kitesurfer1404/WS2812FX #112: Examples Gives Error and don't compile

prof955, 2018-08-14:
Hello,

i can't compile examples successfully.
i checked; 
    using last version of this library
    using last version of Ada Neo library too.

it gives error like this:
```
Arduino\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'void WS2812FX::setPixelColor(uint16_t, uint32_t)':

Arduino\libraries\WS2812FX\src\WS2812FX.cpp:99:49: error: 'gamma8' was not declared in this scope

     Adafruit_NeoPixel::setPixelColor(n, gamma8(r), gamma8(g), gamma8(b), gamma8(w));

                                                 ^

Arduino\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'void WS2812FX::setPixelColor(uint16_t, uint8_t, uint8_t, uint8_t)':

Arduino\libraries\WS2812FX\src\WS2812FX.cpp:107:49: error: 'gamma8' was not declared in this scope

     Adafruit_NeoPixel::setPixelColor(n, gamma8(r), gamma8(g), gamma8(b));

                                                 ^

Arduino\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'void WS2812FX::setPixelColor(uint16_t, uint8_t, uint8_t, uint8_t, uint8_t)':

Arduino\libraries\WS2812FX\src\WS2812FX.cpp:115:49: error: 'gamma8' was not declared in this scope

     Adafruit_NeoPixel::setPixelColor(n, gamma8(r), gamma8(g), gamma8(b), gamma8(w));

                                                 ^

Arduino\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_running_lights()':

Arduino\libraries\WS2812FX\src\WS2812FX.cpp:738:78: error: 'sine8' was not declared in this scope

     int lum = (int)sine8(((i + SEGMENT_RUNTIME.counter_mode_step) * sineIncr));

                                                                              ^

exit status 1
Error compiling for board Adafruit Feather HUZZAH ESP8266.
```

moose4lord, 2018-08-14:
It looks like you're using an old version of the Adafruit NeoPixel library. Try upgrading to the latest version.

prof955, 2018-08-15:
Yes i was. 
But i checked before i wrote issue here... i was at latest version.

i found the main issue. it wasn't related library(es)
it was Megaupload sync program.
when i update my library it was downloading old(backup) one.

so, don't use sync software for library :)

closed 2018-08-15 by prof955