kitesurfer1404/WS2812FX #114: Brightness slider segments web?

Andriejus, 2018-09-09:
Hello, just wanted to ask if its easy or not to insert Brightness slider into ws2812fx_segments_web ? Need to insert few lines of code or its more complicated?? Maybe someone has already done that and could share?:)

moose4lord, 2018-09-12:
Since [materialize](https://materializecss.com/) recently created their v1.0.0 release, I've been wanting to revisit the ws2812fx_segments_web sketch and update it to the latest version of materialize. Materialize has good support for sliders, so I'll look at adding a brightness slider.

moose4lord, 2018-09-13:
I have a working version of ws2812fx_segments_web which incorporates the latest materialize code and adds a brightness slider. Give it a try and let me know what you think.

[ws2812fx_segments_web.zip](https://github.com/kitesurfer1404/WS2812FX/files/2380317/ws2812fx_segments_web.zip)

Andriejus, 2018-09-17:
It works very nice! Thank you very much. Now I can proceed with my stair lighting project:)

Andriejus, 2018-10-01:
Hello again. I understand maybe I'm asking too much, but at least could you please add this PIR triggering into ws2812fx_segments_web sketch? I tried, but its too hard for me. Maybe I will be able to add LDR and other features by myself :/

moose4lord, 2018-10-03:
Hello, the key to developing a moderately complex project, like yours, is to break it down into smaller, manageable experiments, then integrate what you learned from your experiments into a complete application. You would do well to play with the PIR sensors by themselves to get a good understanding of how that hardware and software work. Then do the same with the LDR sensor and the ws2812fx library. Then try to integrate everything together. Being able to decompose a large problem into smaller problems is a valuable skill, which you can only learn by doing.

Having said that, I understand that sometimes projects just drag on and on, and you just want to get it done. So I've hacked the ws2812fx_segments_web sketch to implement some of your ideas. The PIR sensors, segment delay and toggle switch items are in the code for you to look at. Hopefully that will get you going down the right path. I'll leave the LDR integration to you. I know you can do it.

Good luck

[ws2812fx_stairs.zip](https://github.com/kitesurfer1404/WS2812FX/files/2442413/ws2812fx_stairs.zip)

Andriejus, 2018-10-04:
Hello, Thank you so much for your answer and the sketch! I'm really learning a lot by analyzing the code you have written.  Now its clear for me how to add the LDR. Thank you!
**I have the last question**:  as I see in the code that when the PIR sensors are triggered, so it begins to show "fixed" animation FX_MODE_COLOR_WIPE
is it possible to make that when the PIR sensors are triggered, so anim

moose4lord, 2018-10-06:
Oh sorry, when we were talking about your project a few months ago, I think you were hardcoding the animations. Now I understand that you're integrating the web features.

I have refactored the ws2812fx_stairs sketch to use the data supplied by the web server. The web server part was meant to be a standalone app, so it's a bit of a kludge. I also added some code to save the current setup to eeprom, so if the esp8266 is rebooted (or power cycled), it should restore the previous settings. I haven't done a lot of testing, but it seems to work ok.

[ws2812fx_stairs2.zip](https://github.com/kitesurfer1404/WS2812FX/files/2453088/ws2812fx_stairs2.zip)

Andriejus, 2018-10-12:
If im thinking right, so I need to combine two effects together: 1.color wipe and 2.fade
Anyone knows how to do it right?

moose4lord, 2018-10-14:
I gave it some thought and I'm having a hard time coming up with an easy algorithm to implement your custom fade effect. One problem is that the WS2812FX effects are cyclical, looping over and over again, but it seems you want the stair segments to fade-in and stay on, then after a time fade-out and stay off. Another problem is that the custom effect has to be reentrant if it's to be shared among all the segments, meaning it can't modify global or static variables. So while the fade seems pretty basic, the implementation is a bit technical. :(

It might be easiest to make this custom effect use a predefined, lookup-table to manage the LED animation. Let me ask you a couple questions. How many LEDs are on each step and how many steps are there? Also, would each step use the same color palette, or would each step use a different color? If the number of steps is not too large and the steps can share a color palette, I think a lookup-table approach would work.

Andriejus, 2018-10-15:
there are 15 LEDs in each step and 15 steps. Total 225 LEDs.
The plan was to be able to choose a different collor for each segment. Like now I can choose for each segment different color, effect or speed. But if its complicated, so its no problem, that each segment would be the same color. Its the last stages of my project. The most important things are already done.

Andriejus, 2018-10-19:
Thank you! It works! Very nice fade in effect. Need more time to play with different settings of this effect. Now Im thinking how to hardcode inactive state to custom fade out effect. Here are my thoughts:
Now in web browser drop down menu at the bottom I have two custom effects Custom Fade In and Custom Fade Out which as I understand are FX_MODE_CUSTOM_0 and FX_MODE_CUSTOM_1 right?
So if I edit inactive state like this:
if(isStairActive) { // activate step
      memcpy(&segments[stepNum], &savedSegments[stepNum], sizeof(WS2812FX::segment));
      segRts[stepNum].aux_param = 0; // reset the custom fade
    } else { // deactivate step
      ws2812fx.setSegment(stepNum, firstLed, lastLed, FX_MODE_CUSTOM_1, BLUE, 1000, NO_OPTIONS);
    }
should it work? Im not at home for few days, so dont have possibility to try. Thats why Im asking which way I should go in order to hardcode custom fade out effect into inactive state.

moose4lord, 2018-10-23:
This discussion has veered off course. The brightness slider in the web interface has been added. Please open a new issue to discuss other aspects of your project.

closed 2018-10-23 by moose4lord