kitesurfer1404/WS2812FX #116: ws2812fx.init(); causes wdt resets

jasonblewis, 2018-09-20:
I'm unable to get beyond initialising the ws2812fx. In my code, as soon as I try and init(), the esp8266 seems to hang, and the watchdog timer starts firing every few seconds.

```#include <WS2812FX.h>

#define LED_COUNT 8
#define LED_PIN 7

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.print("initialising\n");
  ws2812fx.init();
//  ws2812fx.setBrightness(100);
//  ws2812fx.setSpeed(200);
//  ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
//  ws2812fx.start();

}

void loop() {
  // put your main code here, to run repeatedly:
  //Serial.print("hello world\n");
  //delay(1000);
  ws2812fx.service();

}
```

and on the serial console:
```
08:28:05.818 -> initialising
08:28:14.238 -> 
08:28:14.238 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
08:28:14.238 -> 
08:28:14.238 -> wdt reset
08:28:14.238 -> load 0x4010f000, len 1384, room 16 
08:28:14.238 -> tail 8
08:28:14.238 -> chksum 0x2d
08:28:14.239 -> csum 0x2d
08:28:14.239 -> vbb28d4a3
08:28:14.239 -> ~ld
08:28:14.271 -> initialising
08:28:22.656 -> 
08:28:22.656 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
08:28:22.656 -> 
08:28:22.656 -> wdt reset
08:28:22.656 -> load 0x4010f000, len 1384, room 16 
08:28:22.689 -> tail 8
08:28:22.689 -> chksum 0x2d
08:28:22.689 -> csum 0x2d
08:28:22.689 -> vbb28d4a3
08:28:22.689 -> ~ld
08:28:22.722 -> initialising
```

using a wemos d1 mini, arduino IDE 
ws2812fx v 1.1.0
Adafruit Neopixel library v1.1.6

I have tested on two different wemos d1 mini modules with the same result.

moose4lord, 2018-09-21:
This is a pretty common mistake for beginners. I think you're confusing the GPIO7 pin with the pin that's labelled "D7" on your board. When you specify a pin number in a sketch, you always use the GPIO number. So
```c++
#define LED_PIN 7
```
means use GPIO7 to drive the LED strip. Unfortunately, that's not the pin labelled "D7" on your board.

GPIO7 is used internally as part of the SPI bus that connects the ESP8266 with the onboard flash memory module (it's usually not even exposed on the board as a pin you can make a connection to). So using it in your sketch to drive the LEDs creates a hardware conflict with the flash memory and your WeMos will surely crash.

If you look here [https://wiki.wemos.cc/products:d1:d1_mini](https://wiki.wemos.cc/products:d1:d1_mini) you'll see the WeMos D1 Mini connects the ESP8266's GPIO13 to the pin labeled "D7" on the board. So this what you want in your sketch
```c++
#define LED_PIN 13
```

Or, because this mismatch between ESP8266 GPIO pins and pin labels on the board has caused so many headaches, the ESP8266 core team has created macros for each of the ESP8266 boards that do the correct pin mapping. So you can also use this
```c++
#define LED_PIN D7   // GPIO13
```

jasonblewis, 2018-09-21:
thank you! that was the exact problem

on a small side note, I don't seem to have the D7 type macros. Is there some #include I need to use to access those or something?

closed 2018-09-21 by jasonblewis

moose4lord, 2018-09-21:
Just FYI, when you install the Arduino IDE it dumps all the code into a folder called "Arduino15". On a Mac it's in /Users/{{userid}}/Library/arduino15, but I'm not really sure where it is on a Windows PC. Anyway, in that folder you'll find the folder packages/esp8266/hardware/esp8266/2.4.2/variants, which contains the customization files for each the the boards that the IDE supports. In there you'll find various pins_arduino.h files which define the D7 macro, and many others, specific to the board you're using. The IDE takes care of automatically #including the proper pins_arduino.h file when you compile your sketch.