kitesurfer1404/WS2812FX #117: Step by step detail to creat a custom effect

RAGE-Sonix, 2018-09-24:
Are there  more detailed instructions on how to create a custom effect?  I am really new to this and would like to have an LEDStrip flashing blue and red (like police lights).

My goal would to have 10 lights quickly flash blue 3 times, skipping the next 10 lights (which would flash red when the blue has flashed), then the next 10 quickly flash blue in sync with the first set. 
The red would flash in place where the blue ones are not flashing in sync as well.
 I'd like to have them on a LEDstrip that is 140 lights in length.

Similar to this: https://www.youtube.com/watch?v=LQ4hJD_b_tY

moose4lord, 2018-09-26:
There is a blurb about creating custom effects in the User's Guide found in the library's "extras" folder. Admittedly, it's a pretty terse explanation. For more detailed instructions I'd normally walk you through the ws2812fx_custom_effect sketch in the examples folder, but I just noticed it has a few bugs in it that prevent it from compiling. Doh! I'll fix that.

Meanwhile, the outline for creating a custom effect involves creating a function (usually called "myCustomEffect"), in the function get a reference to the active segment, use the segment reference to get the start and stop LED index, then write an algorithm to manipulate the LEDs to create the effect you're looking for. The skeleton for a myCustomEffect function looks like this:
```c++
uint16_t myCustomEffect(void) {
  WS2812FX::Segment* seg = ws2812fx.getSegment(); // get the current segment

  for(uint16_t i=seg->start; i<=seg->stop; i++) { // loop over each LED in the segment
    // some algorithm to manipulate the LEDs with setPixelColor() calls
  }

  return seg->speed; // return the delay until the next animation step (in msec)
}
```

To be sure, it's not trivial, especially for a beginner.

To make amends for not catching the bugs in the ws2812fx_custom_effect example sketch, I took a stab at creating the custom effect you're looking for and came up with the following sketch. It may not be exactly what you had in mind, but it should get you started.
```c++
#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN 5
#define LEDS_PER_SECTION 10

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

// colors
#define NUM_COLORS 12
uint32_t colors1[] = {BLACK, BLACK, BLACK, BLACK, BLACK, BLACK, BLUE,  BLACK, BLUE,  BLACK, BLUE,  BLACK}; // even sections
uint32_t colors2[] = {RED,   BLACK, RED,   BLACK, RED,   BLACK, BLACK, BLACK, BLACK, BLACK, BLACK, BLACK}; // odd sections
uint8_t  colorIndex = 0;
  
void setup() {
  Serial.begin(115200);
  
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setCustomMode(myCustomEffect);

  ws2812fx.setSegment(0, 0, LED_COUNT-1, FX_MODE_CUSTOM, RED, 100, false);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}

uint16_t myCustomEffect(void) {
  WS2812FX::Segment* seg = ws2812fx.getSegment(); // get the current segment

  for(uint16_t i=seg->start; i<=seg->stop; i++) {
    int section = (i - seg->start) / LEDS_PER_SECTION;
    if(section % 2) { // is it an even or odd section?
      ws2812fx.setPixelColor(i, colors2[colorIndex]); // odd section, i.e. 10-19, 30-39, 50-59, etc
    } else {
      ws2812fx.setPixelColor(i, colors1[colorIndex]); // even section, i.e. 0-9, 20-29, 40-49, etc
    }
  }

  colorIndex = (colorIndex + 1) % NUM_COLORS;
  return seg->speed; // return the delay until the next animation step (in msec)
}
```

Andriejus, 2018-10-11:
could you explain how to insert such effect into existing sketch "segments web" custom effect0? Or how to insert for example "BlockDissolve.h" which I can find in "WS2812FX / src / custom / " into custom effect1? I find it a little confusing as I can not find step by step instructions :)

moose4lord, 2018-10-11:
Have you looked at the ws2812fx_custom_effect2 example sketch in the library's "examples" folder? That sketch shows how to use one of the pre-made custom effects from the src/custom folder.

For each custom effect you want to use, you #include the effect at the top of your sketch and then assign it to one of the four custom effect slots by calling the setCustomMode() function. For example, add these two lines to ws2812fx_segments_web.ino:
```c++
#include "custom/BlockDissolve.h"
...
void setup() {
...
  uint8_t blockDissolveMode  = ws2812fx.setCustomMode(F("Block Dissolve"), blockDissolve);
...
}
```
The block dissolve effect will then show up at the bottom of the effects dropdown box in the web interface.

Andriejus, 2018-10-11:
Thank you! So I don't need to edit WS2812FX.cpp file?

moose4lord, 2018-10-11:
That's right, to incorporate a custom effect, you don't need to edit WS2812FX.cpp.

closed 2018-10-23 by moose4lord