kitesurfer1404/WS2812FX #119: Segment effects auto mode cycle

Giuseppe-P, 2018-10-05:
I set in setup in the auto mode cycle example of the led segments to move a number of leds in the preferred direction ... But I noticed that at start the function in the setMode loop useful for changing scene only works on the first segment while the others remain with the effect set in the setup. How can I solve?

moose4lord, 2018-10-06:
The auto_mode_cycle example is an older sketch that was written before the WS2812FX library supported multiple segments. It could be updated, but unfortunately the functions used to dynamically query/update the running animation (getMode/setMode, getSpeed/setSpeed and getColor/setColor) have not been updated to support multiple segments. They only query/update the first segment. So you'd have to refactor those functions if you wanted to use more then one segment.

Giuseppe-P, 2018-10-06:
Could you help me

moose4lord, 2018-10-06:
I'm not sure what you're trying to do. The auto_mode_cycle sketch just loops through all the different effect modes. Are you trying to have each segment loop through the same set of effects, or each segment loops through a different set of effects?

Giuseppe-P, 2018-10-09:
I would like a segment to use an automatic mode of effects for each individual segment, with the possibility of using them manually by setting each individual effect ... let me explain better ..... I have a strip of LEDs of which I cut 9 strips of 17 always connected in series and arranged vertically creating a matrix ... to avoid the problem of the automatic cycle I could connect them in parallel, but I will not be able to use this matrix to create different effects on every single line ... whereas if you could manage both auto that manual, configuring everything in sketched segments the problem would not be there

moose4lord, 2018-10-11:
The esp8266_webinterface example sketch has a feature similar to what you're looking for. In that sketch you can specify a list of modes (the myModes[] array) you want to cycle through, and if you click the auto button in the web interface, it'll start cycling through the list. Although that sketch assumes you only have one segment.

To expand the concept to use more then one segment, you'd need to create a multidimensional myModes[][] array to store modes for each segment. I've created a demo sketch that does that for two segments. You adapt it to your situation. Note, this demo sketch uses some new features in the latest version of the WS2812FX library. Please download the latest version (v1.1.1) using the Arduino IDE Library Manager.

```c++
#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN D1

#define TIMER_MS 5000

#define NUM_SEGMENTS 2
#define NUM_MODES 4
uint8_t myModes[][NUM_MODES] = { // list of modes for each segment
  {FX_MODE_STATIC, FX_MODE_BLINK, FX_MODE_SCAN,      FX_MODE_RAINBOW_CYCLE}, // modes for first segment
  {FX_MODE_FADE,   FX_MODE_COMET, FX_MODE_DUAL_SCAN, FX_MODE_CHASE_COLOR}    // modes for second segment
};
uint8_t modeIndex = 0;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(255);

  // init default segments
  const uint32_t colors[] ={RED, GREEN, BLUE};
  ws2812fx.setSegment(0,           0,  (LED_COUNT/2)-1, FX_MODE_STATIC, colors, 1000, false);
  ws2812fx.setSegment(1, LED_COUNT/2,      LED_COUNT-1, FX_MODE_STATIC, colors, 1000, false);

  ws2812fx.start();
}

void loop() {
  static unsigned long then = 0;
  unsigned long now = millis();

  ws2812fx.service();

  if(now > (then + TIMER_MS)) {
    ws2812fx.resetSegmentRuntimes();
    for(uint8_t i=0; i < NUM_SEGMENTS; i++) { // update the mode for all segments
      ws2812fx.setMode(i, myModes[i][modeIndex]);
    }
    modeIndex = (modeIndex + 1) % NUM_MODES; // increment the mode index
    then = now;
  }
}
```

Giuseppe-P, 2018-10-13:
I HAVE CHANGED AS FOLLOWS, I HAVE TO IMPLEMENT THE WEB SERVER AND THE GAME IS DONE.

THANK YOU WILL BE UPDATED

#include <WS2812FX.h>

#define LED_COUNT 54
#define LED_PIN 2

#define TIMER_MS 5000

#define NUM_SEGMENTS 3
#define NUM_MODES 6
uint8_t myModes[][NUM_MODES] = { // list of modes for each segment
  {FX_MODE_LARSON_SCANNER, FX_MODE_RAINBOW_CYCLE, FX_MODE_THEATER_CHASE,FX_MODE_TWINKLE_FADE,FX_MODE_MERRY_CHRISTMAS,FX_MODE_HALLOWEEN}, // modes for first segment
  {FX_MODE_LARSON_SCANNER, FX_MODE_RAINBOW_CYCLE, FX_MODE_THEATER_CHASE,FX_MODE_TWINKLE_FADE,FX_MODE_MERRY_CHRISTMAS,FX_MODE_HALLOWEEN},    // modes for second segment
  {FX_MODE_LARSON_SCANNER, FX_MODE_RAINBOW_CYCLE, FX_MODE_THEATER_CHASE,FX_MODE_TWINKLE_FADE,FX_MODE_MERRY_CHRISTMAS,FX_MODE_HALLOWEEN}    // modes for second segment
};
uint8_t modeIndex = 0;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(255);

  // init default segments
  const uint32_t colors[] ={RED, GREEN, BLUE};
  ws2812fx.setSegment(0,0,  17, FX_MODE_STATIC, colors, 1000, false);
  ws2812fx.setSegment(1,18,34, FX_MODE_STATIC, colors, 1000, true);
  ws2812fx.setSegment(2,35,51, FX_MODE_STATIC, colors, 1000, false);

  ws2812fx.start();
}

void loop() {
  static unsigned long then = 0;
  unsigned long now = millis();

  ws2812fx.service();

  if(now > (then + TIMER_MS)) {
    ws2812fx.resetSegmentRuntimes();
    for(uint8_t i=0; i < NUM_SEGMENTS; i++) { // update the mode for all segments
      ws2812fx.setMode(i, myModes[i][modeIndex]);
    }
    modeIndex = (modeIndex + 1) % NUM_MODES; // increment the mode index
    then = now;
  }
}

closed 2018-10-23 by moose4lord

GBFerns, 2021-12-18:
Hi, is it possible to set the ws2812fx.setMode to AUTO_CYCLE mode in the WS2812FX Webinterface file. please help

moose4lord, 2021-12-18:
You wouldn't use ws2812fx.setMode() to enable auto cycle mode, since it's used to set the current lighting effect. However, if you're referring to the esp8266_webinterface example sketch, then changing the Initialization of the auto_cycle variable from _false_ to _true_ will enable the auto cycle feature when your device is powered on.
```c++
bool auto_cycle = true; // change from false to true
```