kitesurfer1404/WS2812FX #120: [Q] Increase Led count in 22 - Twinkle Fade Random

Kaibob2, 2018-10-08:
Hi all,
thanks for this amazing library. It is really perfect for combining the power of an ESP8266 with the advantages of the WS2812. Great work.
I have a question:
I'm building a star sky in my little daughthers room and would like to use the Twinkle fade random function for it. Is there a way to modify (increase) the number of LEDs which are lit at the same time. I currently use a 50 LED Array and with the current setting there are only a few LEDs lit at the same time.
Thanks in advance.

kitesurfer1404, 2018-10-08:
Hi there. Thanks for the nice words! :)

Great idea with the star sky! Please add a video or a link to the main README when finished. I really would like to see this. Maybe something I can build for some children in my family. 

Take a look at src/WS2812FX.cpp at line 884.
https://github.com/kitesurfer1404/WS2812FX/blob/master/src/WS2812FX.cpp#L884

There the condition for "new" lights are "hidden": if(random8(3) == 0) means, if a random number between 0 and 3 (0, 1, 2) is 0, another random LED is lit. Which is statistically in 1 of 3 turns the case. For more LEDs coming on, just lower the number inside of the random8() function to 2.

Also, this effect might need further tweaking in the future to control the probability of "newly" lit LEDs by a parameter.

Kaibob2, 2018-10-08:
Thanks. I will upload a video to Youtube and edit the Readme then. I just found this line, but my coding skills are a little limited, so i increased this number which didn't work. Everything i know about C++ comes from try and error by modifying the Marlin 3D printer software and historical Basic programming.
So, thanks for the fast reply and explanation.
And yes, a parameter for the minimum number of LEDs lit at the same time would be great! Especially for larger arrays.

Kaibob2, 2018-10-08:
I tested random8(2) and it's better than before but i guess this is still not enough LEDs :-|
Any other hint in this direction?

Kaibob2, 2018-10-08:
```
 // if(random8(2) == 0) {												
    setPixelColor(SEGMENT.start + random(SEGMENT_LENGTH), color);
 // }
```

This comes close, but a parameter would be the much better alternative :)

kitesurfer1404, 2018-10-08:
Next step would be adding more lines of 
setPixelColor(SEGMENT.start + random(SEGMENT_LENGTH), color);
so more random pixels are set. Not it's just one at a time. Adding some more lines will light up more at every step.

moose4lord, 2018-10-08:
Another trick you can try is to subdivide your LEDs into segments. Since the probability of an LED "twinkling" is the same _**per segment**_, the more segments you have, the more LEDs will twinkle. You can use this quicky code snippet in your sketch to play with the number of segments:
```c++
  uint8_t numSegments = 5;
  for(uint8_t i=0; i<numSegments; i++) {
    uint16_t firstLED = i * (LED_COUNT / numSegments);
    uint16_t lastLED = firstLED + (LED_COUNT / numSegments) - 1;
    ws2812fx.setSegment(i, firstLED, lastLED, FX_MODE_TWINKLE_FADE, WHITE, 1000, false);
  }
```
Adjust the numSegments parameter to try different scenarios. Keep in mind the maximum number of segments is 10.

Kaibob2, 2018-10-10:
Thanks, i chose kitesurfers solution for the beginning, but i keep the segments solution in mind. Maybe it becomes relevant when the whole setup is ready.
I have some more questions:
Is it possible to add more fade stages? The fade out looks quite rough.
Is there a fade in function possible?
Is there a LED limit? I'm planning to use around 600.

Btw.: I got the sketch going pretty well by now. Based on the WS2812FX Webinterface example sketch I added a 32x128 OLED which displays mode, brightness in % and speed. Added the OTA function and reused the auto cycle buttons in the webinterface to be ON and OFF.

moose4lord, 2018-10-11:
You can add more granularity to the fade by using the options parameter in the setSegment() function. For example this code will create a slower, more gradual fade:
```c++
  const uint32_t colors[] = {RED, BLACK, BLACK};
  ws2812fx.setSegment(0,  0, LED_COUNT-1, FX_MODE_TWINKLE_FADE_RANDOM, colors, 200, FADE_XXSLOW);
```
Search the User's Guide for "fade rate" for more details.

The color_blend() function can be used for fade-in effects, although it's currently only used by the FX_MODE_FADE effect. You'd have to create a custom effect if you wanted to use color_blend() in the twinkle effect.

The WS2812FX library can manage 600 LEDs. My longest strip of ws2812s has 300 LEDs in it, and that works fine. You just have to pay attention to the power requirements. 600 LEDs can draw a lot of current. If each RGB element in a ws2812 is at maximum brightness, the device can draw about 60mA. Multiply 600 devices times 5 volts and we're talking roughly a 200 watt power supply. If you're only going to use the twinkle effect, which doesn't have a lot of LEDs on simultaneously, you can probably get by with something smaller.

Kaibob2, 2018-10-11:
@moose4lord I just recognized, you committed a change to examples/ws2812fx_segments_web/index.html.cpp which adds a brightness slider. Uhuh, this is great. I was planning to do this myself tomorrow, but it would have cost me a lot of brain grease :) Thanks for that!
Would it be possible for you to also add a dropdown for the fade options in addition to the reverse switch? I'm struggling with the construction of the options argument when there are 2 options (reverse + fade)

moose4lord, 2018-10-14:
Yeah, updating the **ws2812fx_segments_web** web interface to include the fade, reverse and gamma options has been on my to-do list for a while. I'll see if I can get to it this week.

Kaibob2, 2018-10-16:
I don't get it. Creating a custom effect is beyond my abilities. Would it be possible for one of you to create a Twinkle fade random effect which fades IN and OUT?

moose4lord, 2018-10-16:
I'm not sure I understand your question. Are you trying to get the entire segment of LEDs to fade in and out as the twinkle_fade_random effect is running? Or are you trying to get the "twinkling" LED, which currently just fades out, to fade in as well?

Kaibob2, 2018-10-16:
Thanks for the fast response. I mean the second version. The "twinkling" Led should fade in, maybe stay lit one or to seconds and then fade out. It's still for my stary sky project. So it would be nice if I could implement it as standard effect into the web_segments sketch.

moose4lord, 2018-10-16:
btw, I got a chance to work on the ws2812fx_segments_web sketch to add the reverse/gamma/fade-rate features into the web interface. Try it out and let me know if you see any issues.

[ws2812fx_segments_web.zip](https://github.com/kitesurfer1404/WS2812FX/files/2484864/ws2812fx_segments_web.zip)

Kaibob2, 2018-10-17:
Hi Keith,
nice, but doesn't work correctly. All test done with one segment, 50 pixels, speed 2000, Brightness 180

This
[image]
works

This
[image]
lights all LEDs up in green and twinkles one LEDs with random color and fades XXSLOW back to green.

This
[image]
works, but doesn't respect the fade setting. It fades with default setting.

This
[image]
lights all LEDs up in green and twinkles one LEDs with random color and fades XXSLOW back to green.

This
[image]
works, but doesn't respect the fade setting. It fades with default setting.

moose4lord, 2018-10-17:
Actually it's working as designed. When fade rate is set to DEFAULT, the "fade to" color is always black, meaning regardless of the segment's color values, the LEDs will always fade to off. This is the legacy behavior. If you choose any other value for fade rate, the "fade to" color is the segment's color[1] setting. Since color[1] is set to green in your screen shots, all the LEDs fade to green. If you want to fade to black, set color[1] to black. Or a dim blue might look nice in your application.

Since the Twinkle Fade Random effect doesn't really have a forward or reverse direction (it involves random LEDs and random colors), the reverse setting has no affect.

The Gamma Correction is working correctly, but it's overwhelming the fade rate setting. Essentially setting fade rate to XXSLOW is trying to change the background LEDs to green, but it's doing it so slowly that the gamma correction overpowers the fade and forces the LEDs off. If you set the fade rate to SLOW or XSLOW, the fade it fast enough to prevent gamma correction's overpowering affect, and you'll see a (dim) green background.

Kaibob2, 2018-10-17:
Damn, you're right. Absolutely my fault. Sorry for this. Knowing this i like it perfect and it works like a charm. Saving directly to EEPROM now is also a very appreciated feature.
But...somehow the brightness setting doesn't seem to work properly. Changing the brightness slider and saving blinks the strip but doesn't change brightness. Loading afterwards sets the brightness(slider) back to the previous value. The crazy thing is, that i guess it worked a few times randomly.

Kaibob2, 2018-10-17:
Man, i don't know what's going on with my ESP. I just reflashed the Sketch again without changing anything and the brightness issue disappeared magically. So nevermind the previous posts. Great job.

Kaibob2, 2018-10-17:
Is it possible to set the brightness per segment or is it better, like now, to change the overall brightness. I'm asking because in my setup i could make some stars twinkle brighter than the other.
Or maybe two slider, one overall brightness and one for the segment brightness.

Kaibob2, 2018-10-28:
[image]

It's evolving :) 
Quick question: Is it possible to have more than one memory slot to store different settings? 
I use 9 segments at the moment and changing them is always quite a lot of clicking.

moose4lord, 2018-10-28:
Looks good! What kind of power supply are you using to power so many LEDs?

The ws2812fx_segments_web sketch only has provisions for a single configuration of segments. The ws2812fx_patterns_web may be more of what you're looking for. It has provisions for eight patterns, each with up to ten segments. It does not have a built-in web interface, but it does have a web client that can receive pattern data in JSON format. If you have an Android device, the [LEDfx](https://play.google.com/store/apps/details?id=com.champlainsystems.ledfx&hl=en_US) app provides a frontend that can be used to program the patterns. See the [LEDfx readme](http://champlainsystems.com/ledfx/) for more info.

Kaibob2, 2018-10-29:
I see. Will look into it, but i guess this will not fit my needs. I have a MosFET in line which i need to control via the web interface. The Android app is not changeable to fit this need.
Maybe i just add a "Try" beside the "save" button which sets the segments but dosn't write the settings to the eeprom.
Regarding the Power supply a considered a 30A China switching power supply, but as most of the time it will only "twinkle" up there, there is no need for a 30Amps supply. Beside that, i don't trust the china supplies. Especially not when they are hanging over my daughters bed :)
I now use a 8A Meanwell supply. This can power all LEDs at 100% brightness in a single (R or G or B) colour. The problem is, that it delivers current until the voltage drops below 3.3V and at that point i will be in a loop as i cannot switch of the strip via the webinterface.
I have a ACS758LCB current sensor lying around and I will put in line to monitor the current. If it exceeds 8A open the MosFET and blink some LEDs for help...done.
Yes, ist a little overkill, but why not :)

kitesurfer1404, 2018-10-29:
I have a very huge halloween/christmas lights installation running at the moment. Powered by three 40A Meanwell powersupplies. I selected them over the chinese cheap ones because of the safety as well.
I'll implement a reduced brightness mode so I can run them without active cooling (fan) at night because they are located in my bedroom windows. Also, I wanted to experiment with a software-side current-limit. Like "max 10A --> if exceeded, reduce brightness even more" 
Maybe that's an idea for your star sky as well. Just calculate the current power and cut down on the brightness.

Some impressions so far:

[image]
[image]

I run 3 of these in the windows, 188 LEDs each.
[image]

And additionally this evil face (just red LED strips).
[image]

Adding more for christmas. :)

Kaibob2, 2018-10-29:
Nice! I like well designed power circuitry! This plywood installation and ESP on gfk breadboard and so on is not my favourite, but my daughter is 11 month old and time is rare.
Calculating the current on the software side would make it necessary to know how many Leds are on at what brightness at which color. I have no idea how to do that :)
The current sensor is capable of sensing 100A and output 0.6 to 5V in linear relation. So I have a live feedback of the REAL value.

kitesurfer1404, 2018-10-29:
Thanks!
There are pros and cons for both solutions, I guess. In hard- and software. Software may crash or calculate wrong in some situations. Otherwise: when the hardware solution detects the current as too high, the current is already too high. Resulting in a voltage drop and maybe freezing of the controller again.

To do the calculations in software, you can just iterate over all LEDs after the FX is written to the memory but before the data is send to the strip, using ws2812fx.getPixelColor() to get the RGB-value of each pixel and sum it all up (20mA * (R / 255) + 20mA * (G / 255) + 20mA * (B / 255)) 
Also, incorporate the set brightness of the strip. And add ~1mA for each LEDs internal controller.

That's strongly simplified but I hope, you get the idea. More research is needed to get a good representation of the real world power consumption of the pixels.

Some day I'll add some examples to the library when I got this figured out in detail and working.

Anyway. I have to show your project to my sister, so I have to build a star sky for her child next year as well.

Kaibob2, 2018-10-29:
Sounds interesting as well. Where would be the point to do the calculation? Between line 185 and 186 in the web sketch?
Feel free to distribute the project. My mother in law is currently making a cloth cover for the whole quarter circle, so, in fact no one cares how it looks below ;) 
The cloth blurs the Leds an makes the effect more chilling.

kitesurfer1404, 2018-10-29:
I cannot identify lines 185 in the web example for this.

My guess would be before line 87 in WS2812FX.cpp
https://github.com/kitesurfer1404/WS2812FX/blob/master/src/WS2812FX.cpp#L87
That's where show() sends out the data. For now, you would have to hack it into that code.

For the long run: I'm thinking of some well thought out compiler instructions to have the option to include the current limit functions as well as not including them for some more program memory. Hmmm...

Kaibob2, 2018-10-29:
Oh oh, I suppose this is, again, beyond my coding skills. But I will give it a try tomorrow.

Kaibob2, 2018-10-30:
Well, like expected I reached my limit quite early when trying to implement the software current calculation. I then turned to the current sensor alternative, but this isn't the yellow from the egg, too :)
When I call the current sense function after a settings change via the interface the strip is off and this leads to only sensing the standby current.
When I include the current function into the loop it's called much too often.
Beside that a 100A is not sensitive enough to securely sense 8 or 9 amps PLUS a 10bit ADC doesn't make it better. So this isn't an option anymore. 
Do you think you could do me a favor and dig a little deeper into the current calculation code?

tobi01001, 2018-10-30:
Hi, 

The calculation is estimated anyway. So at best you could verify with an actual measurement but here you go: (don't know if it will work out of the box):

```cpp
// little helper function to calculate and return the current
uint16_t get_req_current_in_mA()
{
  uint16_t current = 1*LED_COUNT; // current drawn from the LED chip (in each pixel)
  uint32 color = 0x0;
  for(uint16_t i=0; i<LED_COUNT; i++)
  {
    color = ws2812fx.getPixelColor(i);
    current += (20 * ((color >> 16) & 0xFF) / 255); // current from red led
    current += (20 * ((color >>  8) & 0xFF) / 255); // current from green led
    current += (20 * ((color >>  0) & 0xFF) / 255); // current from blue led (shift not neccessary (zero)) but looks more complete
  }
  // do not know how to handle global brightness. At the moment, the current reflects the max without adjustment. 
// Linear reduction would lead to something like (uncomment to try):
// current = current * ws2812fx.getBrightness() / 255; // reduce according to brightness
  return current;
}
```
The function / or calculation would needed to be called/done right before `show()` in the library which is not possible without changing the library itself.

However, @kitesurfer1404 (or @moose4lord ?) were kind enough to provide the possibility of a 'customshow`. Therefore, you could use that to calculate the current right before writing the data to the LEDs (not sure if that will work :-)):

For the ease of use, create a global variable keeping the current (somewhere before 'void setup() {...`would do: `uint16_t global_current = 0;

add your custom show function:
```cpp
void my_customShow(void)
{
  global_current = get_req_current_in_mA(); // calculate current for the frame and store it in the global variable...
  Adafruit_NeoPixel::show(); // this is the part of my uncertainty.... 
                                            // I don't know if calling show like this works from the sketch 
}
```

In `setup()`make ws2812fx aware of your custom show function:

```cpp
void setup() {...

...
  ws2812fx.setCustomShow(my_customShow);
...
}
```

Now you just need to make use of the calculated current. At first - in order to verify if it works or not - you could send it to the terminal with Serial.print... Just do that in the `loop()` (not every loop of course)...
You could also add a kind of alarm if the current exceeds a certain value with a simple `if(global_current > MAX_CURRENT) ...`. This could stop the animation, switch the leds to black or even drive a digital output to siwth off power...
Another option would obviously be to limit the LEDs current to be compatible with your installation. But this is another separate task... It should work with reducing the brightness to the acceptable level (as this equally darkens the complete strip)....

@kitesurfer1404 Any improvements or comments (especially on the custom show?)? Would be good to have / get something in the library or actually adafruits library itself...

Kaibob2, 2018-10-30:
Thanks for the effort, but there seems to be a compile problem when adding the above function to the absolute end of WS2812FX.cpp:

WS2812FX.cpp:1513: error: 'LED_COUNT' was not declared in this scope
and
WS2812FX.cpp:1517: error: 'ws2812fx' was not declared in this scope

Kaibob2, 2018-10-30:
Nevermind the above. The function apparently belongs in the sketch :)
But `cannot call member function 'void Adafruit_NeoPixel::show()' without object` seems to be a problem.

tobi01001, 2018-10-30:
That's what I expected / feared....
I cannot check right now (I only have internet) so we have do do this "quick and dirty".
In WS2812FX.h at between `uint16_t _rand16seed;`and `uint8_t _brightness;`  (L575) add a private variable representing our current `uint16_t _current;`
In line 470 (before `getLength(void)`) add `getCurrent(void),`.
In line 583 (below `void (*customShow)(void) = NULL;`) add `uint16_t get_req_current_in_mA(void);`

In WS2812FX.cpp (at e.g. line 92) insert:
```cpp
uint16_t WS2812FX::get_req_current_in_mA()
{
  uint16_t current = 1*LED_COUNT; // current drawn from the LED chip (in each pixel)
  uint32 color = 0x0;
  for(uint16_t i=0; i<LED_COUNT; i++)
  {
    color = ws2812fx.getPixelColor(i);
    current += (20 * ((color >> 16) & 0xFF) / 255); // current from red led
    current += (20 * ((color >>  8) & 0xFF) / 255); // current from green led
    current += (20 * ((color >>  0) & 0xFF) / 255); // current from blue led (shift not neccessary (zero)) but looks more complete
  }
  // do not know how to handle global brightness. At the moment, the current reflects the max without adjustment. 
// Linear reduction would lead to something like (uncomment to try):
// current = current * ws2812fx.getBrightness() / 255; // reduce according to brightness
  return current;
}
```
in line 125 (so between `void WS2812FX::show(void) {` and `if(customShow == NULL) {` add `_current = get_req_current_in_mA();`.

And after that, add:
```uint16_t WS2812FX::getCurrent(void) {
  return _current;
}
```

This will unfortunately modify the library. The custom show would be easier of course but currently I cannot come up with a solution for that.

With the modifications from above you can check the current anywhere in the sketch with:
`ws2812fx.getCurrent();`
This is rather asynchron and will return the value from the last calculation. As we calculate in the show() function, the LEDs were already written with that current.
Next would be to add a current limiting functionality where you could define the maximum allowed current.
A quick and dirty solution to that would be to skip the show function in case the current is too high (animation gets stuck at the last frame or will skip the bright frames...). Further, a callback function could inform the user about the current being exceeded.

As always, there are multiple ways to do this and certainly better ways to implement this but it should give you a start....

Kaibob2, 2018-10-30:
@tobi01001 Nice! It compiles. But i left the hardware at work, as i have more time there (remember...11 month old daughter :) ) so i can first check the function tomorrow. I did a test before i left work with leaving out the `CustomShow` stuff and putting `global_current = get_req_current_in_mA();` in the main loop. That worked like expected and the current calculation updated according to changes made in the web interface.

kitesurfer1404, 2018-10-30:
I send the idea over to the NeoPixel library, because I think it belongs into the main lib. Maybe we can create a good working example and send it over as a pull request.

https://github.com/adafruit/Adafruit_NeoPixel/issues/181

Kaibob2, 2018-11-01:
I spent many hours now getting this current thing to work and there were a lot of problems to be solved. 
First: 
The specified 20mA are not reachable in reality. My 9 strips are 1.2m long each with a supply at each beginning. The voltage drop over the strip is higher than one would expect, even with both sides connected to Vcc. Plus, the higher the current (brighter) the higher the voltage drop and the lower the current through each subpixel.

Second:
The current of the subpixels isn't equal. In my setup it's R=15mA, G=13mA and B=12mA at full on. Disgusting!

Third:
The global brightness isn't linear. Sure, Leds aren't halogen bulbs. So the calculated current is wronger the lower the brightness gets. I got rid of it using MultiMap. Nice function for not linear mapping https://playground.arduino.cc/Main/MultiMap

But the biggest issue is speed! 
Doing the iteration over 526 Leds over and over takes too much time to go below (I guess) 500ms for the comet effect for example. 

BUT: It works really nice and the accuracy is around 100-200mA. Not bad for all the above said. 
Maybe I can upload my sketch and someone can look over it to verify if it is correct like this or if I coded a handbrake into it ;)

Kaibob2, 2018-11-01:
[image]
[Star-sky.zip](https://github.com/kitesurfer1404/WS2812FX/files/2540104/Star-sky.zip)

There's some german in the sketch, but i guess this won't be a big problem.

moose4lord, 2018-11-01:
Sorry I'm getting to the party late. I wrote some code to try to calculate real-time current demand and adjust the brightness to limit the current. The code to calculate the sum of all RGB intensities is this:
```c++
// return the sum of all RGB intensities
#define BYTES_PER_LED 3 // set to 3 for RGB LEDs or 4 for  RGBW LEDs
uint32_t intensitySum() {
  uint32_t sum = 0;
  uint8_t *pixels = ws2812fx.getPixels();
  for(uint16_t i=0; i <LED_COUNT * BYTES_PER_LED; i++) {
    sum += pixels[i];
  }
  return sum;
}
```
It reads the values directly from the Adafruit_NeoPixel pixels[] byte array, so it's much faster than calling the getPixelColor() function over and over again.

The part that adjusted brightness was much more complicated. I refactored the code into a custom show() function and it looks like this:
```c++
#define MAX_CURRENT         300 // maximum allowed current draw for the entire strip (mA)
#define QUIESCENT_CURRENT    56 // current draw for the entire strip with all LEDs off (mA)
#define INCREMENTAL_CURRENT  40 // increase in current for each intensity step per RGB color (uA)
#define MAX_INTENSITY_SUM   ((MAX_CURRENT - QUIESCENT_CURRENT) * 1000) / INCREMENTAL_CURRENT
void myCustomShow(void) {
  static uint32_t lastSum = 0;
  uint32_t newSum = intensitySum();
  if(lastSum != newSum) {
    if(newSum > MAX_INTENSITY_SUM) {
      Serial.print("intensitySum="); Serial.print(newSum);
      Serial.print(", MAX_INTENSITY_SUM="); Serial.println(MAX_INTENSITY_SUM);

      uint32_t estimatedCurrent = ((newSum * INCREMENTAL_CURRENT) / 1000) + QUIESCENT_CURRENT;
      Serial.print("estimatedCurrent="); Serial.print(estimatedCurrent); Serial.println("mA");

      uint8_t oldBrightness = ws2812fx.Adafruit_NeoPixel::getBrightness();
      Serial.print("old brightness="); Serial.println(oldBrightness);

      uint8_t scaleFactor = (MAX_INTENSITY_SUM * 256) / newSum; // brightness scaling factor
      uint8_t newBrightness = constrain(((oldBrightness * scaleFactor) / 256), BRIGHTNESS_MIN, BRIGHTNESS_MAX);
      Serial.print("new brightness="); Serial.println(newBrightness);

      // call the Adafruit_Neopixel::setBrightness() function, because the WS2812FX::setBrightness() function calls show(), which we don't want.
      // this will cause a discrepancy between WS2812FX's _brightness setting and Adafruit_Neopixel's brightness setting.
      ws2812fx.Adafruit_NeoPixel::setBrightness(newBrightness);

      // for verification, recalculate the intensity sum and current estimate
      newSum = intensitySum();
      Serial.print("new intensitySum="); Serial.println(newSum);
      estimatedCurrent = ((newSum * INCREMENTAL_CURRENT) / 1000) + QUIESCENT_CURRENT;
      Serial.print("new estimatedCurrent="); Serial.print(estimatedCurrent); Serial.println("mA");
    }
    lastSum = newSum;
  }
  ws2812fx.Adafruit_NeoPixel::show();
}
```

The #defines at the top are empirical values I got from taking current measurements of my test setup (a strip of 30 WS2812B LEDs). The code calculates a maximum intensity-sum based on the #define parameters, and if the measured intensity-sum exceeds that maximum, calculates a scaling factor which will reduce the overall brightness to keep the current below MAX_CURRENT. My strip of WS2812B LEDs' current profile is pretty linear, so it works reasonably well for me. Although I don't use any of this code in a production environment. I only created it for testing purposes.

Kaibob2, 2018-11-02:
NICE!! Works like a charm. Much faster than the other solution. What calculation leads you to INCREMENTAL_CURRENT = 40 uA? Is this the "per subpixel current" (don't think so) or the RGB current when the Pixel is white or the combined R, G and B current. Whereas 2 and 3 will result in almost the same :)
In my setup INCREMENTAL_CURRENT = 45 leads to quite acceptable accuracy, but it was try and error.

moose4lord, 2018-11-02:
This is the table of current measuremenst I had made for my 30 LED strip:
```
Strip length - 30 WS2812B LEDs
Quiescent current with all LEDs off - 56mA
int   RED (ma)  Green (mA)  Blue (mA)
---   --------  ----------  ---------
  1 -    57         57         58
  2 -    58         58         59
  4 -    61         60         61
  8 -    65         65         66
 16 -    74         74         75
 32 -    93         93         94
 64 -   130        130        131
128 -   205        205        206
160 -   242        243        243
192 -   283        281        281
224 -   320        319        319
255 -   358        357        357
```
It shows the current measurement when programming all 30 LEDs to the specified color at the specified intensity. The three colors behaved almost identically and had a pretty linear relationship between intensity and current. The **INCREMENTAL_CURRENT** was calculated by taking the current measurement for red at intensity=255 , subtracting the quiescent current, dividing by the intensity, then dividing by the number of LEDs in the strip.
((358ma - 56ma) / 255) / 30 = 0.0395mA
So it's the incremental current added when you increase the intensity by 1 for any one of the RGB subpixels.

Kaibob2, 2018-11-02:
Somehow this was what i expected because it's the only calculation which makes sense. For some reason in my setup this is far from reality. Doing the above calculation gives me INCREMENTAL_CURRENT = 61
With this value inserted into the sketch it doesn't work as expected. With INCREMENTAL_CURRENT = 45 it's perfect. 
Btw. in the table you posted above: Don't you think, that 10mA per subpixel isn't a little low? Like i mentioned earlier i get R=15 B=13 G=12 mA. And this is a 1,2m strip. The drop in your 30cm strip should be much lower resulting in higher subpixel current.

Kaibob2, 2018-11-02:
Edit: It works perfect, as long as "red" isn't involved. My red subpixels draw more current than the two others. This makes it impossible with only one current. Is there a way to split INCREMENTAL_CURRENT up into 3 subpixel currents?

moose4lord, 2018-11-02:
10mA per subpixel at full intensity is what I got when I was making measurements. Someone made current measurements with two different WS2812B strips (apparently from two different manufacturers) and found that the currents were quite a bit different. See this [blog](https://www.pjrc.com/how-much-current-do-ws2812-neopixel-leds-really-use/). So apparently the current demands can be different depending on where the LEDs are fabricated. Btw, my 30 LED test strip is 1M long, not 30cm, so I would expect a voltage drop from one end to the other similar to yours, although that too may depend on where the strip is fabricated.

I suppose it's possible to have three separate INCREMENTAL_CURRENT values, but then you have to know which bytes in the pixels[] array are associated with which colors. A strip of RGB LEDs would have a different calculation than a strip of GRB LEDs. It would get messy.

Kaibob2, 2018-11-02:
I see and agree. My statement 

> INCREMENTAL_CURRENT = 61
With this value inserted into the sketch it doesn't work as expected. With INCREMENTAL_CURRENT = 45 it's perfect.

was not correct. It does work perfect with 61 as current, but only for the red subpixel. I was trying to get a middle for all pixels but this will never work. The problem is in the strip and not in the code. I will simply use 61 (the highest current of all pixels) so any other color will be lower. The brightness reduction will come up a little too early then, but this is unfixable for now.

Kaibob2, 2018-11-02:
Something different: Is it normal, that the speed setting doesn't affect for example fireworks or color wipe (and also most other)? The effect is the same with speed set to 20 or 2000.
With Twinkle fade random the speed affects the effect like expected.

moose4lord, 2018-11-02:
All of the effects, except **STATIC** and **BREATHE**, should respond to the speed setting. You should definitely see a difference between s=20 and s=2000.

Kaibob2, 2018-11-03:
I uploaded the original example sketch ws2812fx_segments_web to the ESP to exclude an issue in my heavily modified sketch and only added the Wifi config and a digitalWrite(D6,HIGH) at the beginning of setup().
Please take a look at the video: https://youtu.be/qbK1xUTKYFw

Kaibob2, 2018-11-03:
EDIT: I made another test.
I started with NUM PIXELS = 50 and one segment including all of them Speed set to 20. 
I then increased NUM PIXELS and the segment length by 50 until i reached the maximum of 526 LEDs.
See what happens: https://youtu.be/fEfmzwj6Bx8

tobi01001, 2018-11-03:
The problem is likely the time it takes to write the frame / pixel data to
the strip...

It takes 30 us to update one (+50 is at the end).
So one frame will be 526 * 30 us + 50 us = 15.830 ms for o e frame.

The whole wipe animation from all black to all red then takes at least 8.3
seconds no matter what speed you use.

There are ways to improve this. I may have a look later... But as Scotty
said: you cannot change the laws of physics. :-)

Kaibob2, 2018-11-03:
Thanks for the fast reply. I would greatly appreciate an improvement here.
Hmm, maybe i need to get the flux capacitor back from the storage :)

moose4lord, 2018-11-03:
Now that I think about it some more, I think the reason you're not seeing any change in the animation speed when you change the speed parameter is because you're not using large enough values for speed. The animation frame rate has a minimum value that depends on the number of LEDs. For example, with COLOR_WIPE the time delay between frames is calculated with this equation:
frame_delay = max(speed / (SEGMENT_LENGTH * 2), SPEED_MIN)
where, for an ESP, SPEED_MIN=2.

So for SEGMENT_LENGTH=526, the speed parameter has to be greater than 2104 for it to have any affect on the frame delay. Try changing the speed parameter from 2000 to 20000 and you should see a difference.

tobi01001, 2018-11-05:
Forgot to mention. I also did some measurements with my 250 leds dev strip:
I did several measurements with several led numbers. 2 shown below.
[image]

[image]

to make a long story short: I also have quite linear and equal behaviour across the several colors.

Kaibob2, 2018-11-06:
First of all thanks to all for these detailed explanations. Sorry that my reply took so long.
With knowing that

> 526 * 30 us + 50 us = 15.830 ms for one frame

is the maximum achievable speed i take this as given. I didn't expect that the pixels are that "slow". I will definetly keep the APA102 Strips in mind where speed is more necessary. In fact this is still the star sky project for my daughter ment to calm her before sleeping. Speed is not wanted anyway :)

The mising linearity in my strip still bothers me, but i think this is the hardware i can't change for now. 
In the diagrams posted by tobi01001 above the red subpixels consume even slightly less current than the others. In my setup it's around 15% more for them. Crazy, but unchangeable.

Thanks again, i learned a lot from the last posts.

moose4lord, 2018-11-14:
Just FYI, I released WS2812FX v1.1.3 just now. It contains two new functions, intensitySum() and intensitySums(), which calculate color intensity sums as we have been discussing. The intensitySums() function returns an array of 4 32-bit numbers, one intensity for each of the 4 RGBW colors. That might be useful if you want to fine tune your code to account for the different current parameters for each RGB LED.

In v1.1.3 I also include an example sketch, ws2812fx_limit_current, which incorporates the custom show() code I posted earlier in this thread.

Kaibob2, 2018-11-14:
Sounds good. The problem i have now is, that i already mounted the whole setup under the ceiling :/ 
I think i will leave the code like it is for the moment. I implemented the httpupdater in advance, but it will be hard to verify if real current and calculated current match while standing on a ladder with a multimeter in the one hand and a notebook in the other. 
This is how it looks, the cloth cover is still missing because Version 1 of it was slightly too big.
Some fine tuning in color distribution and brightness and that's it.
[image]
[image]
[Star-sky on youtube](https://www.youtube.com/watch?v=kCyVQ01EIME)

moose4lord, 2019-01-20:
I think this very interesting conversation has run its course. Closing.

closed 2019-01-20 by moose4lord

Kaibob2, 2019-01-21:
Sorry, i had a lot of other stuff to do and forgot to finish this thread.
Like promised, here is the video of the final product :)
[Starsky on YouTube](https://www.youtube.com/watch?v=bKMRjDSuWyY)

Many thanks for the great support and help. My daughter loves it

kitesurfer1404, 2019-01-21:
Very cool! Thanks for sharing!