kitesurfer1404/WS2812FX #122: Please add more Colors

ChristianKnorr, 2018-10-14:
Hi,
if you write this:
`#define NUM_COLORS        4`
in **WS2812FX.h,** and this:
`ws2812fx.setPixelColor(index, seg->colors[3]);` instead of `ws2812fx.setPixelColor(index, BLACK);`
in **VUMeter.h**
you could write this:
`uint32_t colors[] = {GREEN, YELLOW, RED, MAGENTA};`
in
**ws2812fx_custom_effect2.ino**

Thank you :)

moose4lord, 2018-10-15:
If we were going to change the number of colors, I don't think we could get everyone to agree on what the right number is. Is it 3, or 4, or 5 or 8? 3 seems to strike a good balance between versatility and not wasting resources.

If you just want to patch the VUMeter effect, you can copy the function from VUMeter.h into your sketch and change the line to this (creating a custom custom effect :)
```c++
ws2812fx.setPixelColor(index, MAGENTA);
```

ChristianKnorr, 2018-10-18:
Yes, thank You :)

closed 2018-10-23 by moose4lord