kitesurfer1404/WS2812FX #124: Fade to another color and stop animation after that

ovi1337, 2018-10-18:
I currently don't have figured out how I can stop the animation after the first iteration. 
I specially just want to fade pixels from start color to end color.
Perfectly the current color will be used as start color which can be faded to another target color. 

Is this case generally possible and if yes, how?

moose4lord, 2018-10-18:
You've bumped into a drawback of the library. All of it's effects are meant to loop indefinitely, not to be triggered once, run to completion, then wait for the next trigger. However you can accomplish what you want with a custom effect, like so:
```c++
#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN 5

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  
  ws2812fx.init();
  ws2812fx.setBrightness(255);

  // setup the custom fade effect
  uint8_t myFadeMode  = ws2812fx.setCustomMode(F("Custom Fade"), myFade);

  // setup the segment
  uint32_t colors[] = {RED, GREEN, BLUE};
  ws2812fx.setSegment(0, 0, LED_COUNT-1, myFadeMode, colors, 50, NO_OPTIONS);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();

  // some mechanism to trigger the fade
  long static then = 0;
  if(millis() > then + 5000) { // every 5 seconds
    ws2812fx.resetSegmentRuntimes(); // trigger the fade
    then = millis();
  }
}

uint16_t myFade(void) { // fade from color[0] to color[1] then stop
  WS2812FX::Segment* seg = ws2812fx.getSegment();
  WS2812FX::Segment_runtime* segRt = ws2812fx.getSegmentRuntime();

  if(segRt->counter_mode_step < 256) {
    uint32_t color = ws2812fx.color_blend(seg->colors[0], seg->colors[1], segRt->counter_mode_step);
    for(uint16_t i=seg->start; i <= seg->stop; i++) {
      ws2812fx.setPixelColor(i, color);
    }
    segRt->counter_mode_step += 4;
  }

  return seg->speed;
}
```

The custom fade effect is triggered by calling resetSegmentRuntimes(), which resets the counter_mode_step runtime variable back to zero. The custom effect uses counter_mode_step as a parameter to the color_blend() function, which calculates the proper blend between color[0] and color[1]. It's a kludge, but it works.

moose4lord, 2018-10-19:
It ran smoothly for me on an ESP8266. Are you running on an Arduino?

ovi1337, 2018-10-19:
I've found the issue of the lagging now.
I'm using a ESP8266 and i had an issue with a double assignment of the same pixels.

It working well so far, but now i have a problem by setting another color. I only get the initial color and the fading is stopped, after i'm setting something different via API endpoint:

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <ESP8266mDNS.h>
#include <WS2812FX.h>

#define LED_COUNT 8
#define LED_PIN D3

const char *PARAM_ID = "id";
const char *PARAM_FROM = "from";
const char *PARAM_TO = "to";

AsyncWebServer server(80);

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

uint16_t myFade(void) {
  WS2812FX::Segment* seg = ws2812fx.getSegment();
  WS2812FX::Segment_runtime* segRt = ws2812fx.getSegmentRuntime();

  if(segRt->counter_mode_step < 256) {
    uint32_t color = ws2812fx.color_blend(seg->colors[0], seg->colors[1], segRt->counter_mode_step);
    
    for(uint16_t i=seg->start; i <= seg->stop; i++) {
      ws2812fx.setPixelColor(i, color);
    }

    segRt->counter_mode_step += 4;
  }

  return seg->speed;
}

void setup()
{
    Serial.begin(115200);
    Serial.setDebugOutput(true);

    WiFi.hostname("esp-test");
    WiFi.begin("*****", "*****");

    Serial.print("Connecting");
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println();

    Serial.print("Connected, IP address: ");
    Serial.println(WiFi.localIP());

    MDNS.addService("http", "tcp", 80);

    uint8_t myFadeMode = ws2812fx.setCustomMode(F("Custom Fade"), myFade);

    ws2812fx.init();
    ws2812fx.setBrightness(50);
    ws2812fx.setSpeed(1000);
    ws2812fx.setColor(0x000000);
    ws2812fx.setMode(FX_MODE_STATIC);

    ws2812fx.setSegment(0, 0, LED_COUNT-1, myFadeMode, (const uint32_t[]) {RED, BLUE}, 25, NO_OPTIONS);

    ws2812fx.start();

    server.onNotFound([](AsyncWebServerRequest *request) {
        request->send(404);
    });

    server.on("/fade", HTTP_GET, [&](AsyncWebServerRequest *request) {
        if (request->hasParam(PARAM_ID) && request->hasParam(PARAM_FROM) && request->hasParam(PARAM_TO)) {
            uint8_t id = atoi(request->getParam(PARAM_ID)->value().c_str());
            uint32_t from = (uint32_t) strtol(request->getParam(PARAM_FROM)->value().c_str(), NULL, 16);
            uint32_t to = (uint32_t) strtol(request->getParam(PARAM_TO)->value().c_str(), NULL, 16);

            const uint32_t colors[] = {from, to};

            ws2812fx.setSegment(id, 0, LED_COUNT-1, myFadeMode, colors, 25, NO_OPTIONS);

            request->send(200, "text/plain", "Fade LED");
        } else {
            request->send(200, "text/plain", "missing params: id and color");
        }
    });

    server.begin();
}

void loop()
{
    ws2812fx.service();
    
    long static then = 0;
    if(millis() > then + 5000) {
        ws2812fx.resetSegmentRuntimes();
        then = millis();
    }
}
```

`http://xxx.xxx.xxx.xxx/fade?id=0&from=ff0000&to=0000ff`

moose4lord, 2018-10-21:
Did you figure out your problem? Looking over your sketch I see a scope problem with the myFadeMode variable. You're declaring it in the setup() function, but using it in the web server's server.on("/fade") callback. Unfortunately, when the setup() function ends the myFadeMode variable will go out of scope, so when the server.on() callback is called, the variable will not exist. Try declaring myFadeMode as a global variable.

ovi1337, 2018-10-23:
Hi, many thanks, now it's working well. 
But the custom effect still seems to have a problem with smoothness. Compared to FX_MODE_BREATH or FX_MODE_FADE is it very stepped, which is very visible between brightness 0% - 20%.
Is there maybe a way to implement an Option to stop the effect after the first iteration?

moose4lord, 2018-10-23:
The fade will stop automatically after it's triggered. The reason it's repeating in my example sketch is because it's constantly being retriggered every 5 seconds in the loop() function.
```c++
  // some mechanism to trigger the fade
  long static then = 0;
  if(millis() > then + 5000) { // every 5 seconds
    ws2812fx.resetSegmentRuntimes(); // trigger the fade
    then = millis();
  }
```
You'll have to come up with your own scheme to trigger the fade with a button or sensor.

ovi1337, 2018-10-23:
Oh no, that's not what i mean, i totally understand it, but the effect itself seems to be laggy.

This is what i currently use for my experiments:
You can change the current color of part 1 and 2 separately.

```cpp
uint32_t color_1_current = BLUE;
uint32_t color_1_new = BLUE;

uint32_t color_2_current = RED;
uint32_t color_2_new = RED;

server.on("/fade", HTTP_GET, [&](AsyncWebServerRequest *request) {
  if (request->hasParam(PARAM_ID) && request->hasParam(PARAM_COLOR)) {
    uint8_t id = atoi(request->getParam(PARAM_ID)->value().c_str());
    uint32_t color = (uint32_t)strtol(request->getParam(PARAM_COLOR)->value().c_str(), NULL, 16);
    
    color_1_current = color_1_new;
    if (id == 0) {
      color_1_new = color;
    }

    color_2_current = color_2_new;
    if (id == 1) {
      color_2_new = color;
    }

    const uint32_t colors_1[] = {color_1_current, color_1_new};
    const uint32_t colors_2[] = {color_2_current, color_2_new};

    ws2812fx.setSegment(0, 0, 3, myFadeMode, colors_1, 50, NO_OPTIONS);
    ws2812fx.setSegment(1, 4, 7, myFadeMode, colors_2, 50, NO_OPTIONS);

    ws2812fx.resetSegmentRuntimes();

    request->send(200, "text/plain", "Fade LED");
  } else {
    request->send(200, "text/plain", "missing params: id and color");
  }
});
```

ovi1337, 2018-10-23:
Oh, now i've found the ultimate way and i don't have to use `ws2812fx.resetSegmentRuntimes();`. It's stopping on the target color and it's automatically restarting, if the target color is different to the current color.
So i just have to use:
```cpp
ws2812fx.setSegment(0, 0, 3, myFadeMode, (const uint32_t[]) {BLACK, RED}, 1500, NO_OPTIONS);
````
to fade from black to red and if i want to get blue, i just have to set:
```cpp
ws2812fx.setSegment(0, 0, 3, myFadeMode, (const uint32_t[]) {RED, BLUE}, 1500, NO_OPTIONS);
````

Here's the well working function:
```cpp
uint16_t myFade(void) {
    WS2812FX::Segment *segment = ws2812fx.getSegment();
    WS2812FX::Segment_runtime *runtime = ws2812fx.getSegmentRuntime();

    int lum = runtime->counter_mode_step;

    if (lum > 255) {
        lum = 511 - lum;
    }

    uint32_t color = ws2812fx.color_blend(segment->colors[0], segment->colors[1], lum);

    for (uint16_t i = segment->start; i <= segment->stop; i++) {
        ws2812fx.setPixelColor(i, color);
    }

    runtime->counter_mode_step += 4;
    if (runtime->counter_mode_step > 255) {
        runtime->counter_mode_step = 255;
    } 
    else if (runtime->counter_mode_step > 255 && color != segment->colors[1]) {
        runtime->counter_mode_step = 0;
    }

    return (segment->speed / 128);
}
```

It's maybe useful for other users, we could make an example for this special use case.

Maurotb, 2019-05-30:
Hi, is possible to adapt this code to fade in/out  when  off  or off strip led for various effect??