kitesurfer1404/WS2812FX #125: Circuit guidance should probably include a resistor between ESP/Arduino and WS2812

cjcharles0, 2018-10-22:
I spent ages trying to debug why my lights would occasionally flicker, or why my ESP would suddenly crash and looked through heaps of code, tried different version of things, went to simple sketches etc etc.

It turned out the signal reflections from pin of ESP to IN on the WS2812 was causing huge amounts of noise, and it would eventually go mad. Dont really understand why, but have now done plenty of googling, and it seems this can be very common, but is rarely documented - except on some Adafruit/Neopixel pages.

Adding a resistor between ESP output pin and WS2812 input pin creates an RC network which filters the noise. A 300-510Ohm resistor works perfectly (anything in this range) and totally solved all my problems. I thought it was worth adding to your web pages, since it could save somebody else all the hours that I wasted debugging code that was fine all along!

moose4lord, 2018-10-22:
The hardware section of the WS2812FX User Guide does mention the [Adafruit Neopixel Uberguide](https://learn.adafruit.com/adafruit-neopixel-uberguide) which talks about that series resistor as well as other best practices. Sorry you had to stumble on that for so long. Mary Pickford said "_failure is not the falling down, it's the staying down_" Glad you got back up. :)

kitesurfer1404, 2018-10-22:
That's quite funny. I had the same issue just one day ago! In the past I always used the strips without any resistor. No issue so far. Not with Arduino. Not with ESPs. Not on 5V or 3.3V. This weekend I built my Christmas/Halloween decoration using 188 WS2812 for each window and an ESP. Had lots of flickering on the strip. Until I accidentally touched the data-line... 
I added a 470 Ohm resistor and it works since then.

@moose4lord can you provide the frizzing files to the image-directory of the userguid? I will add the resistor into the images then.

cjcharles0, 2018-10-22:
Don't want to create work, so happy to help aswell if needed! Can even tweak the PNGs and submit a PR if you cant be bothered with (or cant find) the fritzing files.

moose4lord, 2018-10-22:
Here are the Fritzing files.
[WS2812FX User Guide Fritzing.zip](https://github.com/kitesurfer1404/WS2812FX/files/2503072/WS2812FX.User.Guide.Fritzing.zip)

cjcharles0, 2018-10-22:
Just done a PR for the updated files, hope that comes through ok. Will email you the updated Fritzing files as I dont know how to PR those! (sorry it came through as three parts, I wasnt using a proper Git client and didnt add the changes properly to the web interface)