kitesurfer1404/WS2812FX #129: WS2812FX custom mode multiple segments

PhuocBull, 2018-10-26:
Hi, I'm trying to create a custom mode that has 10 **segments** on 160 leds, 16 leds each **segment**, what I want is to have 10 rainbow cycle/comet/wipe... on one led strip and implement that into custom mode 0 or 1 , 2 ... in **ws2812fx. cpp**. Can someone help me ? And by the way, can I add more command to **Serial Control**, like _reverse_, _pause_? Sorry if my English is bad!

moose4lord, 2018-10-27:
I'm not sure what you're asking in your first question. Creating 10 segments and have each one run the same or a  different animation shouldn't be a problem, and doesn't require a custom mode. See the ws2812fx_segments example sketch. A custom mode/effect creates a new visual effect that can be applied to a segment.

You can absolutely add more commands to the serial_control sketch. Just use the "if" statements in the process_command() function as a reference. A pause command might be implemented like this:
```c++
if(cmd.startsWith(F("p"))) {
  ws2812fx.pause(); 
  Serial.println("paused");
}
```

PhuocBull, 2018-10-28:
@moose4lord what I meant in the first question is 10 segments and have each one run the same at a custom mode, but when I choose a different mode in **serial control** it will have to run on the full 160 leds strip

I tried this in **Serial Control** - 48 leds strip, 3 segments, 16 leds each,same Rainbow Cycle mode :
`#define REDUCED_MODES // sketch is too big for Arduino w/32k flash, so invoke reduced modes
#include <WS2812FX.h>

#define LED_COUNT 48
#define LED_PIN 5

// Parameter 1 = number of pixels in strip
// Parameter 2 = Arduino pin number (most are valid)
// Parameter 3 = pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

String cmd = "";               // String to store incoming serial commands
boolean cmd_complete = false;  // whether the command string is complete

void setup() {
  Serial.begin(115200);
  ws2812fx.init();
  ws2812fx.setBrightness(30);
  ws2812fx.setSpeed(1000);
  ws2812fx.setColor(0x007BFF);
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.start();

  printModes();
  printUsage();
}

void loop() {
  ws2812fx.service();

  // On Atmega32U4 based boards (leonardo, micro) serialEvent is not called
  // automatically when data arrive on the serial RX. We need to do it ourself
  #if defined(__AVR_ATmega32U4__) || defined(ESP8266)
  serialEvent();
  #endif

  if(cmd_complete) {
    process_command();
  }
}

/*
 * Checks received command and calls corresponding functions.
 */
void process_command() {
  if(cmd == F("modecustom")) { 
    ws2812fx.setSegment(0, 0, 15,   FX_MODE_RAINBOW_CYCLE, WHITE, 10, false);
    ws2812fx.setSegment(1, 16, 31,   FX_MODE_RAINBOW_CYCLE, WHITE, 10, false);
    ws2812fx.setSegment(2, 32, 47,   FX_MODE_RAINBOW_CYCLE, WHITE, 10, false);
  }
}
......
.....
....`
Typed "modecustom" in **Serial Monitor** it runs Rainbow Cycle on all 3 segment perfectly fine
But when I switch to another mode, typed "m 12" it only apply Rainbow Cycle to this segment (first 16 leds on strip) :
`(0, 0, 15,   FX_MODE_RAINBOW_CYCLE, WHITE, 10, false)`

moose4lord, 2018-10-28:
Ah, I see. If you want to change the segment configuration when "m 12" is received, you'll need to reset the segment config, then run the setSegment() function to reconstruct the segment. Something like this:
```c++
if(cmd.startsWith(F("m "))) { 
  uint8_t m = (uint8_t) cmd.substring(2, cmd.length()).toInt();
  ws2812fx.resetSegments();
  ws2812fx.resetSegmentRuntimes();

  ws2812fx.setSegment(0,  0, LED_COUNT-1, m, WHITE, 1000, false);

  Serial.print(F("Set mode to: "));
  Serial.print(ws2812fx.getMode());
  Serial.print(" - ");
  Serial.println(ws2812fx.getModeName(ws2812fx.getMode()));
}
```

PhuocBull, 2018-10-28:
IT WORKED !!!
Thank you for your time ,sir !

closed 2018-10-28 by PhuocBull

antonkuspriyadi, 2020-11-20:
hi, I need your help. how to show 55 modes into standby mode ? 
this is my sketch

#include <WS2812FX.h>

#define LED_Data    10  // Untuk Data Led
#define Jumlah_LED 10  // Untuk Setting Jumlah Led Yang Anda Pakai
#define TIMER_MS 5000
int Signal_REM = 2;
int Sein_Kanan = 3;
int Sein_Kiri = 4;
int Sein_Kanan_Kiri = 5;

int NoLED_Sein_Kanan_Awal = 0;
int NoLED_Sein_Kanan_Akhir = 2;
int NoLED_Sein_Kiri_Awal = 7;
int NoLED_Sein_Kiri_Akhir = 9;
int NoLED_Rem_Awal = 0;
int NoLED_Rem_Akhir = 9;
int NoLED_StandBY_Awal = 0;
int NoLED_StandBY_Akhir = 9;

int Mode_REM = 3; // Pilih Mode Ganti 1-56
int Mode_SEIN = 3; // Pilih Mode Ganti 1-56
int Mode_SEIN_Kanan_Kiri = 6; // Pilih Mode Ganti 1-56
int Mode_STANDBY = 0; // Pilih Mode Ganti 1-56

int Kecepatan_REM = 500;
int Kecepatan_SEIN = 700;
int Kecepatan_STANDBY = 700;

int ON = HIGH;
int OFF = LOW;
int SignalREM = OFF;
int SignalSeinKanan = OFF;
int SignalSeinKiri = OFF;
int SignalSeinKananKiri = OFF;

int ResetREM = OFF;
int ResetSeinKanan = OFF;
int ResetSeinKiri = OFF;
int ResetSeinKananKiri = OFF;

int Kecerahan_LED = 255; // Untuk Setting Keterangan Lampu Led 0 - 255
unsigned long Warna_Dasar_REM = RED; // Setting Warna Dasar Yang Anda Inginkan Bisa Gunakan Kode HTML Misal 0xFF0000
unsigned long Warna_Dasar_SEIN = YELLOW; // Setting Warna Dasar Yang Anda Inginkan Bisa Gunakan Kode HTML Misal 0xFF0000
unsigned long Warna_Dasar_SEIN_Kanan_Kiri = YELLOW; // Setting Warna Dasar Yang Anda Inginkan Bisa Gunakan Kode HTML Misal 0xFF0000
unsigned long Warna_Dasar_STANDBY = RED; // Setting Warna Dasar Yang Anda Inginkan Bisa Gunakan Kode HTML Misal 0xFF0000
unsigned long Matikan_LED = BLACK;

WS2812FX ws2812fx = WS2812FX(Jumlah_LED, LED_Data, NEO_GRB + NEO_KHZ800);
unsigned long last_change = 0;
unsigned long now = 0;
void setup ()
{
  Serial.begin(115200);
  pinMode(Signal_REM, INPUT);
  pinMode(Sein_Kanan, INPUT);
  pinMode(Sein_Kiri, INPUT);
  pinMode(Sein_Kanan_Kiri, INPUT);

  ws2812fx.init();
  ws2812fx.setBrightness(Kecerahan_LED);
  ws2812fx.start();

}

void loop()
{
  SignalREM = digitalRead(Signal_REM);
[... 72 lines cut ...]
ws2812fx.setSegment(0, 0, 2, Mode_SEIN_Kanan_Kiri, Warna_Dasar_SEIN, Kecepatan_SEIN, true);  // segment 2 is leds 20 - 29
ws2812fx.setSegment(1, 3, 6, 28, RED, 500, false);  // segment 2 is leds 20 - 29
ws2812fx.setSegment(2, 7, 9, Mode_SEIN_Kanan_Kiri, Warna_Dasar_SEIN, Kecepatan_SEIN, false);  // segment 2 is leds 20 - 29
}

void LampuRem() {
// parameters: index, Pin Awal, Pin Akhir, Pilihan Mode, Warna Yang Di Inginkan, Kecepatan, Arah Running True / False
ws2812fx.setSegment(0, 0, 4, Mode_REM, Warna_Dasar_REM, Kecepatan_REM, true);  // segment 2 is leds 20 - 29
ws2812fx.setSegment(1, 5, 9, Mode_REM, Warna_Dasar_REM, Kecepatan_REM, false);  // segment 2 is leds 20 - 29
ws2812fx.setSegment(2, 5, 9, Mode_REM, Warna_Dasar_REM, Kecepatan_REM, false);  // segment 2 is leds 20 - 29
}

void LampuRem_JikaBersamaSeinKanan(){
// parameters: index, Pin Awal, Pin Akhir, Pilihan Mode, Warna Yang Di Inginkan, Kecepatan, Arah Running True / False
ws2812fx.setSegment(0, 0, 6, Mode_SEIN, Warna_Dasar_SEIN, Kecepatan_SEIN, true);  // segment 2 is leds 20 - 29
ws2812fx.setSegment(1, 7, 9, 14, RED, 500, true);  // segment 2 is leds 20 - 29
ws2812fx.setSegment(2, NoLED_Sein_Kiri_Awal, NoLED_Sein_Kiri_Akhir, NO_OPTIONS, Matikan_LED, Kecepatan_SEIN, true);  // segment 2 is leds 20 - 29  
}

void LampuRem_JikaBersamaSeinKiri(){
// parameters: index, Pin Awal, Pin Akhir, Pilihan Mode, Warna Yang Di Inginkan, Kecepatan, Arah Running True / False
ws2812fx.setSegment(0, 3, 9, Mode_SEIN, Warna_Dasar_SEIN, Kecepatan_SEIN, false);  // segment 2 is leds 20 - 29
ws2812fx.setSegment(1, 0, 2, 14, RED, 500, true);  // segment 2 is leds 20 - 29
ws2812fx.setSegment(2, NoLED_Sein_Kanan_Awal, NoLED_Sein_Kanan_Akhir, NO_OPTIONS, Matikan_LED, Kecepatan_SEIN, true);  // segment 2 is leds 20 - 29  
}

moose4lord, 2020-11-20:
@antonkuspriyadi did you mean to append to this old issue? This issue has been closed for quite some time. Please open a new issue with your question.

Thanks