kitesurfer1404/WS2812FX #131: Some modes don't properly work on Serial Control

PhuocBull, 2018-10-29:
I just started a new project a few days ago using **Serial Control** from **WS2812FX** examples to start with and happen to find out that some modes don't work properly on Serial Control.

1.**Fade** - Fades the LEDs on and (almost) off again. - When it fades to the maximum brightness, all led instantly turn to white, then go back to the picked color.
2.**Running Lights** - Running lights effect with smooth sine transition. - All leds freeze.
3.**Running Random** - Random colored pixels running. - Only run full strip with higher brightness.
4.**ICU** - Two eyes looking around. - All leds freeze.

All modes woks fine in **McLightning/esp8266_webinterface** using ESP8266 so I don't think there any problem with the library, maybe the **Serial Control** sketch have bugs.
I have a bunch of nano and just tested on all of them still same problems, I'm gonna go downtown tomorrow and buy some other board like pro micro,uno,mega,... just to make sure its not the hardware limit.
Lets me know if you guys find any other arduino board have the same problems with **Serial Control**.

moose4lord, 2018-10-29:
**Fade** does have a bug that only affects Arduino devices that use 16-bit integers (see [int](https://www.arduino.cc/reference/en/language/variables/data-types/int/)). It has to do with how sign extension is implemented during math operations. Devices that use 32-bit integers work fine. I do so little testing on Arduinos these days, that I forgot about it, but I'll revisit it.

The serial_control example sketch is quite big and won't fit in the standard Arduino's 32k of flash memory. So it has ```#define REDUCED_MODES``` at the top, which disables the **BREATH**, **RUNNING_LIGHTS** and **ICU** effects to conserve flash memory. (If you're running on an Arduino with more than 32k of flash memory or on an ESP8266, you can comment out the _#define_ and those effects will be available.)

PhuocBull, 2018-10-30:
Ah, so that's why, thank you, that's helped a lot !

closed 2018-10-30 by PhuocBull