kitesurfer1404/WS2812FX #141: White strip coming on randomly for ws2812fx_patterns_web.ino

chunkysteveo, 2018-12-11:
Hi, First up - great library and work here, and the Android app too! Am trying to get several ESP8266 controlled WS2811 strips working as Christmas string lights. I think my WiFi is being a little laggy as the Android app seems to show the devices (currently have three set up) as unresponsive when turning them on/off with the "play" "Upload" button.

But the main issue I am seeing is that the strips (occurring on two of them) will power up when I have "stopped them" with the android app and on full power white. They will come on randomly, and also switch off on their own. I end up unplugging them to make sure they stay off through the night.

It's odd behaviour to say the least! Not sure if you have come across it also? Going to try the Alexa sketch as my fiancee only seems to like them all on one effect anyhow, so a simple Alexa command should be easier to use than configuring all the devices through the app.  I will let you know if the random white power on occurs with this sketch, then it will no doubt be something to do with my wiring, power, data lines, etc?

Thanks,

Steve

moose4lord, 2018-12-11:
The very first strip of LEDs that I bought would power on with all LEDs at full intensity. All the others I have bought since then power on with all LEDs off. So it seems they have fixed that in later versions of the driver hardware.

Randomly turning on and off is something else. When the LEDs randomly turn on, is the animation playing, or are the LEDs just static? If there's noise being injected onto the data line, you might need a small resistor in series with DI pin. A few hundred uF capacitor between GND and VDD can also clear up flaky behavior. These tips are covered in Adafruit's Neopixel Uber Guide, which is definitely worth a read: [https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices](https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices)

chunkysteveo, 2018-12-12:
The strips are solid white on, and randomly turning solid on, and then off, whenever they wanted to.

One setting I may not have had correct was the 800Khz/400Khz option in the Adafruit library. Was set to 800, but the strips are using the 2811 driver chips and they recommend 400KHz option. Unsure on what difference that may have caused. Have never had to put the inline resistor on the data pin, but always known that was recommended. Have got a 1000uF cap on all the power lines, so that's OK.

BUT the big update now is that I have switched to the Alexa sketch and now both strips have remained off when I tell them/her to switch off, and stay off, no more random on or off. I have also set the strips to 400KHz too, so there are now two updates, 800KHz to 400KHz, and the change from the pattern sketch and app, to the Alexa sketch and Echo control. Now they are working I am not wanting to test them at 800KHz on the Alexa sketch to see which if either step fixed it... Alexa control is a lot more useful to us now than the LEDfx app, and adding them all into their own lighting groups allows us to control all the lights in one command - neat!