kitesurfer1404/WS2812FX #143: lvalue required as unary

redstar2k1, 2018-12-15:
I'm trying to compile the wsp8266_webinterface but I get this error messages:

Arduino: 1.9.0-beta (Linux), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

/home/mihai/Arduino/sketchbook/libraries/WS2812FX/examples/esp8266_webinterface/esp8266_webinterface.ino: In function 'void srv_handle_set()':
esp8266_webinterface:226:56: error: lvalue required as unary '&' operand
       uint32_t tmp = (uint32_t) strtol(&server.arg(i)[0], NULL, 16);
                                                        ^
esp8266_webinterface:233:54: error: lvalue required as unary '&' operand
       uint8_t tmp = (uint8_t) strtol(&server.arg(i)[0], NULL, 10);
                                                      ^
esp8266_webinterface:244:56: error: lvalue required as unary '&' operand
         uint8_t tmp = (uint8_t) strtol(&server.arg(i)[0], NULL, 10);
                                                        ^
exit status 1
lvalue required as unary '&' operand

I had this working ok about one year ago. Did anything changed since then in the IDE? Any idea how can I fix this?

moose4lord, 2018-12-15:
That example sketch is compiling OK on my Mac with IDE v1.8.7. Can you try that version?

Otherwise it looks like your compiler is balking at the odd casting that is being done when processing the server.arg() query parameters. You can try rewriting the statements like this:
```c++
//    uint32_t tmp = (uint32_t) strtol(&server.arg(i)[0], NULL, 16);
      uint32_t tmp = (uint32_t) strtol(server.arg(i).c_str(), NULL, 16);
```

redstar2k1, 2018-12-16:
Rewriting the statements worked. Thank's for quick fix.

closed 2018-12-16 by redstar2k1