kitesurfer1404/WS2812FX #144: ws2812fx_patterns_web:222:5: error: 'DynamicJsonBuffer' was not declared in this scope

frankvdh, 2018-12-16:
Documenting for others...

The web examples of ws2812fx need v5.x.x of the ArduinoJson library. They don't work with v6.7.0-beta (and I'm guessing with any v6.x.x). v5.13.4 does work.

Compiling ws2812fx_patterns_web with v6.7.0-beta gives
`ws2812fx_patterns_web:222:5: error: 'DynamicJsonBuffer' was not declared in this scope
`and a bunch of consequential errors.

moose4lord, 2018-12-16:
Yes, there are a few breaking changes in ArduinoJson v6.x-beta. There is a nice v5 to v6 migration guide, if you want to go that route:
[https://arduinojson.org/v6/doc/upgrade/?utm_source=github&utm_medium=release](https://arduinojson.org/v6/doc/upgrade/?utm_source=github&utm_medium=release)

Unfortunately, the Arduino IDE doesn't distinguish between stable versions and beta or release candidate versions of libraries, so sometimes it goads unsuspecting users into updating libraries when they probably shouldn't.

Thanks for bringing this up.

moose4lord, 2019-01-20:
V1.1.7 added code to allow ws2812fx_patterns_web to be compiled with ArduinoJson v6. Although, ArduinoJson v6 is still beta code and still needs some polish, so I would advise sticking with ArduinoJson v5 for now. Closing this issue.

closed 2019-01-20 by moose4lord