kitesurfer1404/WS2812FX #147: Issue with setBrightness(0) within custom effect

gpmaxx, 2018-12-23:
Hi,

I'm trying to use brightness from within a custom effect and encountering an issue when setting brightness to 0.   Once setting brightness to 0 they LEDs stay off even though the getBrightness reads as > 0. Changing the mode restores the LEDs.  This can be demonstrated in the code below. If setBrightness(1) is used then the code works as expected (blinks for 5 seconds and then goes solid) however if setBrightness(0) is used the LEDs stay off until the 5 seconds is elapsed. This is even though the calls to getBrightness() show the value alternating between 0 and 255 as expected during this time.  I can't find any obvious reason for this behaviour but understanding what is going on in the Adafruit library's setBrightness function is difficult. 

```
#include <WS2812FX.h>
#define DEFAULT_LED_COUNT 50
#define LED_PIN 14

WS2812FX ws2812fx = WS2812FX(DEFAULT_LED_COUNT, LED_PIN, NEO_RGB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setCustomMode(F("Custom Blink"),customBlink);
  ws2812fx.setSegment(0,0,ws2812fx.getLength()-1,FX_MODE_CUSTOM_0,BLUE,1000,false);
  ws2812fx.start();

}

uint32_t timer = millis();
void loop() {

    ws2812fx.service();

    if ((millis() - timer) > 5000) {
      ws2812fx.setBrightness(255);
      ws2812fx.setMode(0,FX_MODE_STATIC);
    }

    Serial.println(ws2812fx.getBrightness());

}

uint16_t customBlink() {

  WS2812FX::Segment* seg = ws2812fx.getSegment();
  WS2812FX::Segment_runtime* segrt = ws2812fx.getSegmentRuntime();
 
  for (uint16_t i = seg->start; i < seg->stop; i++) {
    ws2812fx.setPixelColor(i,BLUE);
  }

  if ((segrt->counter_mode_call & 1) == 0) {
    ws2812fx.setBrightness(255);
  }
  else {
    ws2812fx.setBrightness(0);   // doesn't work
    //ws2812fx.setBrightness(1); // works fine
  }
 
  return 100;
}

```

moose4lord, 2018-12-23:
What's happening is when brightness=0 and customBlink() is called, the setpixelColor() function will store zeros in the Adafruit library's internal pixels[] array (the LEDs will be black/off, not BLUE). Since the pixels[] array has all zeros in it, there's simply no way to recover the BLUE color information when setBrightness(255) is called. This 'destructive' brightness processing is mentioned near the getPixelColor() function in the Adafruit_Neopixel.cpp file:
```
// Stored color was decimated by setBrightness().  Returned value
// attempts to scale back to an approximation of the original 24-bit
// value used when setting the pixel color, but there will always be
// some error -- those bits are simply gone.
```

To fix, you just need to move setBrightness() before setPixelColor(). Like so:
```c++
  if ((segrt->counter_mode_call & 1) == 0) {
    ws2812fx.setBrightness(255);
  } else {
    ws2812fx.setBrightness(0);   // doesn't work
    //ws2812fx.setBrightness(1); // works fine
  }

  for (uint16_t i = seg->start; i <= seg->stop; i++) {
    ws2812fx.setPixelColor(i,BLUE);
  }
```

P.S. note there's a bug in your for loop. It should be
```
for (uint16_t i = seg->start; i <= seg->stop; i++) {
```
not
```
for (uint16_t i = seg->start; i < seg->stop; i++) {
```

gpmaxx, 2018-12-23:
thank you!   After all the fiddling I went through and of course the solution is so obvious.   What threw me off (and I still don't fully grasp is the difference in behaviour/logic flow in calling setBrightness(0) from the main execution loop. 

With respect to the for loop bounds.  Wow.  So bizzare I missed that. And switching batch and forth between activating custom and built in effects made it not at all obvious that 1 LED wasn't working. 

But please note that I was going off the example in the Custom Effects section of the [extras](https://github.com/kitesurfer1404/WS2812FX/blob/master/extras/WS2812FX%20Users%20Guide.md) page, which uses i < seg.stop in the example. 

thank you very much for the quick assistance.

moose4lord, 2018-12-23:
Ah, I see what you mean about the custom effect example in the User Guide. That is old code that definitely won't work. The whole User Guide could probably use an update, since the library has changed quite a bit since it was written. I'll try to do that over the holidays.

Thanks for mentioning that.

moose4lord, 2019-01-20:
User Guide has been updated in v1.1.7. Closing this issue.

closed 2019-01-20 by moose4lord