kitesurfer1404/WS2812FX #148: Possible enhancement: add on library to do animations across segments

sprior, 2018-12-25:
Being a geek one thing I'd like to do with my Neopixel ring Christmas tree would to have a "warp core" mode.  I currently have 5 rings and one star which I've defined as 6 segments.  I'd imagine the best way to implement warp core mode would be to define segments for each level and then have an addon animation library cycle the modes of the segments.  It'd be like having a wrapper library like WS2812FX but instead of manipulating individual Neopixels it would manipulate WS2812FX segments.

If I get time I might poke around with this.

paulkugener, 2019-09-24:
Hi, did you make any progress on this feature? I would like to use a similar functionality with my project. Let's say I have:

``` c++
ws2812fx.setSegment(0, 0, LED_COUNT/3-1, FX_MODE_CHASE_COLOR, GREEN, 1000, false);
ws2812fx.setSegment(1, LED_COUNT/3, LED_COUNT*2/3-1, FX_MODE_STATIC, RED, 1000, false);
ws2812fx.setSegment(2, LED_COUNT*2/3, LED_COUNT-1, FX_MODE_CHASE_COLOR, GREEN, 1000, false);
```
Now I would like to _combine_ segment 0 and segment 2, that there is only one chase animation using both segments. (Here the LED-strip would be in the form of a ring.)

I believe this feature would be very useful. If I get time to learn some c++, I might look into this.

sprior, 2019-09-24:
I spent a couple of nights hacking the ws2812fx code a while back to take out the segment animation code and see how the mode coordination code could be used across segments, but then got distracted by other shiny projects and haven't gotten back to it.  My thought was that this new function wouldn't be a new feature of the ws2812fx code as much as new code which would coordinate multiple ws2812fx objects.

closed 2020-04-06 by moose4lord