kitesurfer1404/WS2812FX #152: What about a code with all the effects?

OURLEDS, 2019-01-08:
Hey, after about 2 days of searching and almost 10 hours of trying different codes and projects ...
This is the one I consider the Roseta stonne for WS2812 ... but it is in an encrypted languaje.

Because I am just a amateur code programmer, I just can't find a way to the project I want in the time I need. 
It would be very very helpfull if the creator or another person that understand this library explain me or share a code where it is used all of the effects, and I just need to change the times, speed and order of sequences,.
I know how to make loops and perhaps even insert some buttons and other things, but defientely I don't know how to write the effects to appear in the ws2812 ring.
This is the only code I found and I managed to change jsut to make something simialr to 1 of the aniamtions I need, but is just a very bad use of the whole library potential.
`/*
  Demo sketch showing how to use the functions and features of the FastLED
  library to create custom effects. For this example we're going to implement
  Mark Kriegsman's Fire2012 effect from the FastLED examples folder here:
  https://github.com/FastLED/FastLED/tree/master/examples/Fire2012

  The basic idea is to use FastLED to create the LED color data, then
  copy the data to the ws2812fx instance for display.
  
  Keith Lord - 2018

  LICENSE

  The MIT License (MIT)

  Copyright (c) 2018  Keith Lord 

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sub-license, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
  THE SOFTWARE.
  
  CHANGELOG
  2018-05-13 initial version
*/
#include "FastLED.h" // be sure to install and include the FastLED lib
#include <WS2812FX.h>

#define NUM_LEDS 32
#define LED_PIN 5

WS2812FX ws2812fx = WS2812FX(NUM_LEDS, LED_PIN, NEO_GRB + NEO_KHZ800);

// declare global parameters used by Fire2012
bool gReverseDirection = false;

void setup() {
  Serial.begin(115200);

  // init WS2812FX to use a custom effect
  ws2812fx.init();
  ws2812fx.setBrightness(55);
  ws2812fx.setColor(BLUE);
  ws2812fx.setSpeed(100);
  ws2812fx.setMode(FX_MODE_CUSTOM);
  ws2812fx.setCustomMode(myCustomEffect);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}

// in the custom effect run the Fire2012 algorithm
uint16_t myCustomEffect() {
  Fire2012();

  // return the animation speed based on the ws2812fx speed setting
  return (ws2812fx.getSpeed() / NUM_LEDS);
}

/*
 * paste in the Fire2012 code with a small edit at the end which uses the
 * setPixelColor() function to copy the color data to the ws2812fx instance. 
*/

// Fire2012 by Mark Kriegsman, July 2012
// as part of "Five Elements" shown here: http://youtu.be/knWiGsmgycY
//// 
// This basic one-dimensional 'fire' simulation works roughly as follows:
// There's a underlying array of 'heat' cells, that model the temperature
// at each point along the line.  Every cycle through the simulation, 
// four steps are performed:
//  1) All cells cool down a little bit, losing heat to the air
//  2) The heat from each cell drifts 'up' and diffuses a little
//  3) Sometimes randomly new 'sparks' of heat are added at the bottom
//  4) The heat from each cell is rendered as a color into the leds array
//     The heat-to-color mapping uses a black-body radiation approximation.
//
// Temperature is in arbitrary units from 0 (cold black) to 255 (white hot).
//
// This simulation scales it self a bit depending on NUM_LEDS; it should look
// "OK" on anywhere from 20 to 100 LEDs without too much tweaking. 
//
// I recommend running this simulation at anywhere from 30-100 frames per second,
// meaning an interframe delay of about 10-35 milliseconds.
//
// Looks best on a high-density LED setup (60+ pixels/meter).
//
//
// There are two main parameters you can play with to control the look and
// feel of your fire: COOLING (used in step 1 above), and SPARKING (used
// in step 3 above).
//
// COOLING: How much does the air cool as it rises?
// Less cooling = taller flames.  More cooling = shorter flames.
// Default 50, suggested range 20-100 
#define COOLING  50

// SPARKING: What chance (out of 255) is there that a new spark will be lit?
// Higher chance = more roaring fire.  Lower chance = more flickery fire.
// Default 120, suggested range 50-200.
#define SPARKING 20

void Fire2012()
{
// Array of temperature readings at each simulation cell
  static byte heat[NUM_LEDS];

  // Step 1.  Cool down every cell a little
    for( int i = 0; i < NUM_LEDS; i++) {
      heat[i] = qsub8( heat[i],  random8(0, ((COOLING * 5) / NUM_LEDS) + 2));
    }
  
    // Step 2.  Heat from each cell drifts 'up' and diffuses a little
    for( int k= NUM_LEDS - 1; k >= 2; k--) {
      heat[k] = (heat[k - 1] + heat[k - 2] + heat[k - 2] ) / 3;
    }
    
    // Step 3.  Randomly ignite new 'sparks' of heat near the bottom
    if( random8() < SPARKING ) {
      int y = random8(7);
      heat[y] = qadd8( heat[y], random8(160,255) );
    }

    // Step 4.  Map from heat cells to LED colors
    for( int j = 0; j < NUM_LEDS; j++) {
      CRGB color = HeatColor( heat[j]);
      int pixelnumber;
      if( gReverseDirection ) {
        pixelnumber = (NUM_LEDS-1) - j;
      } else {
        pixelnumber = j;
      }

// **** modified for use with WS2812FX ****
//    leds[pixelnumber] = color;
      ws2812fx.setPixelColor(pixelnumber, color.red, color.green, color.blue);
// **** modified for use with WS2812FX ****

    }
}`

OURLEDS, 2019-01-08:
Just to let readers know what I am looking for, it is something like the min 4:10 of this video
https://www.youtube.com/watch?v=4fzjkswrgDs

moose4lord, 2019-01-08:
I'm not exactly sure what your goal is. It seems like you're interested in cycling through your own list of custom effects, which really isn't the WS2812FX libs forte. To cycle through effects, where each effect has independent parameters (i.e. duration, brightness, speed, colors, options), you can look at the ws2812fx_patterns_web sketch in the examples folder. Strip out all the web interface stuff and just hardcode the patterns array with the effects you like:
For example, replace this:
```c++
Pattern patterns[MAX_NUM_PATTERNS] = { DEFAULT_PATTERN };
int numPatterns = 1;
```
with this:
```c++
// duration, brightness, numSegments, [ { first, last, speed, mode, options, colors[] } ]
Pattern patterns[MAX_NUM_PATTERNS] = {
  {30, 64, 1, { {0, numLeds-1, 1000, FX_MODE_FADE, NO_OPTIONS, {RED,  BLACK, BLACK}} }},
  {30,128, 1, { {0, numLeds-1, 2000, FX_MODE_SCAN, NO_OPTIONS, {BLUE, BLACK, BLACK}} }},
  {30,255, 1, { {0, numLeds-1, 2000, FX_MODE_BLINK,NO_OPTIONS, {GREEN,BLACK, BLACK}} }}
};
int numPatterns = 3;
```

Creating custom effects is a different animal. The User Guide has some info about creating a custom effect and as you've found, there's a couple good example sketches dealing with custom effects. But honestly, it's not trivial and probably not something a novice programmer can just dive into. I'd stick with the built-in effects and the handful of custom effects in the src/custom folder while you're still learning to use the lib.

OURLEDS, 2019-01-08:
Thanks for the reply;
Yes probably I will end up learning everything from scratch as what I want is very simple but very specific.
In the sencond answer you can see some example videos attached, where you can see the kind of aniamtion I am looking for, but I just simple don't know how to even start, as many of them will requiere of many if, for, voids and creating paramerrter and variables that I am not even remotely used to work with.
So probably if at elast someone guide me to a palce where can I start learning the basic of Arduino code creating for LEDS? if that even exist.

kitesurfer1404, 2019-01-08:
Take a look at https://github.com/scottkildall/NeoPixel-Samples
There are some basic examples. My advice would be: try them, modify them, see what happens. Start with neopixel_basic. Try to understand what each line does. Try to modify the code, so only one half lights up. Try to modify the code, so the first half lights up red, the second half blue. Find more ideas to implement and try try try. ;-)
Move on to neopixel_3color and play with it. Move on to neopixel_moving1 and try to modify the code, so the animation runs backwards. Implement 4 segments. Implement one static segment and one moving. Implement, that the static segment grows every time the moving segment starts from the beginning... At that point you will get very near to some of your effects from the videos. ;-)
Also: create a repository where you check in your code and link your progress here. I would be happy to see you also sharing your learning experience and was to solve the problem. I promise: there is a high hurdle only at the beginning. Overcome that and you will get to the point where you create your own effects very soon.

kitesurfer1404, 2019-01-08:
Also: use the built in examples of Arduino. Read and understand the code line by line. Modify small parts, see what happens. 
--> Do small steps you can manage. Break down your ideas into small code segments that can be understood, tested and used again later.

OURLEDS, 2019-01-08:
Look my main issues are the kind like this:

I have this simple code running:
https://gist.github.com/OURLEDS/22f613b3b8374fc32e7d05d855012453
Which awesome, but I can't control nothing

So I just went to all the file of this library and I found codes like this:

https://gist.github.com/OURLEDS/cb9b674359e3f55785855d94756cc689
It is what I wanted, a effect with options to customize ... but how I add it to a simple new file code from scratch?

Thanks

moose4lord, 2019-01-10:
Look at the ws2812fx_custom_effect2 sketch in the examples folder. It shows how to use the custom effects found in the src/custom folder.

OURLEDS, 2019-01-15:
> 
> 
> Look at the ws2812fx_custom_effect2 sketch in the examples folder. It shows how to use the custom effects found in the src/custom folder.

It doesn't explain, each effect has a different way of code writing it with the unit 8, 16 or 32...
it requieres intermediater advanced code knoweldege I asm afraid

moose4lord, 2019-01-15:
You're right, the effects code can be quite difficult. @kitesurfer1404 mentions some nice tutorials (NeoPixel-Samples, see above) for novice software developers that want to work with NeoPixels. You should start there.

closed 2019-09-11 by moose4lord