kitesurfer1404/WS2812FX #153: Twinkle fade

GaseGasovic, 2019-01-15:
Hi guys. Thanks for amazing work on this masterpiece. I have one question. I have tried to make a Twinkle_Fade effect on WS2812b strip. It runs like a charm, but i noticed that pixels fades in a bit "choppy" way. It seems to fade lets say in 20 or 30 (out of 255) increments each time instead of 1 at the time for smooth fade. I tried to look at WS2812FX.h and WS2812FX.cpp files, but for my lack of coding knowledge, i did not find something to change. Can you help me with this? Thanks in advance. :)

moose4lord, 2019-01-15:
Can you show us your sketch?

GaseGasovic, 2019-01-15:
My sketch is one of the examples provided with library:

#include <WS2812FX.h>

#define LED_PIN    5  // digital pin used to drive the LED strip
#define LED_COUNT 10  // number of LEDs on the strip

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setBrightness(255);

  // parameters: index, start, stop, mode, color, speed, reverse
  ws2812fx.setSegment(0,  0,  9, FX_MODE_TWINKLE_FADE, #ffffff, 1000, false); // segment 0 is leds 0 - 9

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}

moose4lord, 2019-01-16:
Your setSegment() statement uses the default fade behavior, which simply divides the color intensity by 2 for each fade "step". So you're right, with brightness set to 255, you get only 8 fade steps.

But you can use the options parameter to set a different fade rate and produce a smoother animation.
```
ws2812fx.setSegment(0,  0,  29, FX_MODE_TWINKLE_FADE, 0xffffff, 1000, FADE_SLOW);
```
(note there was a typo in the code you pasted above, the color is specified as '0xffffff', not '#ffffff')
You can read more about the options parameter in the Users Guide that's in the WS2812FX lib's extras folder.

GaseGasovic, 2019-01-16:
I have tested it. Its working!!! Thank you for your support and thank you for keeping this comunity alive and updated. :)

closed 2019-01-16 by GaseGasovic