kitesurfer1404/WS2812FX #154: Add WS2812FX ArduBadge (Installation Guide)

gilmaimon, 2019-01-18:
Hi!
I saw you got a "Download  and Install" section in your readme. I think a nice addition (mainly for newer users) will be to use an ArduBadge "Badge" aswell. It looks like that:
[image]

It shows:
1. That WS2812FX is in the "Library Manager" in the Arduino IDE
2. The latest release version
3. Clicking on it redirects to a custom page with explanation on how to install WS2812FX.
What do you think?

You can check out ardu-badge here: www.ardu-badge.com

Tnx.

kitesurfer1404, 2019-01-19:
Nice idea. I added it to the readme. But ArduBadge does not support https. So there will be an unencrypted connection to the server. I'll add it again as soon as they support https.

gilmaimon, 2019-01-19:
> Nice idea. I added it to the readme. But ArduBadge does not support https. So there will be an unencrypted connection to the server. I'll add it again as soon as they support https.

Hi, Tnx!
I will notify you when it does :)