kitesurfer1404/WS2812FX #157: Doesn't work on Teensy 3.6 for me

adrianhinge, 2019-02-01:
`#include <Adafruit_NeoPixel.h>
#include <FastLED.h>
#include <WS2812FX.h>

#define LED_COUNT 144
#define LED_PIN 6

#define TIMER_MS 5000

// Parameter 1 = number of pixels in strip
// Parameter 2 = Arduino pin number (most are valid)
// Parameter 3 = pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_RGBW + NEO_KHZ800);

unsigned long last_change = 0;
unsigned long now = 0;

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setSpeed(1000);
  ws2812fx.setColor(0x007BFF);
  ws2812fx.setMode(FX_MODE_BLINK);
  ws2812fx.start();
}

void loop() {
  now = millis();

  ws2812fx.service();

  if(now - last_change > TIMER_MS) {
    ws2812fx.setMode((ws2812fx.getMode() + 1) % ws2812fx.getModeCount());
    last_change = now;
  }
}`

It the problem that I use a Teensy 3.6 (with an OctoWS2811)?

moose4lord, 2019-02-01:
I don't think the Adafruit_NeoPixel library is compatible with the OctoWS2811 adaptor. You should probably be using the OctoWS2811 library.

adrianhinge, 2019-02-02:
The NeoPixel library works fine :)

closed 2020-04-06 by moose4lord