kitesurfer1404/WS2812FX #158: Curl to invoke effect

sommo, 2019-02-12:
Hi,
Amazing project! I flashed a wemos and all works!
I need to trigger it via cli command to integrate it in other project (homebridge).
How could it be the command?
Is there a guide to looking for?
Thank you!

kitesurfer1404, 2019-02-12:
Hi there,
that's very easy. Take a look at
https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/esp8266_webinterface/main.js.cpp#L23
and 
https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/esp8266_webinterface/esp8266_webinterface.ino#L223

You have to call http://<IP-OF-ESP>/set?PARAMETER=VALUE
with PARAMETER being in c for color, m for more, b for brightness, s for speed, a for autocycle.
c takes hex encoded RGB-value, m takes integer for mode-number, b and s take - for reducing or something else for increasing brightness/speed. a takes - for disabling or anything else for enabling autocycle.

For example: 
http://192.168.0.23/set?m=15
http://192.168.0.23/set?s=-
http://192.168.0.23/set?a=x

sommo, 2019-02-13:
Thank you @kitesurfer1404

sommo, 2019-02-13:
Thank you! I will try!

Inviato da iPhone

EherXtrem, 2019-03-17:
Hello from the baltic sea. I like to send one command (or 5 commands in Line) to 5 different Wemos D1 Mini to switch all together (or in very short time) to the same modus / color. But it doesn't work (just the first 2 or sometimes 3 commands are working). Is there a kind of timing-problem? 

Details:
1. Wemos Nr. 1 works as AP with IP 192.168.4.1
2. Handy as switch with 192.168.4.2 <- sends get (or curls) following commands (for example to set all Wemos to RED and Multistrobe):

http://192.168.4.3/set?m=28 
http://192.168.4.4/set?m=28 
http://192.168.4.5/set?m=28 
http://192.168.4.6/set?m=28 
http://192.168.4.7/set?m=28 
http:// 192.168.4.3/set?c=0xff0000
http:// 192.168.4.4/set?c=0xff0000
http:// 192.168.4.5/set?c=0xff0000
http:// 192.168.4.6/set?c=0xff0000
http:// 192.168.4.7/set?c=0xff0000

3. Wemos Nr. 2 works with IP 192.168.4.3 (WS2812FX)
4. Wemos Nr. 3 works with IP 192.168.4.4 (WS2812FX)
5. Wemos Nr. 4 works with IP 192.168.4.5 (WS2812FX)
6. Wemos Nr. 5 works with IP 192.168.4.6 (WS2812FX)
7. Wemos Nr. 6 works with IP 192.168.4.7 (WS2812FX)

(max_connection @ WiFi.softAP increased to 8)

moose4lord, 2019-03-17:
Are you using Android ESP8266 Core v2.5.0? If so, I would recommend using v2.4.2 instead. V2.5.0 has network issues with some ESP8266s (https://github.com/esp8266/Arduino/issues/5736).
Also, can you show us the sketch running on Wemos Nr. 1?

EherXtrem, 2019-03-17:
The code isn't mine: [https://pastebin.com/embed_js/zvcRveru](url)

Just testing: I wrote a little batch-file on WindowsPC to switch over Wlan (not the wemos softAP above but this time my fritzbox-WLan) from one mode to the next (mode 0 to mode 55) on a single wemos (another one for testing 192.168.9.162):

2019-03-17 15:29:06 (49,3 B/s) - 'set@c=FF0000'
2019-03-17 15:29:06--  http://192.168.9.162/set?m=1
...
...
2019-03-17 15:29:10--  http://192.168.9.162/set?m=55

Switch all modes in 4 seconds without problems. The problem seems to be the softAP-Wlan @ Wemos Nr. 1.

moose4lord, 2019-03-17:
So you're using a Windows PC to send the HTTP requests to the Wemos devices? Wemos Nr. 1 is just used as the AP for your network?

EherXtrem, 2019-03-17:
Just a test-setup using a Windows PC to send the HTTP requests to the Wemos devices (over Fritzbox-Wlan @ Home). Usually Wemos Nr. 1 is used as the AP for your network (with the other wemos)!

Sorry for the difficult situation.

moose4lord, 2019-03-17:
I setup a similar configuration and I could not reproduce your issue. I have two NodeMCU's, the first setup with the SoftAP sketch, and the second runs the esp8266_webinterface sketch. I then used a Windows 10 PC to connect to the SoftAP network and run a Powershell script to send HTTP requests to the second NodeMCU. Everything worked fine.

Here is my SoftAP sketch:
```c++
#include <ESP8266WiFi.h>
void setup() {
  Serial.begin(115200);
  delay(500);
  Serial.println();

  Serial.print("Setting soft-AP ... ");
  Serial.println(WiFi.softAP("raspi", "12345678", 1, 0, 8 ) ? "Ready" : "Failed!");

  Serial.print("Soft-AP IP address = ");
  Serial.println(WiFi.softAPIP());
}
 
void loop() {
  Serial.printf("Stations connected to soft-AP = %d\n", WiFi.softAPgetStationNum());
  delay(10000);
}
```
and my Windows Powershell script:
```
# Set-ExecutionPolicy -Scope CurrentUser -ExecutionPolicy Unrestricted
Invoke-WebRequest http://192.168.4.2/set?c=0x0000ff
Invoke-WebRequest http://192.168.4.2/set?m=0
Invoke-WebRequest http://192.168.4.2/set?m=1
Invoke-WebRequest http://192.168.4.2/set?m=2
Invoke-WebRequest http://192.168.4.2/set?m=3
Invoke-WebRequest http://192.168.4.2/set?m=4
Invoke-WebRequest http://192.168.4.2/set?m=5
Invoke-WebRequest http://192.168.4.2/set?m=6
Invoke-WebRequest http://192.168.4.2/set?m=7
Invoke-WebRequest http://192.168.4.2/set?m=8
Invoke-WebRequest http://192.168.4.2/set?m=9
Invoke-WebRequest http://192.168.4.2/set?m=10
Invoke-WebRequest http://192.168.4.2/set?m=11
Invoke-WebRequest http://192.168.4.2/set?m=12
Invoke-WebRequest http://192.168.4.2/set?m=13
Invoke-WebRequest http://192.168.4.2/set?m=14
Invoke-WebRequest http://192.168.4.2/set?m=15
Invoke-WebRequest http://192.168.4.2/set?m=16
Invoke-WebRequest http://192.168.4.2/set?m=17
Invoke-WebRequest http://192.168.4.2/set?m=18
Invoke-WebRequest http://192.168.4.2/set?m=19
Invoke-WebRequest http://192.168.4.2/set?m=20
Invoke-WebRequest http://192.168.4.2/set?m=21
Invoke-WebRequest http://192.168.4.2/set?m=22
Invoke-WebRequest http://192.168.4.2/set?m=23
Invoke-WebRequest http://192.168.4.2/set?m=24
Invoke-WebRequest http://192.168.4.2/set?m=25
Invoke-WebRequest http://192.168.4.2/set?m=26
Invoke-WebRequest http://192.168.4.2/set?m=27
Invoke-WebRequest http://192.168.4.2/set?m=28
Invoke-WebRequest http://192.168.4.2/set?m=29
Invoke-WebRequest http://192.168.4.2/set?m=30
Invoke-WebRequest http://192.168.4.2/set?m=31
Invoke-WebRequest http://192.168.4.2/set?m=32
Invoke-WebRequest http://192.168.4.2/set?m=33
Invoke-WebRequest http://192.168.4.2/set?m=34
Invoke-WebRequest http://192.168.4.2/set?m=35
Invoke-WebRequest http://192.168.4.2/set?m=36
Invoke-WebRequest http://192.168.4.2/set?m=37
Invoke-WebRequest http://192.168.4.2/set?m=38
Invoke-WebRequest http://192.168.4.2/set?m=39
Invoke-WebRequest http://192.168.4.2/set?m=40
Invoke-WebRequest http://192.168.4.2/set?m=41
Invoke-WebRequest http://192.168.4.2/set?m=42
Invoke-WebRequest http://192.168.4.2/set?m=43
Invoke-WebRequest http://192.168.4.2/set?m=44
Invoke-WebRequest http://192.168.4.2/set?m=45
Invoke-WebRequest http://192.168.4.2/set?m=46
Invoke-WebRequest http://192.168.4.2/set?m=47
Invoke-WebRequest http://192.168.4.2/set?m=48
Invoke-WebRequest http://192.168.4.2/set?m=49
Invoke-WebRequest http://192.168.4.2/set?m=50
Invoke-WebRequest http://192.168.4.2/set?m=51
Invoke-WebRequest http://192.168.4.2/set?m=52
Invoke-WebRequest http://192.168.4.2/set?m=53
Invoke-WebRequest http://192.168.4.2/set?m=54
Invoke-WebRequest http://192.168.4.2/set?m=55
```
Sorry, I don't think I can be much help. You didn't mention if you were using v2.4.2 or v2.5.0 of the Arduino ESP8266 Core.

EherXtrem, 2019-03-17:
First: thank you very much for you support. But in this setup which you configurred I have no issue too. With my PC the setup works fine. My issue starts when I use my Handy instead of the PC. I want to setup 5 Wemos from the Handy at the same time. In this setup I can't start reliable the wemos Nr 2, Nr 3, Nr 4, Nr 5 or Nr 6 in one cmd-line-command. If I use many commands one after another it seems to work. Probably a  kind of timing-problem inside the WLan.

moose4lord, 2019-03-17:
Oh sorry, I misunderstood. What's a Handy?

kitesurfer1404, 2019-03-18:
Mobilephone or Smartphone.

EherXtrem, 2019-04-09:
Pardon @ break off a conversation. Some things @ real life needed attention. 

First: WS2018FX is amazing and working. My issues are not the software itself but the hardware I use. The Handy is a Samsung Galaxy S6. I use some app for automatisation like macrodroid to take the http get function on a widget-button. Macrodroid works with an kind of time lag by execute commands. Much better performence goes with tasker app. No time lag but no execute one command for all Wemos at the same time too. Is there any possibilitiy to syncron one command to differend hardware (wemos d1 mini)?

PS: If the project comes to an end and the presentation @ contest ist done I will show some pictures iff wanted :)

closed 2019-09-11 by moose4lord