kitesurfer1404/WS2812FX #159: Why fall off WI-FI in AP mode with WS2812FX / NeoPixel?

smhmousavi, 2019-02-13:
Hi and Thank you for the ws2812fx library.
There is a direct dependence on the number of LEDs in the WS2812FX() class and the stability of WI-Fi in the access point mode. (in client mode, stability is much higher.).
Why ?? :(
Note: the start page address is 192.168.4.1

1. no LED strip is needed to check the sketch
2. before uploading the sketch, first upload the data (from the data folder)
3. I am using an older ws2812fx slightly modified (attached)
4. to check free memory-192.168.4.1/mem
Problem - Wi-Fi suddenly falls off in access point mode:
1. when loading web pages
2. when you turn on the buttons on the start page
3. the problem is stronger/faster manifested, the higher the LED_COUNT value for the WS2812FX class()
   LED_COUNT = 600 - web pages can not be loaded at all. Wi-Fi falls off right away
   LED_COUNT = 100 - it may seem that everything is fine, but after a while (the number of requests, connections), Wi-Fi also falls off.

smhmousavi, 2019-02-17:
I need help for problem

kitesurfer1404, 2019-02-17:
I have running several ESPs in AP mode and never encountered something similar. Maybe someone else reads this and can help. Sounds like something memory related.

smhmousavi, 2019-02-17:
Which esp8266 model did you use?

kitesurfer1404, 2019-02-17:
Mostly ESP-01 and NodeMCU dev-boards.

smhmousavi, 2019-02-17:
I tested the esp-01 and esp-07 board

mindforger, 2019-02-17:
my closest guess after searching through the code for a bit, the used show methode from adafruit for the esp ist bitbanging the data in a continuos loop without yielding to the network stack (understandably this would ruin the function of setting the LEDs, so no real way around it)

WS2812 LEDs have an protocol which can not be replicated using any common peripherals (afaik) so bitbanging is the only option here, maybe someone optimizied the adafruit lib to use DMA or something to take load off the main loop to improve performance

smhmousavi, 2019-02-17:
Why it works in client mode ?

marissv, 2019-02-17:
I can confirm there is long term wifi connection stability issue. 
I use sample sketch segments with web interface. After some time (5min...30min.. 2 h) it losts wifi connection, cpu is runing, wdt reset doesnt happen. 
I haven't found a direct association with used led count, NeoPixel lib used, esp8266 core lib used, even tried diferent wifi AP models.
Before Xmas I programmed one wemos d1, it was more stable, worked 1..2 days online as I remember. But I dont know lib versions used.

mindforger, 2019-02-17:
> 
> 
> Why it works in client mode ?

less work to do in the stack i guess, in AP mode you are the network host

moose4lord, 2019-02-19:
@smhmousavi, can you attach your code?

smhmousavi, 2019-02-19:
Hi  @moos4lord
Please reaplace code the following function with the function in the "esp8266_webinterface.ino"example WS2812FX library and 
LED_COUNT = 600 

void wifi_setup() {

  WiFi.softAP("myAP", "12345678");

  IPAddress myIP = WiFi.softAPIP();

  Serial.print("myAP IP address: ");

  Serial.println(myIP);

  Serial.println();

}

Biketrial981, 2019-03-07:
Hi, I have the same problem.
I would be very interested in solving it since it is perfect for my project, but it depends on the changes of the web, there are times when the wifi disappears in 1-2 min.

I use DMS mini and WS2812b 5V

The CPU works since the effects continue, but the Wi-Fi does not appear again until the power is restarted

moose4lord, 2019-03-07:
Which version of the ESP8266 Arduino Core are you using? The latest version (v2.5.0) has a known problem with the WiFi firmware (see https://github.com/esp8266/Arduino/issues/5736). You should be using v2.4.2 or download/install the latest version from GitHub, which works around the issue by reverting back to an older SDK (see https://github.com/esp8266/Arduino/issues/5784).

pradeepshankara, 2019-06-27:
> Hi, I have the same problem.
> I would be very interested in solving it since it is perfect for my project, but it depends on the changes of the web, there are times when the wifi disappears in 1-2 [...]

Even I am facing the same issue, any solution here?
I am using ESP8266 12F module.

closed 2020-04-06 by moose4lord

rahulbaruri, 2022-07-31:
> Hi @moos4lord Please reaplace code the following function with the function in the "esp8266_webinterface.ino"example WS2812FX library and edit LED_COUNT = 600
> 
> void wifi_setup() {
> 
>   WiFi.softAP("myAP", "12345678");
> 
>   IPAddress myIP = WiFi.softAPIP();
> 
>   Serial.print("myAP IP address: ");
> 
>   Serial.println(myIP);
> 
>   Serial.println();
> 
> }
> 
> and remove in loop code
> 
> if(now - last_wifi_check_time > WIFI_TIMEOUT) { Serial.print("Checking WiFi... "); if(WiFi.status() != WL_CONNECTED) { Serial.println("WiFi connection lost. Reconnecting..."); wifi_setup(); } else { Serial.println("OK"); } last_wifi_check_time = now; }

After 5 second it is switched off