kitesurfer1404/WS2812FX #160: Wrong colors

kylix34, 2019-02-17:
I have a WS2812B LED strip and it's not displaying the correct colors.
I tried all the possible combinations: NEO_RGB, NEO_RBG, NEO_GRB, NEO_GBR,NEO_BRG, NEO_BGR but the colors are still inverted....

moose4lord, 2019-02-19:
Does it still show the wrong colors with a simple sketch like this:
```c++
#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN D1

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(100);
  ws2812fx.setSpeed(1000);
  ws2812fx.setColor(RED); // or GREEN or BLUE
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}
```

kylix34, 2019-02-20:
Your simple sketch worked. After that I loaded again the full sketch and the colors are loading almost OK. It's the color picker on the webpage that it's not very accurate. Thanks!

kitesurfer1404, 2019-02-20:
Do all recent mobile phone support the HTML-color-input-tag?
`<input name="fx_color" type="color" value="#ff0000">`
Then we can maybe upgrade the code to use the systems colorpicker which might be more reliable.

kylix34, 2019-02-20:
The color palette template used here: https://github.com/jasoncoon/esp8266-fastled-webserver/blob/master/data/index.htm, works on my phone (that project uses the FastLED library)

debsahu, 2019-02-25:
I use [iro.js](https://iro.js.org) works flawlessly on Android, iOS and chrome on PC.

closed 2019-09-11 by moose4lord