kitesurfer1404/WS2812FX #161: some effects not working (twinkle)

sumantakec, 2019-02-22:
HI, Kitesurfer

thanks for this amazing library and hard work. 
I thought it's my responsibility to inform users about the issue I faced and the temporary solution I found.

I am using Chinese clone of Arduino and Chinese WS2812b LED strip. After loading auto cycle example found some effects not working and only blinking first led.
starting from 19 to 24 and 45,46.
Rest effects are working fine.
so I tried putting capacitor and changed resistor value but problem remains same.

after bit more debugging, I found random16 function on WS2812FX.cpp  is not working and always giving output "0".

I am not yet sure why but changing random16 to random8 solve the issue.
on line number 880  
  setPixelColor(SEGMENT.start + random16(SEGMENT_LENGTH), color1);

changed to 

  setPixelColor(SEGMENT.start + random8(SEGMENT_LENGTH), color1);

now twinkle effect working fine

moose4lord, 2019-02-22:
I tried the twinkle effect and it's working ok for me. What microprocessor are you using? It sounds like your processor isn't doing 16-bit math the same as an ESP8266.

sumantakec, 2019-02-22:
Dear Sir,  thanks for your quick response, I am using Arduino Uno chinese clone.On IC written ATMEL atmega328p.

sumantakec, 2019-02-22:
also found 
" return twinkle_fade(color_wheel(random16())); " on line 1002 is working also i didnt change Line 484, 682,

moose4lord, 2019-02-22:
Ah, I see. You are correct. I dusted off my old Arduino Leonardo and it too has problems with twinkle and other effects that use random16(). The problem stems from this line in the random16() function:
```r = (r * lim) >> 16;```
On an Arduino microprocessor the operation is performed with 16-bit arithmetic, so the 16-bit shift-right operation causes r to be set to zero all the time. :(

This bug can be fixed by explicitly casting the variables in the two random16() functions (near line 478):
```c++
uint16_t WS2812FX::random16() {
    return (uint16_t)random8() * 256 + random8(); // add the cast to (uint16_t)
}

// note random16(lim) generates numbers in the range 0 to (lim - 1)
uint16_t WS2812FX::random16(uint16_t lim) {
    uint16_t r = random16();
    r = ((uint32_t)r * lim) >> 16; // add the cast to (uint32_t)
    return r;
}
```
I'll make these changes to the WS2812FX library. Note, your workaround to change random16() to random8() also fixes the issue, as long as your strip has less then 256 LEDs.

Sorry about the bug. I don't normally do much testing with Arduino processors, so it slipped through the cracks.

sumantakec, 2019-02-22:
thanks a lot sir,  I will try that, also I will try it on esp nodemcu and will let you know.

sumantakec, 2019-02-23:
I tested the original code without changing anything on ESP nodemcu and all effects working fine. thanks again.

closed 2019-02-23 by sumantakec