kitesurfer1404/WS2812FX #162: Max segment size?

CraigASanford, 2019-02-22:
Hello,
        Sorry for posting without testing first. I was just wondering what is the reasoning for maxing the number of LEDs in a segment to 10? This library seems perfect for what I'm trying to do but I have 2 segments that are 20+ LEDs long and 2 shorter ones. Just wondering what Issues I'll run int.

Thanks

moose4lord, 2019-02-22:
I think you misunderstood. A single LED strip can be subdivided into 10 segments (maximum). Each segment can have up to 32767 LEDs (but you would need a power supply as big as a house :)

CraigASanford, 2019-02-22:
Ahhh. ok. I assumed that was the case only because the examples also only had 10 LEDs per segment! Thanks!

closed 2019-02-22 by CraigASanford