kitesurfer1404/WS2812FX #164: Save last state & Sound to light

PhuocBull, 2019-02-26:
Hi, I'm working on a Serial Control project using mega 2560 pro, I wonder if I can somehow implement Save Last State (mode, speed, brightness or even segments) to Serial Control like Mclightning but on a mega 2560 pro? And btw, External Trigger (sound to light) - Is it possible to add this to Serial Control as a stand alone mode?

moose4lord, 2019-02-28:
I think what you're asking is to save the WS2812FX settings to nonvolatile eeprom, so they can be restored automatically after a power loss. The ws2812fx_segment_web sketch in the examples folder has the code to do that. Look for the saveToEEPROM() and restoreFromEEPROM() functions.

To add a trigger mode to the serial_control sketch I think you'd just need to add a global triggerEnable variable:
```
boolean triggerEnable = false;
```
Add the command logic:
```c++
  if (strcmp(cmd,"t+") == 0) {
    triggerEnable = true;
    Serial.println(F("Enable trigger mode"));
  }
  if (strcmp(cmd,"t-") == 0) {
    triggerEnable = false;
    Serial.println(F("Disable trigger mode"));
  }
```
And add the code from the external_trigger example sketch to your sketch's loop() function:
```c++
if(triggerEnable) {
// trigger on a regular basis
  if(now - last_trigger > TIMER_MS) {
    ws2812fx.trigger();
    last_trigger = now;
  }

  // trigger, if analog value is above threshold
  // this comes in handy, when using a microphone on analog input
  if(analogRead(ANALOG_PIN) > ANALOG_THRESHOLD) {
    ws2812fx.trigger();
  }
}
```
Give it a shot.

PhuocBull, 2019-03-01:
Thank you so much, I'll give it a try

closed 2019-03-01 by PhuocBull

PhuocBull, 2019-03-05:
@moose4lord One more question, I saw some of the custom modes in scr folder,how do I add them to serial control too ?

moose4lord, 2019-03-05:
Take a look at the ws2812fx_custom_effect2 example sketch. You'll just need to #include the custom effect file and run the setCustomMode() function to add the effect to the list of effects. There are 4 effect "slots" available, so you can run setCustomMode() four times to add up to four custom effects to your sketch. The serial_control code itself shouldn't need any modification.