kitesurfer1404/WS2812FX #165: 2 or 3 serveral Output Pins?

kollimann, 2019-03-03:
can i work with more than 1 Output Pin?

kitesurfer1404, 2019-03-03:
For the same FXs on different strips with one WS2812FX object? --> not quite sure how to mirror the output the best way. Might be possible. Also, it might be possible to drive two (several?) strips from one pin without overloading the pin.
Or for several strips with different FXs on different pins? --> just initialize more WS2812FX objects on different pins.

kollimann, 2019-03-03:
thanks for answer....
----->just initialize more WS2812FX objects on different pins.

 just initialize more WS2812FX objects on different pins.
WS2812FX ws2812fx = WS2812FX(50, 1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx = WS2812FX(30,2, NEO_GRB + NEO_KHZ800);

is not working.......

I have this programmed in FastLED and this runs perfect
  //FastLED.addLeds<LED_TYPE, DATA_PIN, COLOR_ORDER>(leds, NUM_LEDS);         
  FastLED.addLeds<WS2812, 1, GRB>(leds, 0, 51); //51
  FastLED.addLeds<WS2812, 2, GRB>(leds, 52, 100); 

so i can drive WS2812 on 5 Pins/Ports on a ESP.....

thanks Heiko

moose4lord, 2019-03-04:
You need to run init(), start() and service() on all the ws2812fx instances you create.

This sketch works for me:
```c++
#include <WS2812FX.h>

#define LED_COUNT 20  // number of LEDs on the strip

WS2812FX ws2812fx1 = WS2812FX(LED_COUNT, D1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx2 = WS2812FX(LED_COUNT, D2, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  delay(500);

  ws2812fx1.init();
  ws2812fx2.init();

  // parameters: index, start, stop, mode, color, speed, reverse
  ws2812fx1.setSegment(0,  0, LED_COUNT-1, FX_MODE_BLINK, RED,   1000, NO_OPTIONS);
  ws2812fx2.setSegment(0,  0, LED_COUNT-1, FX_MODE_BLINK, GREEN, 1000, NO_OPTIONS);

  ws2812fx1.start();
  ws2812fx2.start();
}

void loop() {
  ws2812fx1.service();
  ws2812fx2.service();
}
```

EherXtrem, 2019-05-04:
Is there a solution for the WS2812FX original code to drive same FXs on different strips with one WS2812FX object on more than 1 Output Pin ?

moose4lord, 2019-05-04:
Not really. The Adafruit_NeoPixel lib assigns a GPIO to each object you initialize. As @kitesurfer1404 mentions you can probably drive more then one WS2812 strip from a single GPIO pin. If not, you'll have to add some kind of buffer circuit to fan out the GPIO. A 74HC125 should do the trick.

kitesurfer1404, 2019-05-07:
Read this for driving 20000 WS2812FX ;-)
https://hackaday.com/2019/05/07/lots-of-blinky-esp32-drives-20000-ws2812-leds

closed 2019-09-11 by moose4lord