kitesurfer1404/WS2812FX #171: Brightness per segment

somebuddy87, 2019-03-12:
Hi,

Is it possible to add the segment_brightness as parameter of the setSegment function?

Thank you

moose4lord, 2019-03-13:
Sadly, no. Brightness is implemented in the Adafruit_NeoPixel library, upon which WS2812FX is built, and affects the brightness of the entire strip of LEDs.

I suppose you could use the setColor() and setColors() functions to dynamically adjust a segment's colors and so their brightness.
```c++
setColor(uint8_t seg, uint32_t c);
setColors(uint8_t seg, uint32_t* c);
```
Other "hardcoded" effect colors, like rainbow effects, cannot be adjusted per segment.

dmak, 2019-07-30:
Would it be possible to add support of several LED strips to the library (each strip is connected to it's own data pin)? Than the brightness of each strip can be controlled individually.

moose4lord, 2019-07-30:
The lib already supports multiple strips on different GPIO pins. All strips are programmed independently. See #165

closed 2019-09-11 by moose4lord