kitesurfer1404/WS2812FX #177: LEDs flashing when using WiFi

JulianS-Uni, 2019-04-15:
I am having issues running the WS2812FX with WiFi on a ESP32 (Doit v1 clone). When WiFi is enabled and connected during LED animations the LEDs tent to flash in random intervals, position and color. Check my code below. I reduced the brightness because the flashing gets more noticable then because its very bright. I have choosed `FX_MODE_BREATH` because its mono color and the flashing tents to be in random colors. Please let me know if you had or have similar expirences. 

``` c++
#include <WiFi.h>
#include <WS2812FX.h>

WS2812FX ws2812fx = WS2812FX(3, 27, NEO_GRB + NEO_KHZ800);

void connectToWiFi(){
	Serial.println("Connecting to WiFi:");
	WiFi.mode(WIFI_STA);
	WiFi.begin("SSID", "PW");
	//  WiFi.mode(WIFI_STA);
	while (WiFi.status() != WL_CONNECTED){
		delay(100);
		Serial.print(".");
	}
	Serial.println();
	Serial.println("Successfully connected to WiFi!");	
}

void setupWS2812(){
	ws2812fx.init();
	ws2812fx.setBrightness(5);
	ws2812fx.setSpeed(10);
	ws2812fx.setMode(FX_MODE_BREATH);
	ws2812fx.start();
}

void setup() {
	connectToWiFi();
	setupWS2812();
}

void loop(){
	ws2812fx.service();
}
```

kitesurfer1404, 2019-04-15:
Do you have a resistor in the data-line as mentioned here https://github.com/kitesurfer1404/WS2812FX/blob/master/extras/WS2812FX%20Users%20Guide.md and in the Adafruit WS2812 Uberguide?

JulianS-Uni, 2019-04-17:
Now i have a 470 ohm resistor and a 1000uF capacitor. Still flashing. Maybe i try different LEDs next.

moose4lord, 2019-04-17:
I see the same flashing with my ESP32 setup. It used to work, but sadly, I think espressif has changed the way the ESP32 handles interrupts and now WiFi is interfering with the timing of the signal driving the LEDs.

As an alternative to the timing sensitive "bit-bang" technique, the ESP32's [RMT](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/rmt.html) hardware can be used to drive the LEDs. I've attached a demo sketch that demonstrates this technique. (My demo is a variant of the **example/ws2812fx_esp32** sketch that's in the WS2812FX GitHub repository.)

[ws2812fx_ESP32.zip](https://github.com/kitesurfer1404/WS2812FX/files/3091565/ws2812fx_ESP32.zip)

JulianS-Uni, 2019-04-18:
Thank you for the confirmation and workaround. So does this mean the Adafruit NeoPixel lib wont work anymore with esp32 and wifi at all? So its probably worth greating a issue at adafruit neopixel?

moose4lord, 2019-04-18:
The issue with ESP32 and Neopixels has been discussed for quite some time ([139](https://github.com/adafruit/Adafruit_NeoPixel/issues/139)).

JulianS-Uni, 2019-04-20:
@moose4lord Thank you for the RMT driver.
In case someone else needs this "fix", in `ESP32_RMT_Driver.h` provided by @moose4lord in his [comment](https://github.com/kitesurfer1404/WS2812FX/issues/177#issuecomment-484263083) i changed the Tick times in line 9 to 11 to the following to work with my ws2812.
```c++
// timing parameters for WS2812B LEDs. you may need to
// tweek these if you're using a different kind of LED
#define T1_TICKS      350 / RMT_TICK // 250ns
#define T2_TICKS      300 / RMT_TICK // 625ns
#define T3_TICKS      600 / RMT_TICK // 375ns
#define RESET_TICKS 50000 / RMT_TICK // 50us
```

moose4lord, 2019-04-20:
Hi Julian, I'm curious how you came up with the T1/T2/T3 timing parameters for your setup. The numbers don't look like anything I see in the spec sheets, and the T2=300ns setting provides a very short pulse that distinguishes a "0" from a "1". I would not think you'd get reliable results with those timings. Did you arrive at those values through trial and error?

JulianS-Uni, 2019-05-03:
I have to reopen this sadly. I encountered a problem with @moose4lord 's workarround when using 96 or more LEDs. I tried to do the rainbow animation for 95 LEDs which works just fine. When trying 96 LEDs the first 30 or 40 LEDs are just lighting static and the following LEDs are doing the animation. When trying more than 96 LEDs they all stay dark. Do you have any idea what happens here?

moose4lord, 2019-05-04:
I connected my ESP32 to a strip of 144 WS2812B LEDs and did not see the problem you describe. All 144 LEDs seem to work fine. I don't have any WS2812 LEDs, like yours, so I can't say if it's a subtle difference between the LEDs or something else is going on. Is it just the rainbow animation that fails or does it happen with any animation?

JulianS-Uni, 2019-05-04:
I tried the rainbow animation and the rainbow cycle. Both do not work. But then i decreased the animation speed from 5 to 5000 and the animation started to work. The breath animation does work even with very high speeds (5).

moose4lord, 2019-05-04:
The breath animation code ignores the speed parameter completely and uses hardcoded timing delays, so that's probably why it works for you while other animations do not. Setting speed=5 is very fast. It's more typical to set speed in the range of 500-2000, or if you are driving lots of LEDs (like hundreds of them), then setting speed to 10000-30000 is typical.

Having said that, I set the speed to 5 and ran the rainbow and rainbow_cycle animations with no problems on my strip of 144 WS2812Bs. So there may be something else going on with your setup.

JulianS-Uni, 2019-05-11:
I did some testing today. The problems seem to correlate with the number of LEDs and the animation speed. I was able to drive 128 LEDs with an animation speed of 1000. I post my code so that someone can evaluate it. With 129 LEDs the first LEDs does not light the rest is doing fine and with 130 LEDs the LEDs all LEDs stay dark.
```c++
#include <WS2812FX.h>
#include "ESP32_RMT_Driver.h"

WS2812FX ws2812fx = WS2812FX(130, 27, NEO_GRB + NEO_KHZ800);

void setBootLed(uint8_t _OnOff){
	pinMode (LED_BUILTIN, OUTPUT);
  	digitalWrite(LED_BUILTIN, _OnOff);
}

void startBoot(){
	setBootLed(LOW);
	Serial.begin(115200);
	Serial.println("\n\nBoot in:");
	for(uint8_t t = 3; t > 0; t--){
		Serial.printf(".. %d ..\n", t);
		delay(1000);
	}
}

// Needed for ESP32 workaround for flashing lights when using WiFi
void rmtShow(void) {
  	uint8_t *pixels = ws2812fx.getPixels();
  	uint16_t numBytes = ws2812fx.getNumBytes() + 1;
  	rmt_write_sample(RMT_CHANNEL_0, pixels, numBytes, false);
}

void setupWS2812(){
	// ESP32 workaround for flashing lights when using WiFi
	rmt_tx_int(RMT_CHANNEL_0, ws2812fx.getPin());
	ws2812fx.setCustomShow(rmtShow);
	// setup ws2812fx
	ws2812fx.init();
	ws2812fx.setBrightness(10);
	ws2812fx.setSpeed(1000);
	ws2812fx.setMode(FX_MODE_RAINBOW);
	ws2812fx.setColor(0xff0000);
	ws2812fx.start();
}

void setup() {
	startBoot();
	setupWS2812();
	setBootLed(HIGH);
}

void loop(){
	ws2812fx.service();
}
```
For the timings in the ESP32_RMT_Driver.h i use following timings for my WS2812b LEDs.
```c++
// timing parameters for WS2812B LEDs. you may need to
// tweek these if you're using a different kind of LED
#define T1_TICKS      350 / RMT_TICK // 250ns
#define T2_TICKS      300 / RMT_TICK // 625ns
#define T3_TICKS      600 / RMT_TICK // 375ns
#define RESET_TICKS 50000 / RMT_TICK // 50us
```

moose4lord, 2019-05-11:
I ran your sketch with a DOIT ESP32 DEVKIT V1 and a string of 144 WS2812B LEDs, and didn't see any problem. It works if I use 128, 129, 130 or all 144 LEDs. Sorry, I'm not really sure why it's not working for you.

JulianS-Uni, 2019-05-16:
One last idea. Are you using PlatformIO or Arduino to flash your devices? I am using PlatfromIO.

moose4lord, 2019-05-16:
I use the Arduino IDE on a Mac.

JulianS-Uni, 2019-12-21:
I got some time and motivation to look further into this issue. And made a few more observations.

So firstly the problem again:
I am using the [WebServer example scetch](https://github.com/kitesurfer1404/WS2812FX/tree/master/examples/esp8266_webinterface) from this repo with the modification that I am using a ESP32 and the RMT driver.
Sometimes when i change animations the new choosen animation won't play.

My observations:

What helps:

- When I have the situation that a animation won't play a can reduce the animation speed and the animation will start.
- Alternatively i can reduce the number of LEDs in my WS2812FX constructor
- Using the default show() function instead of the custumShowFunction() with the RMT patch does help too. I can go down to a animation speed of 2 and animation will still work (sadly thats not a option because i have LED flicker without RMT and WiFi)

What does not help:

- Chaging the timing at the RMT driver does not really have an effect
- I tried using a resistor and a capacitor, both do not have any effect
- I tried powering the LEDs with 3,3V or 5V both no impact
- Using Arduino IDE instead of PlatformIO

What I did not test:

- Powering the LEDs with 5V and using a levelshifter for the signal
- Using another ESP32 model (tried 3 DOIT clones from ebay which i got from the same store)
- Using another OS (I am using Win10 atm)

My conclusion so far:
I could imagine that when I increase the animation speed and LED count is high enough the time to update the whole LED strand is longer than the time between the animation frames.
What i dont understand with this is that the default show() does work and the RMT show() does not.

moose4lord, 2019-12-21:
I hooked up my ESP32 (an DOIT ESP32 DEVKIT V1) again and still do not see the problems you're describing. I've attached the sketch I wrote which merges the esp8266_webinterface sketch with the RMT driver. Maybe that will help debug your issue.

As for hardware, I power my 144 LEDs with a 5V 40W power supply with an 820uF decoupling capacitor wired close to the LED strip. I use a 74HC125 chip to level shift the ESP32's 3.3V GPIO and the 5V WS2812B LED strip. There's a 100 ohm resistor between the 74HC125 output and the LED's Din wire. I'm using Arduino IDE v1.8.10, ESP32 Core v1.0.4 and WS21812FX v1.2.2. I use a Mac and the Arduino IDE, but I doubt your Windows 10 setup is the problem.

[esp32_test.zip](https://github.com/kitesurfer1404/WS2812FX/files/3991898/esp32_test.zip)

JulianS-Uni, 2019-12-22:
Hey there, first I want to tank you for your constant replies.
I tested another LED strand today and that worked. So I think it is not the software or the esp I use. I will another strand tomorrow and than hopefully close this issue!

JulianS-Uni, 2019-12-29:
I had some LEDs left from the 5m strand I had trouble with and tested them totay. Those LEDs have the same issue whereas all other strands I tested did not have any problems. So I am safe to say now that the LEDs I used are the problem. I really do not know why and why it is only happening with the RMT driver but I think I will mark this as solved for now. Thanks for all your help!

closed 2019-12-29 by JulianS-Uni