kitesurfer1404/WS2812FX #178: Change start led

Maurotb, 2019-04-21:
Hi,
i have 894 led ws2815 ,
If i connect my d1 mini pro af first led , i have 6mt of cable and effect not work correctly.
If i connect my board at led ~100 , i have 1mt of cable and all work correctly but some
effect start at bad led obviously...
I can't put the board near LED 1 and I don't have the possibility to use repeaters...
There is a way to shift start led compared to where is the signal cable connected at program?
Thanks

kitesurfer1404, 2019-04-21:
You might want to take a look at the segments-feature in the examples. You can select any "segment" of the LEDs to run your effect on. This way you can shift the starting LED to any position.

moose4lord, 2019-04-21:
I have a setup similar to yours, with about 3 meters of wire between the ESP8266 and the first WS2812B. Make sure you're using a 3.3V-to-5V level shifter on the output of the ESP8266. I've used both a [bidirectional level shifter](https://www.adafruit.com/product/757) and a [74HC125](https://www.adafruit.com/product/1787), and they work well. Also, I did NOT put a series resistor between the level shifter and the LEDs. When driving such a long wire, the series resistor can distort the signal enough to cause problems.

Maurotb, 2019-04-21:
> I have a setup similar to yours, with about 3 meters of wire between the ESP8266 and the first WS2812B. Make sure you're using a 3.3V-to-5V level shifter on the output of the ESP8266. I've used both [...]

I have a ws2815 and 12v level shifter, no way to work with 6mt...
in datasheet of ws2812 i see max lenght two led 5mt in my ws2815 in datashet see max lenght 3mt
I use mclighting , i need to make a patch to set 'segment' as @kitesurfer1404

moose4lord, 2019-04-21:
Oh sorry, I didn't realize the WS2815 is a 12V part.

The parameters for the setSegment() function are: segment_index (0-7), start_led, stop_led, mode, color, speed, reverse (boolean). So I think your setSegment should be this:
```c++
ws2812fx.setSegment(0, 100, 300, FX_MODE_COMET, 0xFF0000, 1000, false);
```
Unfortunately, that doesn't completely solve your problem, since leds 0-99 would still be unassigned and would be off. The comet effect would NOT run from leds 100-300, then wrap around to leds 0-99.

I'm not really sure how your leds are configured. Are they one long string of 894 leds and you're injecting the signal from the WiMos at led 100? Does the Dout of LED 893 connect to the Din of LED 0? Can you describe how the LEDs are wired together?

Maurotb, 2019-04-21:
okay so i have a problem :(
I have 894 LEDs in the entrance arranged in a circle on the ceiling, applied to a polystyrene profile.
For the comet effect and others, I need the beginning of the strip to be a precise point to have a nice effect, but unfortunately compared to my wemos d1 mini pro it is 5mt away...
Starting the strip where there is currently the LED n.100, the cable is reduced a lot and led works correctly but unfortunately the comet effect starts now from the wrong point...
I don't want to put external boxes to get closer to the wemos and I have little space in the polystyrene profile, I hoped we could change the starting point from software ...
I also don't like the idea of ​​putting a LED repeater without a protection box 
Thanks

moose4lord, 2019-04-22:
OK, if the LEDs are arranged in a circle, can you connect the Dout pin of LED 893 to the Din pin of LED 0? If you're using the WeMos to drive the Din pin of LED 100, I'm still not clear on what will drive the Din pin of LED 0. Without driving the Din pin of LED 0, LEDs 0-99 will do nothing.

Maurotb, 2019-04-22:
>  can you connect the Dout pin of LED 893 to the Din pin of LED 0?
no, i'ts not possible...

imagine a room with a door, I want the comet effect to start from above the door for my 1000 led circle.
I should inject the signal over the door and at that point interrupt DI / DO
In this case, the visual LED 0 corresponds to the electric LED 0
In this way, however, the signal cable is too long for me
Then above the door I leave connected di/do and 300 LEDs later I interrupt di/do and I inject the signal
In this case, I would like visual led n.0 to be led n.700 electric to make start of comet above the door

moose4lord, 2019-04-23:
Ok, I think I see what you're trying to do. Unfortunately, I couldn't come up with an elegant solution. Dividing the LEDs into segments, and trying to coordinate the animations in each segment, didn't work well. I created a custom show() function that copied the pixel data, manipulated it to offset the "first" LED, ran the default show() function, then copied the original pixel data back. It's an awful hack, but it worked.

```c++
#include <WS2812FX.h>

#define LED_COUNT 144
#define LED_PIN    D2

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  
  ws2812fx.init();
  ws2812fx.setBrightness(255);

  ws2812fx.setCustomShow(myCustomShow);

  ws2812fx.setSegment(0, 0, LED_COUNT - 1, FX_MODE_COMET, RED, 2000, NO_OPTIONS);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}

#define OFFSET 25
void myCustomShow(void) {
  static uint8_t tmpPixels[LED_COUNT * 4]; // allocate space for a copy of the ADafruit pixel data
  uint8_t bytesPerPixel = ws2812fx.getNumBytes() / ws2812fx.getLength(); // 3=RGB, 4=RGBW

  // copy the Adafruit pixel array
  memcpy(tmpPixels, ws2812fx.getPixels(), ws2812fx.getNumBytes());

  // offset the data in the Adafruit pixel array
  memcpy(ws2812fx.getPixels() + OFFSET * bytesPerPixel, tmpPixels, (LED_COUNT - OFFSET) * bytesPerPixel);
  memcpy(ws2812fx.getPixels(), tmpPixels + (LED_COUNT - OFFSET) * bytesPerPixel, OFFSET * bytesPerPixel);
  
  // run the standard Adafruit show() function
  ws2812fx.Adafruit_NeoPixel::show();

  // copy the original pixel data back to the Adafruit pixel array
  memcpy(ws2812fx.getPixels(), tmpPixels, ws2812fx.getNumBytes());
}
```

Maurotb, 2019-04-23:
I try with this, Thanks!

Maurotb, 2019-06-01:
Tested and fully working!

closed 2019-09-11 by moose4lord

khammel2, 2021-05-18:
I'm new to github, not sure where to post this question,

I have a similar issue, although was looking for a different kind of work around. 
I have a nonlinear, non symmetrical 3d object of led's. Imagine a tangle of led's on a 3d object, and I'd like to be able to map each individual led pin order, into a specific linear line.

I would like to be able to re-map the led's 1-50, but instead of sequencial order have them mapped in a specific set order. It would be like having a cross over function before the led's are sent out.
so instead of led order 1,2,3,4,5,6,7....,49,50
instead I'd want            6,17,2,32,.... 45,3
led1 = led6
led2 = led 17
....
led49 = led 45
led50 = led 3
and then have the re-mapped order sent to the pre-built functions in WS2812fx

Is this possible, and is there an easy workaround?

By doing the above cross over code, would allow for easier use of setting segment ranges in certain circumstances. 
(less ranges to be mapped if the led order is changed)