kitesurfer1404/WS2812FX #182: Deselect effects in auto mode cycle

aeromojito, 2019-05-14:
I would like an easy way to use auto mode cycle and deselect some modes. I see this was previously answered but the links take me to a 404 error.

moose4lord, 2019-05-14:
I assume you're referring to the esp8266_webinterface example sketch. To create a custom list of effects, add the effects/modes you want to use to the myModes array. Something like this:
```c++
uint8_t myModes[] = { // *** optionally create a custom list of effect/mode numbers
    FX_MODE_STATIC,
    FX_MODE_BLINK,
    FX_MODE_COLOR_WIPE,
    FX_MODE_RAINBOW_CYCLE,
    FX_MODE_CHASE_COLOR,
    FX_MODE_RUNNING_COLOR,
    FX_MODE_LARSON_SCANNER,
    FX_MODE_FIREWORKS
};
```

aeromojito, 2019-05-14:
Actually, I was referring to the auto_mode_cycle sketch
`#include <WS2812FX.h>

#define LED_COUNT 144
#define LED_PIN 6

#define TIMER_MS 5000

// Parameter 1 = number of pixels in strip
// Parameter 2 = Arduino pin number (most are valid)
// Parameter 3 = pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_RGB + NEO_KHZ800);

unsigned long last_change = 0;
unsigned long now = 0;

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setSpeed(1000);
  ws2812fx.setColor(0x007BFF);
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.start();
}

void loop() {
  now = millis();

  ws2812fx.service();
  
    if(now - last_change > TIMER_MS) {
    ws2812fx.setMode((ws2812fx.getMode() + 1) % ws2812fx.getModeCount());
    last_change = now;
  }
}
`

aeromojito, 2019-05-14:
I see that it counts up by one every 5000ms. I would like to add exceptions for the effects I don't want to see. For instance, I don't like 2, 6, or 8. If the mode = 2, 6, or 8, it will skip it and count one higher.

moose4lord, 2019-05-15:
Oh, sorry. I think if the list of modes you want to skip is small, you can add some code to test the mode and reassign it. Something like this:
```c++
if(now - last_change > TIMER_MS) {
    uint8_t nextMode = ws2812fx.getMode() + 1;
    if(nextMode == 2) nextMode = 3;
    if(nextMode == 6) nextMode = 7;
    if(nextMode == 8) nextMode = 9;
    ws2812fx.setMode(nextMode % ws2812fx.getModeCount());
    last_change = now;
  }
```

closed 2019-09-11 by moose4lord