kitesurfer1404/WS2812FX #183: Enhance effects/usability for this library

HiranChaudhuri, 2019-05-21:
Hello,

I am using this library as it is embedded into McLighting, and I want to spice up my living room with it.I fully understand WS2812fx is just an effect library and McLighting actually adds functions like OTA, Webservice and MQTT and other APIs to control these effects.

Now I realize that these continuous effects actually are great for a disco party but in an comfortable illumination scene I'd like to use something more simplistic. I thought of entry/exit sequences like chase or fade but do them once only, then keep the colors static.

So now the question, and I am unsure if it has been asked/implemented already:
Would it be possible to set a repetition count for the effects? Or is there a way to halt the system for not repeating a chase, fade, or other effect?

moose4lord, 2019-05-21:
There is an isFrame() function which returns a boolean value signifying an animation has stepped to the next "frame" (i.e the LEDs have changed). You can use this to count animation frames and do something different when the count exceeds a value. For example:
```c++
#include <WS2812FX.h>

#define LED_PIN    27  // digital pin used to drive the LED strip
#define LED_COUNT 144  // number of LEDs on the strip

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setBrightness(64);

  // parameters: index, start, stop, mode, color, speed, reverse
  ws2812fx.setSegment(0, 0, LED_COUNT-1, FX_MODE_BLINK, RED, 1000, NO_OPTIONS);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();

  static int frameCount = 0;
  if(ws2812fx.isFrame()) { // did the animation step to the next frame?
    frameCount++;          // increment the frame counter
    if(frameCount == 10) { // after 10 frames (blink on/off 5 times), pause the animation
      ws2812fx.pause();
    }
  }
}
```

HiranChaudhuri, 2019-05-22:
It sounds good there is a possibility to track effect repetitions. The example given looks confusing for me though.

```
void loop() {
  ws2812fx.service();

  static int frameCount = 0;
  if(ws2812fx.isFrame()) { // did the animation step to the next frame?
    frameCount++;          // increment the frame counter
    if(frameCount == 10) { // after 10 frames (blink on/off 5 times), pause the animation
      ws2812fx.pause();
    }
  }
}
```

I guess in ws2812fx.service() the effect is moved forward one step.
But then frameCount is introduced and initialized with 0, just to be increased when isFrame() returns true. Does isFrame() also move the effect one step? Or how would frameCount ever reach the value 10 in this example?

moose4lord, 2019-05-23:
Sorry, I used a static variable in the loop() function, which can be confusing. The static frameCount variable is initialized once, the first time loop() is executed, and retains its data each time loop() is called. I might be easier to understand if it were a global variable:
```c++
int frameCount = 0;
void loop() {
  ws2812fx.service();

  if(ws2812fx.isFrame()) { // did the animation step to the next frame?
    frameCount++;          // increment the frame counter
    if(frameCount == 10) { // after 10 frames (blink on/off 5 times), pause the animation
      ws2812fx.pause();
    }
  }
}
```
To answer your question, the service() function does not always change the LEDs. The function checks to see if it's time to display the next animation frame, and if so, updates the LEDs (and sets a flag so isFrame() will return true). Since service() is embedded in the loop() function, it gets called over and over again, and between animation frames it simply returns without making any changes to the LEDs.

Hope that makes it clearer.

HiranChaudhuri, 2019-05-23:
Thank you for the explanation. Yes, now it is clear (and I learned a bit about static variables inside functions).
I will give this a try. :-)

moose4lord, 2019-05-24:
You are absolutely correct. Counting frames is a flexible way of getting feedback about how the animation is progressing, but it does require some finesse, just as you describe.

Some of the effects (specifically the CHASE, COLOR_WIPE and LARSON_SCANNER effects) implement the isCycle() function, which is similar to isFrame(), but flags when the animation has completed one cycle. Unfortunately, isCycle() isn't implemented everywhere, but If you're using those effects, you can use it to detect end-of-animation.

Another option is to simply change the animation for a specific amount of time, using a simple timer. It might take a little trial and error testing to get it timed right, but you don't have to count frames and do some goofy calculation to account for the number of LEDs or brightness steps
```c++
long timer = 0;
void loop() {
  ws2812fx.service();

  if(millis() > timer + 10000) { // wait 10 seconds, then pause
      ws2812fx.pause();
  }
}
```

HiranChaudhuri, 2019-05-24:
Well, I think isCycle() is what I was searching.
So I'd take your example from above but replace isFrame() with isCycle() and should be done.
Let me have some time to try it out. :-)

Maurotb, 2019-06-01:
Hi , i have similar request,
for usability with mclighing in my home, is possible to make
1)
from  off to  on (whatever effect)-> luminosity from 0 to 254 gradually
from  on to  off  (whatever effect)-> luminosity from 254 to 0  gradually
2)
in static mode, when change from color A to color B , blend from A to B?
Thanks!

moose4lord, 2019-06-01:
Unfortunately, WS2812FX doesn't really provide for transitions between effects. There's a couple problems you run into. The first is that many of the effects can use more than one color (up to three), so it gets messy transitioning all three colors simultaneously. The second problem is that not all the LEDs are updated when an animation "frame" is displayed. For example, the COLOR_WIPE animation only changes one LED for each frame, the other LEDs simply maintain their state. So this would cause problems if you're trying to update all LEDs during some kind of color transition.

It's also tempting to use the global brightness parameter to create fadein/fadeout effects, but the problem with that is the Adafruit brightness feature does not keep track of the original color data (i.e the color data _before_ being modified by the brightness setting) when setting each LEDs color. From the Adafruit Neopixel Uberguide:
> setBrightness() was intended to be called once, in setup(), to limit the current/brightness of the LEDs throughout the life of the sketch. It is not intended as an animation effect itself! The operation of this function is “lossy” — it modifies the current pixel data in RAM, not in the show() call — in order to meet NeoPixels’ strict timing requirements. Certain animation effects are better served by leaving the brightness setting at the default maximum, modulating pixel brightness in your own sketch logic and redrawing the full strip with setPixel().

Having said all that, your second question, creating a color transition in static mode, can be implemented using the libs color_blend() function. Here's an example:
```c++
#include <WS2812FX.h>

#define LED_PIN   D2   // digital pin used to drive the LED strip
#define LED_COUNT 144  // number of LEDs on the strip

#define START_COLOR RED // start the blend with this color
#define END_COLOR BLUE  // end the blend with this color
#define BLEND_RATE 20   // change blended color every 20ms

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setBrightness(255);

  // parameters: index, start, stop, mode, color, speed, reverse
  ws2812fx.setSegment(0,  0,  LED_COUNT-1, FX_MODE_STATIC, START_COLOR, 1000, NO_OPTIONS);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
  blend();
}

long blendTimer = 0;     // timer to trigger when a new blended color is calculated
uint8_t blendAmount = 0; // range 0-255, where 0 is START_COLOR and 255 is END_COLOR
void blend() {
  if(millis() > blendTimer + BLEND_RATE) { // time to update color?
    uint32_t newColor = ws2812fx.color_blend(START_COLOR, END_COLOR, blendAmount); // calc the new color
    ws2812fx.setColor(newColor); // set the new color
    blendTimer = millis();       // reset the timer
    blendAmount++;               // incr the blend amount
  }
}
```
If you want to create a fadein/fadeout transition, you would make the START_COLOR or END_COLOR equal to BLACK.
```c++
#define END_COLOR BLACK
```

Maurotb, 2019-06-02:
And without transition?
Normally i use "running light" but when i tun on or off , i get immediatly 100% or 0%
I have try to use setBrightness in a loop from 0 to 254 but i see flickering , is bad...
No other way?
Thanks

moose4lord, 2019-06-02:
The RUNNING_LIGHTS effect only uses one color and updates every LED for each frame, so it can be used in place of the STATIC effect in my demo sketch. There is one caveat though. In order to get RUNNING_LIGHTS to work in this way, you'd need to comment out the line in the library's WS2812FX.cpp file that resets the segment runtime when the setColor() function is called (near line 204):
```c++
void WS2812FX::setColor(uint8_t seg, uint32_t c) {
//  resetSegmentRuntime(seg); // comment out this line
  _segments[seg].colors[0] = c;
}
```
It's a hack, but resetting the segment runtime while the RUNNING_LIGHTS effect is in progress causes it to misbehave, so the line needs to be commented out.

Maurotb, 2019-06-05:
@moose4lord I'm trying to modify mclighting but it's very very difficult to be able to do the fade like this on mclighting ...
In your opinion, it is possible to do the fade by acting with the customshow function and making a strip-> getPixel - FADE - strip-> setPixelColor ()?

moose4lord, 2019-06-05:
That's certainly possible. Here is an example that might get you going. It creates a brightnessTarget variable that can be changed dynamically. Set it to 0 and the LEDs will fade to black. Set it to 255 and the LEDs will slowly increase to full intensity.
```c++
// add a global variable called brightnessTarget in definitions.h
int brightnessTarget = 255;
```
```c++
void myCustomShow(void) {
  static uint8_t tmpPixels[NUMLEDS * 4]; // space for a copy of pixel data
  static int myBrightness = 0;
  static long myBrightnessTimer = 0;

  uint8_t bytesPerPixel = strip->getNumBytes() / strip->getLength(); // 3=RGB, 4=RGBW
  uint8_t *pixels = strip->getPixels(); // the raw Adafruit pixel array

  // copy the Adafruit pixel array
  memcpy(tmpPixels, strip->getPixels(), strip->getNumBytes());

  // adjust the data in the Adafruit pixel array by myBrightness
  for(uint16_t i=0; i <strip->getNumBytes(); i++) {
    pixels[i] = (myBrightness * pixels[i]) / 256;
  }

  // run the standard Adafruit show() function
  strip->Adafruit_NeoPixel::show();

  // copy the original pixel data back to the Adafruit pixel array
  memcpy(strip->getPixels(), tmpPixels, strip->getNumBytes());

  // adjust myBrightness up or down depending on myBrightnessTarget
  if(myBrightness != brightnessTarget && millis() > myBrightnessTimer) {
    myBrightnessTimer = millis() + 20; // next fade step in 20ms

    if(myBrightness < brightnessTarget) myBrightness++;
    else myBrightness--;
  }
}
```
Test the custom show function by adding this at the end of the loop() function.
```c++
  static long fadeTimer = 0;
  if(millis() > fadeTimer + 10000) { // change brightnessTarget every 10 seconds
    brightnessTarget = brightnessTarget == 0 ? 255 : 0; // fadein or fadeout
    Serial.print("brightnessTarget ");Serial.println(brightnessTarget);
    fadeTimer = millis();
  }
```
The problem with using a custom show function is that it's tied to the animation speed (it only gets executed once for each animation frame). So if you have a very slow animation, the fadein/fadeout will be very slow too. But this might be OK for your purposes.

HiranChaudhuri, 2019-06-10:
Hello again.

I played around with my own custom effect in McLighting, and actually it was easy to develop. I created a clock style effect that would display the time across 60 LEDs. Currently it shows a wrong time, and I would probably have to sync it with NTP. Nothing too fancy but I like it. On the other hand it is not connected to the fading I described earlier.

However I'd like to share my experience:
I am getting a bit annoyed by the fact that adjusting something (color, speed, brightness) would reset an effect rather than just tweaking it. This reset does not happen to my clock, but the time runs only when the effect is selected. So there are two conditions when I'd like to NTP-sync the clock:
a) at midnight, when the hours are reset to 0
b) when the effect gets started

And at B) I realize we can only specify the effect's run function, but nothing else.
How about making an effect a class of its own? I then could have all the variables and methods it needs, plus some methods to manage the lifecycle:
- setup: Initialize the effect. I'd use this for an NTP-sync
- entry: take over from the previous effect. This could be a fade in or what
- run: the existing effect method
- exit: get ready to handover to the next effect
- cleanup: if necessary close resources

moose4lord, 2019-06-11:
I think you're right about resetting the segment runtime when the segment color or speed is changed. It's probably not necessary and can cause problems as mentioned in this thread. I have a tweak to the fireworks effect that I've wanted to release, so I'll look into removing the runtime reset at the same time.

As far as refactoring the library to be more object oriented, that's a tall order. It would be quite a bit of work, and when it was done, I'm not sure it would fit in an Arduino's limited flash memory. A nice idea, but probably not worth the effort. Just my 2 cents.

HiranChaudhuri, 2019-06-11:
> I think you're right about resetting the segment runtime when the segment color or speed is changed. It's probably not necessary and can cause problems as mentioned in this thread. I have a tweak to [...]

Sounds great. :-)
 
> As far as refactoring the library to be more object oriented, that's a tall order. It would be quite a bit of work, and when it was done, I'm not sure it would fit in an Arduino's limited flash memory. A nice idea, but probably not worth the effort. Just my 2 cents.

Ok, how about having more than one function per effect? Those other functions would be optional, but we could create a struct with pointers to several methods. Strip setup, as it could be run during the setup of the whole sketch. Strip cleanup, as we would never cleanup resources within a microcontroller. With the remaining three functions you could still create a nice sequence of effects including handovers. So would two more function pointers (I guess 32 bits each) waste too much memory already? You have about 64 effects, I am talking about 64*2*4 bytes = 512 bytes. Still too much?

kitesurfer1404, 2019-06-11:
Can you program a short sketch for such a layout? I was thinking about the object oriented FXs some time ago. I would love to see some modularity which allows for simple adding and removing FXs to/from the library and removing the name-strings from the code if you don't need them and so on.
I have no deep knowledge on the possibilities of the compiler or preprocessor. I would love to have some options like
#define fx_include_running_lights
#define fx_include_blink

or
#define fx_no_names

Sth. like this. Or a simple "registry" for FX objects, so you can build your own order of FXs and/or simply remove them from your compiled sketch.

Or maybe just some links to good compiler/preprocessor tutorials that show all the options with good examples.

HiranChaudhuri, 2019-06-11:
Indeed the struct I spoke about is very close to what compilers today generate for classes with virtual methods. Let me try something, but I I cannot promise how fast I can get it done. I am not a C++ guy although I love objects.

Right now I assume by not including an effect you do not want to include unused code but really exclude that functionality from the binary code.

moose4lord, 2019-06-12:
The "pick and choose" feature you're talking about is already used to a small extent in WS2812FX. In WS2812FX.h you'll see this code:
```c++
#ifdef REDUCED_MODES
      _mode[FX_MODE_BREATH]                  = &WS2812FX::mode_static;
      _mode[FX_MODE_RUNNING_LIGHTS]          = &WS2812FX::mode_static;
      _mode[FX_MODE_ICU]                     = &WS2812FX::mode_static;
#else
      _mode[FX_MODE_BREATH]                  = &WS2812FX::mode_breath;
      _mode[FX_MODE_RUNNING_LIGHTS]          = &WS2812FX::mode_running_lights;
      _mode[FX_MODE_ICU]                     = &WS2812FX::mode_icu;
#endif
```
If the REDUCED_MODES preprocessor flag is set, the STATIC effect will be substituted for the BREATH, RUNNING_LIGHTS and ICU effects. The BREATH, RUNNING_LIGHTS and ICU effects functions will be compiled, but the linker will notice they are never referenced and not include them in the executable uploaded to the Arduino. You can accomplish the same thing by simply commenting out the effects in the _mode array:
```c++
      _mode[FX_MODE_SCAN]                    = &WS2812FX::mode_scan;
      _mode[FX_MODE_DUAL_SCAN]               = &WS2812FX::mode_dual_scan;
      _mode[FX_MODE_FADE]                    = &WS2812FX::mode_fade;
//      _mode[FX_MODE_THEATER_CHASE]           = &WS2812FX::mode_theater_chase;
//      _mode[FX_MODE_THEATER_CHASE_RAINBOW]   = &WS2812FX::mode_theater_chase_rainbow;
      _mode[FX_MODE_TWINKLE]                 = &WS2812FX::mode_twinkle;
```
In this case, the THEATER_CHASE effects won't be included in the executable, so will not take up space in the Arduino's flash memory.

That said, I think one of WS2812FX's strengths is that, unless you're severely limited by the microprocessors flash memory, all of the effects are available at runtime. You can create a control sketch to choose any of the effects on the fly. You don't have to choose the effects you want at compile time.

moose4lord, 2019-06-12:
I have released WS2812FX v1.2.0, which contains the segment runtime reset fix.

closed 2020-04-06 by moose4lord