kitesurfer1404/WS2812FX #185: Example fo ws2801fx and espalexa

ricardo-brenner, 2019-07-05:
Do you have any example of the use of the ws2801fx with alexa?

moose4lord, 2019-07-05:
There is an Alexa demo sketch in the examples folder.

ricardo-brenner, 2019-07-05:
The example is for the ws2812fx, it will work with the ws2801fx? what changes are necessary?

moose4lord, 2019-07-05:
For my ws2801 LEDs the order of each color is RGB, not GRB as in the example. Not sure if all ws2801s are built that way, but if yours are the same, just use the NEO_RGB macro when you create the WS2812FX object. Other then that the demo sketch should work for you.

ricardo-brenner, 2019-07-05:
i haven't unsderstand your answer, theres no NEO_RGB on ws2801fx library.
Can you send me your code for the alexa with ws2801 strips?

moose4lord, 2019-07-05:
Look in the ws2812fx/examples folder and load the **ws2812fx_alexa** sketch into your IDE. On line 122 you'll see this line:
```c++
WS2812FX ws2812fx = WS2812FX(NUMLEDS, DATAPIN, NEO_GRB + NEO_KHZ800);
```
If your ws2801 LEDs are like mine, you'll need to change that line to this:
```c++
// change NEO_GRB to NEO_RGB
WS2812FX ws2812fx = WS2812FX(NUMLEDS, DATAPIN, NEO_RGB + NEO_KHZ800);
```
Also, please read the documentation at the top of the ws2812fx_alexa sketch. Depending on the firmware in your Alexa device and the version of ESP8266 core software you're using, you may need to implement the hack mentioned in the documentation.

ricardo-brenner, 2019-07-05:
but ws2801 has 4 pins, I need to set data and clock, i can't use the ws2812fx

moose4lord, 2019-07-06:
Oh, I apologize, my LED strings use ws2811 chips, not ws2801. The ws2811 chip supports the one-wire interface used by the WS2812FX library. Unfortunately, the ws2801 is not supported. Sorry.

closed 2019-09-11 by moose4lord