kitesurfer1404/WS2812FX #188: Switch from Adafruit Library to NeoPixelBus

Zefiro, 2019-08-23:
This is a nice project and the provided collection of build-in examples and web interface spared me from building all of those myself for my ESP8266-based project. So thank you for that.

However, I'm facing severe wifi connectivity issues, and I suspect the Adafruit library uses bitbanging and disabled interrupts. For the ESP line of chips, there are alternatives, namely the [NeoPixelBus](https://github.com/Makuna/NeoPixelBus) library, which can use UART/DMA on certain hardware pins.

I took a quick look at the sources, however unfortunately this library uses inheritance instead of composition, so I did not manage to swap out the Adafruit library for the NeoPixelBus one easily.

Maybe someone with more C++ knowledge could provide a PR that uses the other library, or let's us choose with some #ifdef which variant we would like to use.

moose4lord, 2019-08-23:
You should look at the example/ws2812fx_dma sketch, which incorporates the NeoPixelBus DMA code into a custom show function. It uses the ESP8266's DMA hardware, rather than bit-banging, to drive the LEDs.

If you have an ESP32, look at the example/ws2812fx_esp32 sketch. It uses the ESP32's RMT hardware in a similar fashion to avoid the problems caused by disabling interrrupts.

Zefiro, 2019-08-24:
That's a very nice way of handling this, thank you.

Makuna, 2019-09-08:
PLEASE do not use the DMA method they way you do it.  I may change the internal API and it will break you.

Zefiro, 2019-09-09:
@Makuna Thanks for the warning.
If it breaks someday, I'll change to looping over all pixels and using getPixelColor / SetPixelColor
However this needs color (object) conversion, and using both libraries like that also twice the amount of RAM.
Is there a better way?

Makuna, 2019-09-09:
See the [Wiki API here](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object-API#uint8_t-pixels), the buffer is still exposed in a similar way you are using it.  So you should be good to move everything over.

moose4lord, 2019-09-13:
I've released WS2812FX v1.2.1, which, among other things, refactored the ws2812fx_dma example sketch to use the proper NeoPixelBus API. The example sketch still allocates twice the memory, since both WS2812FX and NeoPixelBus allocate buffers for the pixel data. The example sketch is a quick and dirty way to implement a DMA solution. If you wanted a more memory efficient implementation, you'd need to forgo NeoPixelBus and recreate the DMA code in your sketch. Blah. Not worth the effort IMHO.

Also, if you're concerned about memory usage, I should mention that the DMA implementation requires an additional memory buffer for the I2S hardware. The I2S buffer is four times larger than the pixel buffer. So it is quite a memory intensive solution.

Zefiro, 2019-09-13:
Thanks, that's cool :)

For my current project the memory does fit, though if there would have been an easy way to slash the redundancy, that's worth asking.

Since my initial goal has been reached, I'm closing this issue.

closed 2019-09-13 by Zefiro