kitesurfer1404/WS2812FX #190: Auto mode cycle on digispark Attiny85 USB

thealienx, 2019-08-30:
Hi,
i am trying to compile the Auto Mode cycle on Digispark Attiny 85 USB board, but getting weird errors on compiling. 
------------------------
C:\................\Arduino\libraries\WS2812FX\src/WS2812FX.h:245:14: error: '__FlashStringHelper' does not name a type
 static const __FlashStringHelper* _names[] = {
---------------------------------------------------------
can someone help me in removing this error and compile the same.
the reason to run the code on Attiny85 is that it will minimize the use of external voltage regulators as used in esp8266-01 Board.

Thanks

moose4lord, 2019-08-31:
It doesn't look like the ATtiny85 core supports the __FlashStringHelper datatype.
See [https://forum.arduino.cc/index.php?topic=397627.0](https://forum.arduino.cc/index.php?topic=397627.0)

You could work around that by modifying the WS2812FX library to use char datatypes instead of __FlashStringHelper, but really the bigger issue is the ATtiny's very limited memory. Only 8k of flash (the bootloader uses 2k, so really only 6k available for your sketch) and only 512 bytes of SRAM. WS2812FX is a rather big library, so there's no way it's going to fit on a device with so little flash memory and SRAM.

You might want to consider an [Arduino Nano](https://projectiot123.com/2019/04/08/arduino-nano-for-beginners/).

thealienx, 2019-09-01:
Thanks for your reply.
no point in changing the original library. i was just trying to see if Attiny works or not by simple modification. Anyways let it be with esp8266-01. it works good on it.
your support is commendable.
Thanks

closed 2019-09-01 by thealienx