kitesurfer1404/WS2812FX #197: Using both WS2812FX and Adafruit_NeoPixel

trizin, 2019-11-08:
```
#define LED_PIN 6
#define LED_COUNT 288
#define MoveCens 7
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

```
Here is my code. When I set LED_COUNT to 144 I can use both Adafruits strip and ws2812fx together but when I set the LED_COUNT to 288 only the one which is written first is working.
Like:
```
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
```
Here strip is working but ws2812fx won't work.

```
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
```
and here ws2812fx effects are working but I can't change the color using strip.

This won't happen when I set the LED_COUNT below 200

moose4lord, 2019-11-08:
Creating an Adafruit_NeoPixel instance and a WS2812FX instance with the same LED_PIN is probably not going to work the way you want. Are you calling both strip.show() and ws2812fx.service() in your loop() function? If so, you're going to run into conflicts between the two instances trying to update the LEDs at different times.

Since the WS2812FX class inherits all the functionality of the Adafruit_NeoPixel class, I'm not sure why you would need both instances. What are you trying to do?

trizin, 2019-11-08:
> Creating an Adafruit_NeoPixel instance and a WS2812FX instance with the same LED_PIN is probably not going to work the way you want. Are you calling both strip.show() and ws2812fx.service() in your [...]

Thanks for your reply. I'm using ws2812fx for animations. I have a boolean called animation when it is true the loop is calling ws2812fx.service() otherwise I only use strip.show() etc. to set color. It was working properly till I increased the number of leds.

moose4lord, 2019-11-09:
WS2812FX should have no trouble with more than 167 LEDs. What kind of processor are you using? You might be running out of memory. Remember each instance of WS2812FX and Adafruit_NeoPixel allocates memory to store LED information, so running on some older Arduino boards, which only have one or two kilobytes of memory, could be a problem..

Can you post your entire sketch? I really think there's a better way to accomplish you goal other then creating a separate Adafruit_NeoPixel class. If you're just trying to switch between an animation and static color, maybe something like this would work in your loop() function:
```c++
if(animation) {
  ws2812fx.setMode(some_mode);
} else {
  ws2812fx.setMode(FX_MODE_STATIC);
}
ws2812fx.service();
```

trizin, 2019-11-09:
I'm using an arduino uno clone
IDE tells that 24658 bytes (76 %) been used, Max 32256 bytes

Here is my full code:
```#include <Adafruit_NeoPixel.h>
#include <WS2812FX.h>
#define LED_PIN 6
#define LED_COUNT 288
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
boolean animation = false;
void setup() {
  Serial.begin(9600);
  strip.begin();        
  strip.show();            
  strip.setBrightness(255); 
  ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
  ws2812fx.setColor(strip.Color(255,0,255));
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.start();
}
void loop() {
  if(animation){
    ws2812fx.service();
  }
  if (Serial.available() > 0) {
      int dataV = Serial.read();
      Serial.println(dataV);
      switch(dataV){
         case 48:
          if(animation)animation=false;else animation=true;
          break;
      case 49:
          colorWipe(strip.Color(255,0,0));
          break;
      case 50:
          colorWipe(strip.Color(0,255,0));
          break;
      }
  }
}

void colorWipe(uint32_t color) {
   for(int i=0; i<strip.numPixels()/4 ; i++) { 
    strip.setPixelColor(strip.numPixels()/4+i, color);
    strip.setPixelColor(strip.numPixels()/4-i, color);

    strip.setPixelColor(strip.numPixels()/4*3+i, color);
    strip.setPixelColor(strip.numPixels()/4*3-i, color);
    
    strip.show();                   
  }
}

moose4lord, 2019-11-09:
When the IDE says your sketch is using 24658 bytes (76 %) and the max is 32256 bytes, it is referring to the amount of Flash memory in your Arduino Uno.  The IDE should also say something similar to "Global variables use 820 bytes (32%) of dynamic memory, leaving 1740 bytes for local variables. Maximum is 2560 bytes", which is referring to the amount of SRAM memory. My earlier comment about "running out of memory" was referring to SRAM memory, not Flash memory.

Still, looking at your full code, it doesn't look like you should have a problem running out of memory when trying to use 288 LEDs. I tried your sketch with my Arduino Leonardo and it worked fine with 288 LEDs, so I'm not sure why you're having trouble with more than 167 LEDs.

P.S. your sketch really doesn't need to have both an Adafruit_NeoPixel instance and a WS2812FX instance. You can get rid of the "strip" code and replace all the references to "strip" with "ws2812fx" in the colorWipe() function.
```c++
void colorWipe(uint32_t color) {
   for(int i=0; i<ws2812fx.numPixels()/4 ; i++) { 
    ws2812fx.setPixelColor(ws2812fx.numPixels()/4+i, color);
    ws2812fx.setPixelColor(ws2812fx.numPixels()/4-i, color);

    ws2812fx.setPixelColor(ws2812fx.numPixels()/4*3+i, color);
    ws2812fx.setPixelColor(ws2812fx.numPixels()/4*3-i, color);
    
    ws2812fx.show();                   
  }
}
```

trizin, 2019-11-10:
Thanks, I changed all "strip" with "ws2812fx" and it is working now.

closed 2019-11-10 by trizin