kitesurfer1404/WS2812FX #199: How to count number of cycles in an effect?

SilverWings777, 2019-11-11:
Hello, and THANK YOU for your outstanding work!

I would like to know what would be the best way to keep track of how many "cycles" have occurred for a given effect?

For instance, I want to Blink a LED a certain number of times, and then stop.  Or I want to Fade a LED through a complete 0 - 255 - 0 "cycle" a certain number of times and then stop.

Is there a good way to do this?

Thanks,
Bill W.

moose4lord, 2019-11-12:
The library provides two functions that give feedback about how an effect is progressing:

1) the isCycle() function returns _true_ when an effect has completed one cycle. For example, with the LARSON_SCANNER effect, when the light travels from one end of the strip to the other, that's considered one cycle. You could change the color of the effect after each cycle with this code:
```c++
void loop() {
  ws2812fx.service();

  if(ws2812fx.isCycle()) {
    uint32_t randomColor = ws2812fx.color_wheel(ws2812fx.random8());
    ws2812fx.setColor(randomColor);
  }
}
```
Unfortunately, the isCycle() function is only implemented for the LARSON_SCANNER, COLOR_WIPE and CHASE modes. For other modes, or more fine grained feedback about a mode's progress, you should use the isFrame() function.

2) The isFrame() function returns _true_ every time the LEDs change (if you think of an effect as an animation, this marks one animation frame). Normally you would count frames and trigger a change after a certain number of frames. For example, with the BLINK mode, all LEDs on is one frame and all LEDs off is a second frame, so a blink "cycle" would be two frames. You could change the color of the effect after 5 blinks with this code:
```c++
void loop() {
  static int frameCount = 0;
  ws2812fx.service();

  if(ws2812fx.isFrame()) {
    frameCount++;
    if(frameCount == 10) { // 10 frames == 5 blinks
      uint32_t randomColor = ws2812fx.color_wheel(ws2812fx.random8());
      ws2812fx.setColor(randomColor);
      frameCount = 0; // reset the frame counter
    }
  }
}
```
Counting frames to trigger changes can be tedious, and often depends on the number of LEDs in your strip or the mode's fade rate, but it does provide quite a bit of control.

SilverWings777, 2019-11-12:
Moose - thanks VERY much for your expertise.  I thought there was a way to track every completed cycle.... but there's not (except for LARSON  and COLOR_SCANNER).   Saved me lots of time digging.  MUCH appreciated !!

closed 2020-04-06 by moose4lord

Devilscave, 2020-05-27:
And how do I have to use the cycle or frame function with the desired mode?

moose4lord, 2020-05-27:
Sorry, I don't understand your question.

Devilscave, 2020-05-28:
How can I apply "frame" to the Mode "Comet"? It does not work. And can I use several modes in one sketch and control them individually? Or to stop?

moose4lord, 2020-05-29:
Since a "frame" of the COMET mode is the comet advancing one LED, you'll need to factor in the number of LEDs into your "frame count" calculation. Here is an example of counting 5 comets and then stopping the effect.
```c++
#include <WS2812FX.h>

#define LED_PIN   10  // digital pin used to drive the LED strip
#define LED_COUNT 30  // number of LEDs on the strip

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(128);

  ws2812fx.setSegment(0,  0,  LED_COUNT-1, FX_MODE_COMET, RED, 1000, NO_OPTIONS);

  ws2812fx.start();
}

void loop() {
  static int frameCount = 0;
  ws2812fx.service();

  if(ws2812fx.isFrame()) {
    frameCount++;
    if(frameCount == LED_COUNT * 5) { // count 5 comets (30 LEDs/strip * 5)
      ws2812fx.stop();
    }
  }
}
```
You can run many modes in one sketch by executing the ws2812fx.setMode() function at different times to switch to a different mode.

Devilscave, 2020-05-30:
ahhh OK, now I understand. Thank you for your little demo. This would be nice as a demo in the example folder as "ws2812fx_frameCount". I saved it so.

But one more question, why can't I change variables in the loop to move the LEDs? He changes it, but the line "setSegment" does not accept it.

#include <WS2812FX.h>
#define LED_PIN   **3**  // digital pin used to drive the LED strip
#define LED_COUNT **300**  // number of LEDs on the strip
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

int **x**, **y**;

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(50);
  ws2812fx.setSegment(0,  **x**,  **y**, FX_MODE_COMET, RED, 1000, NO_OPTIONS);
  ws2812fx.start();
}

void loop() {
  loopi++;
  static int frameCount = 0;
  ws2812fx.service();

  if(ws2812fx.isFrame()) {
    frameCount++;
    if(frameCount == 150) {
      ws2812fx.stop();
    }
  }
 
  Serial.print(x);
  Serial.print(", ");
  Serial.print(y);
  Serial.print(", ");  
  Serial.println(y - x);
  if(frameCount == 6000){
    x = x + random(0,100);
    y = x + random(15,30);
    frameCount =0;
    ws2812fx.start();
  }
}

moose4lord, 2020-05-30:
To update the comet position you would need to execute the ws2812fx.setSegment() statement after updating the x and y values.

Also, you can't just keep adding a random amount to x and y, since they would eventually exceed the size of your LED strip.

There's also a problem with how you're restarting the animation after a delay. Counting frames works fine while an animation is running, but after you've called ws2812fx.stop() to stop the animation, you can't use isFrame() to start it up again. After executing ws2812fx.stop(), there's no animation running, so no frames to count. Rather than stopping the animation, it might be better to just use ```ws2812fx.setColor(BLACK);``` to make the comet color black (all LEDs off), so that the animation continues running, but the comet is hidden. Then use ```ws2812fx.setColor(RED);``` when you want to show it again.

Devilscave, 2020-06-01:
Of course I am aware of this and the random values are selected in a range in which I have LEDs, of course. The last sketch from me is as it is functional only that the X, Y values are not adopted. It is generally very difficult to combine different effects "simultaneously" with your Libaray, this would be a great thing if you could. If I add a static as background color to the Comet Mode and the Comet is stopped, the problem is that the static is stopped, so either everything or nothing lights up. Since then I was already in my previous overall sketch with other Libaray with Neopixel and Pixelpaointer. I just thought that since all of your effects in one it would be easier, unfortunately not.