kitesurfer1404/WS2812FX #204: WS2812FX ESP8266_Webinterface example gives error on compiling

JazekerXX, 2019-12-22:
Hi all,
i've been trying to compile the example sketch but it gives an error on line 226, 233 and 244. it says lvalue required as unary "&" operant. i just don't understand what that means. i have tried it on several pc's also without changing anything in the example. when i exclude the function void srv_handle_set() it compiles ok and webfunction works good (except for the set functions but we excluded it)   
[image]

moose4lord, 2019-12-22:
You're using an old version of the esp8266_webinterface sketch. Use the Library Manager to download the latest version of WS2812FX and use the version of the sketch in the examples folder.

JazekerXX, 2019-12-22:
Hi Moose4lord, thx for your reply! that was the solution :) i used the library provided with the Neopixel Xmas tree on Thingiverse. i really like this project and i'm gonna check out how it all works and use it for other projects too. so thanks big time!

closed 2019-12-22 by JazekerXX