kitesurfer1404/WS2812FX #206: ws2812 blink in random color - like noise

wchpikus, 2020-01-12:
i using esp68266 ale strip 8 led.
i tested it using  Adafruit_NeoPixel and works well.
when i use WS2812FX, i see random blinking on led depend of LED_COUNT.
i using pin13.
That is normal?

moose4lord, 2020-01-13:
That is not normal. What kind of ESP8266 are you using? When you say "pin13", do you mean GPIO13? Do you have any other hardware connected to the ESP8266?  Looks to me like a hardware problem, but posting your sketch might be helpful.

kitesurfer1404, 2020-01-13:
I had the same problem. Adding a 470 Ohm resistor in series in the data line solved the problem.
Also, see https://github.com/kitesurfer1404/WS2812FX/blob/master/extras/WS2812FX%20Users%20Guide.md#hardware-setup

wchpikus, 2020-01-13:
Hi
I used gpio13.
Resistor added, no changing.

i have in setup only init, start and in loop service.
in code in some function i changing parameters of this led. (ws2812fx.setSegment...)
so in any loop called service ws2812fx.service();

ESP8622, harware, one software serial, one hardware serial, led.
Without software serial also i had this issue.
But you see, only when esp is not connect (searching wifi) i have this..
in other status of wifi, i do not see any problems..sometimes blink once from time to time.

moose4lord, 2020-01-13:
So the problem occurs when WiFi is initializing? Did you try the esp8266_webinterface example sketch? That has a working WiFi implementation.

AustinSaintAubin, 2020-02-01:
You might consider using a logic level converter to convert the signal from the esp8266 from 3v to 5v into the LED string. I have had issues in the past with using 3v control of the LEDs, converting to 5v logic fixed that. https://learn.adafruit.com/neopixel-levelshifter

I used something similar to this.
[4-channel I2C-safe Bi-directional Logic Level Converter - BSS138](https://www.adafruit.com/product/757?gclid=Cj0KCQiAvc_xBRCYARIsAC5QT9kR61HkT5HxhZzf655xk2_pnhwFEFyCOqD47iOhdITNB-8N7UWtsHIaAoUVEALw_wcB
)

closed 2020-04-06 by moose4lord