kitesurfer1404/WS2812FX #207: Triggering different effects with different time

FotrLuk, 2020-01-20:
Hi, I'm not so adept at Arduino. I need advice. Would it be possible to run individual functions and give them a delay? Thank you

FotrLuk, 2020-01-21:
void loop() {
  now = millis();

LED.service();

  if((now >= 0) || (now <= 5000)) {
    LED.setMode(FX_MODE_RAINBOW_CYCLE);
    
  }

    if((now > 5000) || (now <= 10000)) {
    LED.setMode(FX_MODE_BLINK);
    
  }

  
}

moose4lord, 2020-01-21:
Although your logic is sound, that actually won't work. The setMode() function resets some of internal WS2812FX state variables, so calling it constantly in the loop() function will cause some of the effects to misbehave. You should be calling setMode() only when you want to actually change the effect.

You can use the setSegment() function to change the effects, instead of setMode(). Something like this:

```c++
void loop() {
  LED.service();
  
  long now = millis();
  now = now % 10000; // reset 'now' so the sequence loops

  if(now >= 0) {
    LED.setSegment(0, 0, LED_COUNT - 1, FX_MODE_RAINBOW_CYCLE, BLACK, 2000, NO_OPTIONS);
  }

  if(now >= 5000) {
    LED.setSegment(0, 0, LED_COUNT - 1, FX_MODE_BLINK, RED, 2000, NO_OPTIONS);
  }
}
```

FotrLuk, 2020-01-21:
Thank you very much, I thought using setMode I can call effects effects.

closed 2020-01-22 by FotrLuk