kitesurfer1404/WS2812FX #208: Wrong colors - MQTT

LAKY911, 2020-01-29:
Hi, 
I use thi lib with MQTT, via Home Assistant app I control the color that is sent via MQTT to my Arduino. Sometimes I get wrong color. In the following example - I set orange in the app, but LEDs are illuminating green. I checked that the color hex is correct, that means the problem must be somewhere else.
[image]

Can anyone tell me some hints, where can be the problem?
Thank you

slootsky, 2020-01-29:
I would check your primary colours first.  Red, Green, Blue. If any of those are wrong, then you have likely defined your LEDs to be of the wrong type.

LAKY911, 2020-01-29:
These colors are correct, but when I want for example some shade of orange, I get green. 
Strange.

gaijinsr, 2020-01-29:
Correct as in "I get red, green and blue" or as in "I get red when I select red, I get green when I select green, I get blue when I select blue"?

LAKY911, 2020-01-29:
Sorry :) The second option is what I meant. I get red when I select red, I get green when I select green, I get blue when I select blue.
But when I select orange, I get purple. When I select purple, I get light blue. 
Seems like I can't get some colors with "ws2812fx.setColor((uint32_t)number);"
thank you

moose4lord, 2020-01-29:
That's odd. What happens if you add this to your sketch?
```c++
ws2812fx.setColor(ORANGE);
Serial.print("\nColor is: 0x"); Serial.println(ws2812fx.getColor(), HEX);
```
Do your LEDs show orange and does the serial monitor display "Color is: FF3000"?

LAKY911, 2020-01-30:
Thank you, that helped. 
It was quite a hidden bug, at least for me.
Serial monitor showed "Color is: 0xFF30009". That was strange, because when I used the Serial.println(number, HEX);, I always got correct color - for example "FF3000", that is why I thought I set the right color. 
The bug was in conversion string to long, there was always a "9" at the end of the string, so the color I set to the RGB strip was "0xRRGGBB**9**".

Thank you all once again for solving this.

closed 2020-01-30 by LAKY911