kitesurfer1404/WS2812FX #210: Change Default Color to Random

haganwalker, 2020-02-18:
Hi there - I have some code that was working just fine, however any code that had one primary color always showed red. I was looking for some way to change this at random. Currently I select random effects by using an array with the effects I'd like to use and then using this in my loop: 
`ws2812fx.setMode(random(myModeCount));`
My thought was that I could do the same for choosing colors, but it doesn't seem to work by having an array like this:
`uint32_t colors[] = {RED, GREEN, BLUE, YELLOW, CYAN, MAGENTA, PURPLE, ORANGE, PINK};
const uint8_t myColors = (sizeof(colors)/sizeof(colors[0]));` 
and then calling that in the code, like this:
ws2812fx.setColor(random(myColors));
however, nothing displays for the single color effects like FX_MODE_STATIC. If a random effect that has multiple colors is selected though, like FX_MODE_RAINBOW, the animation works as expected. How can I choose a random color to use with the effects that utilize only one color, and change the color every time?

moose4lord, 2020-02-18:
You're setting the color to a random number. It looks like what you want to do is set the color to a color in your colors[] array, indexed by a random number.
```c++
ws2812fx.setColor(colors[random(myColors)]);
```

haganwalker, 2020-02-19:
Ahh - sorry, just being dumb. I appreciate that!

closed 2020-02-19 by haganwalker