kitesurfer1404/WS2812FX #211: Some bugs in WS2812FX Web frontend.

slackwaree, 2020-02-20:
Hello,

First of all thanks for the great library and sketch. Good to see an out of the box solution with web control.

I'm using this with a Wemos D1 mini. 

I'm not sure if the bufs are related to the FX library or the web sketch but here is what I have noticed.

Switching between the modes like:

color wipe inverse
color wipe reverse
color wipe reverse inverse
color wipe random

Can cause the ESP8266 to go offline for good, it is still running the color logic but no longer reachable.

Bug2:

Using the running lights for ~20 mins will reset the ESP8266. After reset the default program is always red staic lights.

Flash Sparkle: this doesn't do much

Also just a suggestion you could rerganize the html table to fit on one page instead of having to scroll down on multiple pages.

moose4lord, 2020-02-20:
Sorry you're having problems. I don't have any Wemos hardware, so I can't say if that's the problem, but I don't see the issues you mention on my NodeMCU.

I used to see the WiFi lockup occasionally (like your first bug), but the flakey WiFi seems to have been fixed after the SDK issues were resolved about a year ago ([https://github.com/esp8266/Arduino/issues/5784](https://github.com/esp8266/Arduino/issues/5784)). Are you using the latest ESP8266 core (v2.6.3)?

The default speed is a little too slow to effectively show the Flash Sparkle effect. Crank the speed up and you'll see the effect.

The example sketches are meant to be short demos really, not full fledged applications. Feel free to change them to suit your taste.

moose4lord, 2020-02-20:
I had another thought. Your comment about the ESP8266 resetting itself after a few minutes reminded me of a situation I encounter. I have a Mac mini connected to my test setup where my NodeMCU is powered by the Mac's USB port and the LEDs are powered by an external power supply. If the NodeMCU is running a sketch and the Mac goes to sleep, initially the sketch continues to run, because even while sleeping the Mac supplies power to its USB ports. However, after a while the Mac enters a deep sleep mode and shuts off its USB ports, and in the NodeMCU's death throes it resets, causing the LEDs to display the default static effect.

Just a thought.

slackwaree, 2020-02-20:
Hello,

First I was thinking on power supply issues as well that's why I double checked before opening a ticket about it with 2 power supplies. The second one can provide anywhere up to 5Amp, I was feeding the Wemos from the 5V pin + the leds and there is a resitor between the weemos and the ledstrip like in this guide:

https://www.instructables.com/id/ESP8266-controlling-Neopixel-LEDs-using-Arduino-ID/

This is an 5m strip with 300 leds draining up to 2A when it's full on however the reset issues even happen with the larson scan when it only drains like 0.20A so that's why I doubt it has anything to do with the PSU.

If you can recommend some other ESP32/8266 boards which are quite reliable please let me know their type I will order them but I want to run these led projects over very long term multiple years with turn on and offs. 

Thanks

moose4lord, 2020-02-20:
I have been quite happy with the Amica NodeMCU boards I've had for a couple years. They have an ESP-12E and a CP2102 USB/UART bridge (not the CH340 found on the Wemos D1 mini) and have been rock solid hardware-wise. If you use a breadboard, just make sure you get the newer, narrower version, since the original (very old) version of the NodeMCU was too wide to use in a breadboard.
You can find them on AliExpress/eBay/Amazon just by searching for "NodeMCU".

tobi01001, 2020-02-21:
> Hello,
>
> First I was thinking on power supply issues as well that's why I double checked before opening a ticket about it with 2 power supplies. The second one can provide anywhere up to 5Amp, I was [...]

Hi,

please be careful with your setup. 300 leds at 5V supply could drain (theoretically) up to 18 Amps (300 Leds * (0.02A [red] + 0.02 [green] + 0.02 [blue]))!

In reality its quite less but at full white you could expect up to 10 Amps.

I have two nodeMCU's running (old version) since I think almost two years without any issue. In addition I have 3 Wemos D1 mini running permanently and another one used for development.

I never had any HW issue with them (other than being caused by myself of course) and they are running that stable that I do not need any physical access at all (provided you do not mess up with the OTA :-D).

However, lately there was one issue I had with the application where I have a OLED display together with a rotary/button encoder for controlling the leds in addition to the webserver:
With the latest ESP8266 core I had sudden resets being caused by some (I think) bad interrupt handling. I reverted back to espressif8266@2.2.3 and the issues were gone.

One think which still brings some instability is when one of the devices looses the WiFi connection. But this I need to solve in the SW (especially for the Knob/Display-Control which run perfectly well even without WiFi).

Regards,
Toby

slackwaree, 2020-02-21:
Ok so you mean this model:
https://www.ebay.com/itm/Neu-NodeMCU-v3-2-ESP8266-Dev-Kit-WIFI-Lolin-Amica-CP2102-v2-Arduino-Gro%C3%9Fa-wsh/283758301869

Is your connecting of the strip is the same as it's described on instructables?

About the amps I know the exact value because I use lab psu which shows me. 

Also one more interesting thing about many of these LED strips that they don't show their correctly selected color on maximum brightness. Like it is red I set it to white but have to take down the brightness with 4-5 levels for them to turn actually to white.

Did you also try to interface with 12V strips?

moose4lord, 2020-02-21:
Yes, your eBay link shows the model of nodeMCU I use.

My connections are similar to the ones in the instructables article, but I include a 74HC125 Quad Buffer to do the level shifting between the 3.3V ESP8266 GPIO pin and the 5V WS2812 Din pin. This is similar to the level shifting arrangement described in the WS2812FX User Guide.

I have not seen the situation you describe about having to turn the brightness down to show white. That sounds like a power supply problem to me. Displaying white takes a lot of power. Maybe your power supply isn't as good as you think it is. Are you injecting power at both ends of the LED strip? That may help.

I have not tried 12V LEDs, so can't really say if they're any better or not.

closed 2021-03-01 by moose4lord