kitesurfer1404/WS2812FX #212: Custom Effects and Multiple zones

henmel, 2020-02-21:
I am just starting out and have been playing around with this great library. I am running into an issue with  using custom effects with multiple zones. I am using NodeMCU and have setup 2 test zones, ws2812fx and ws2812fx_z2, using  my skit and they both work well with built in effects, but when when it comes to custom effects, second zone does not work. I was able to add a custom effect using the examples, provided with the library, and it works fine with the default zone, but the second zone does not work, only for the custom effect. Looking at the sample code for TwinkleFox, it appears that it is hardcoded for ws2812fx segment, unlike the built in effects in the WS2812FX library. Is there a way to add/modify the custom effects so that they take the SEGMENT information, as the built in effects parameter.

moose4lord, 2020-02-22:
I'm not sure what you mean by "zone", but it sounds like you have two separate strips of LEDs, each being driven by it's own GPIO pin, and so have two WS2812FX instances in your sketch. You are correct, the custom effects are all written with only one instance in mind and assumes the instance is called "ws2812fx". To run the custom effect on a second instance you'd need to copy the custom effect code and change the lines that reference "ws2812fx" to instead reference "ws2812fx_z2".
Something like this:

```c++
#include <WS2812FX.h>

WS2812FX ws2812fx = WS2812FX(30, 4, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_z2 = WS2812FX(30, 5, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);

  // init zone 1
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setCustomMode(myCustomEffect);
  ws2812fx.setSegment(0, 0, 29, FX_MODE_CUSTOM, RED, 100, false);

  // init zone 2
  ws2812fx_z2.init();
  ws2812fx_z2.setBrightness(255);
  ws2812fx_z2.setCustomMode(myCustomEffect_z2);
  ws2812fx_z2.setSegment(0, 0, 29, FX_MODE_CUSTOM, RED, 100, false);

  // start each zone
  ws2812fx.start();
  ws2812fx_z2.start();
}

void loop() {
  // service each zone
  ws2812fx.service();
  ws2812fx_z2.service();
}

uint16_t myCustomEffect(void) { // random chase custom effect for zone 1
  WS2812FX::Segment* seg = ws2812fx.getSegment(); // get the current segment
  for(uint16_t i=seg->stop; i>seg->start; i--) {
    ws2812fx.setPixelColor(i, ws2812fx.getPixelColor(i-1));
  }
  ws2812fx.setPixelColor(seg->start, ws2812fx.color_wheel(ws2812fx.random8()));
  return seg->speed;
}

uint16_t myCustomEffect_z2(void) { // random chase custom effect for zone 2
  WS2812FX::Segment* seg = ws2812fx_z2.getSegment(); // get the current segment
  for(uint16_t i=seg->stop; i>seg->start; i--) {
    ws2812fx_z2.setPixelColor(i, ws2812fx_z2.getPixelColor(i-1));
  }
  ws2812fx_z2.setPixelColor(seg->start, ws2812fx_z2.color_wheel(ws2812fx_z2.random8()));
  return seg->speed;
}
```

henmel, 2020-02-22:
Thanks for a quick reply, but I don't think this will work for my situation. You are correct. I have setup multiple zones using two GPIO pins and I have modified your esp8266_webinterface example, adding the second zone, similar to what you are suggesting. I have also added 4 custom effects using the example ws2812fx_custom_effect2, and the included sample effects. The sketch works great for both zones and the effects are the in sync, same for each zone, until I try to use the 4 custom effects, which only work on the first zone. I don't see how, your suggestion of having two copies of the effect, one for each zone, will work, unless I am missing something. Is there any way to pass the segment information to the custom effect, like is done in the main library? I was actually hoping this can be done, since the ultimate plan was to drive 3-4 zones, using different GPIOs.

moose4lord, 2020-02-22:
I'm afraid passing the "segment information" to the custom effect will do you no good. Segments partition a zone into multiple, independent sections, each with its own animation config. That's a little different from your multi-zone setup.

I realize duplicating the custom effect code for each zone is far from elegant, with lots of duplicate code, but it will work. What makes you think it will not?

Also, the esp8266_webinterface example sketch is really only setup for one zone. You'll see references to "ws2812fx" peppered throughout the code that updates the LEDs based on the web interface (the srv_handle_set() function). How are you switching between zones in the web interface? I guess I would need you to share your sketch to get a better understanding of what's going on.

henmel, 2020-02-22:
Thank you. Maybe I have not explaining myself well. In the Web example I basically did just what you suggested initially. For each instance of change in mode, speed, brightness or color, I have a segment instance, like this:
    ws2812fx.setSegment(0, 0, LED_COUNT-1, next_mode, colors, speed, options);
    ws2812fx_z2.setSegment(0, 0, LED2_COUNT-1, next_mode, colors, speed, options);

This is both inside the srv_handle_set() function and the auto_cycle in the void loop. This works well for built in effects, but not the custom added ones. When I said "segment information", I really meant the strip(zone) info, in the form of ws2812fx. and ws2812fx_z2.

My code has gotten a bit long, since I have changed the web interface and the sketch to handle 3 colors, added option control, MQTT,  OTA and more. I have been using it to learn the programming by examples. The sketch is now about 1500 lines.

Since it sounds like the custom effect functions don't seem to work the same way as the ones built into the library, I will try your suggestion, setting up a copy of each effect, specific to each zone. Will I still able to add the 4 custom effects, for each zone?

tobi01001, 2020-02-22:
Do you actually need / want to run different effects on your two instances
or do you want them to show the same anyway? I followed your discussion but
still not clear about your needs...

The latter would be easier but if you need different effects effects than
you may have to do as you described.

henmel, 2020-02-22:
For now the plan is to have the same effect for each instance(zone). Really the only difference between each is number of LEDs and the PIN. 

This will hopefully be installed on the roof line of a two story house so I was planning on having the same effect on each and if this works adding more zones down the line. It would be nice to have an ability for different effects on each zone, bot for now not necessary.

henmel, 2020-02-22:
I just tried your original suggestion, but it does not appear to work. Here is a copy of the two relevant sections. void setup:

`
  uint8_t dualLarsonMode  = ws2812fx.setCustomMode(F("Dual Larson"), dualLarson);
  uint8_t dualLarson_z2Mode  = ws2812fx_z2.setCustomMode(F("Dual Larson"), dualLarson_z2);

`
effect functions:
`
uint16_t dualLarson(void) {
  WS2812FX::Segment* seg = ws2812fx.getSegment(); // get the current segment
  WS2812FX::Segment_runtime* segrt = ws2812fx.getSegmentRuntime();
  int seglen = seg->stop - seg->start + 1;

  static int16_t offset = 0;
  int8_t dir = segrt->aux_param ? -1 : 1;
  segrt->aux_param3 += dir;

  ws2812fx.fade_out();

  int16_t segIndex = (segrt->aux_param3 + offset) % seglen;
  ws2812fx.setPixelColor(seg->start + segIndex, seg->colors[0]);
  ws2812fx.setPixelColor(seg->stop  - segIndex, seg->colors[2]);

  if(segrt->aux_param3 >= (seg->stop - seg->start) || segrt->aux_param3 <= 0) {
    segrt->aux_param = !segrt->aux_param;
    if(seg->options & REVERSE) offset = (offset + 1) % seglen;
  }

  return (seg->speed / (seglen * 2));
}

uint16_t dualLarson_z2(void) {
  WS2812FX::Segment* seg = ws2812fx_z2.getSegment(); // get the current segment
  WS2812FX::Segment_runtime* segrt = ws2812fx_z2.getSegmentRuntime();
  int seglen = seg->stop - seg->start + 1;

  static int16_t offset = 0;
  int8_t dir = segrt->aux_param ? -1 : 1;
  segrt->aux_param3 += dir;

  ws2812fx_z2.fade_out();

  int16_t segIndex = (segrt->aux_param3 + offset) % seglen;
  ws2812fx_z2.setPixelColor(seg->start + segIndex, seg->colors[0]);
  ws2812fx_z2.setPixelColor(seg->stop  - segIndex, seg->colors[2]);

  if(segrt->aux_param3 >= (seg->stop - seg->start) || segrt->aux_param3 <= 0) {
    segrt->aux_param = !segrt->aux_param;
    if(seg->options & REVERSE) offset = (offset + 1) % seglen;
  }

  return (seg->speed / (seglen * 2));
}
`

I do get the LEDs lighting, but the effect is not working for the second instance.

henmel, 2020-02-22:
I just realized that adding the above code actually inserted two custom effects, both Dual Larson. It was not added for just the second instance(zone). They both show up in the Web interface part now.

moose4lord, 2020-02-22:
Ah, you have uncovered a bug. When using this form of the setCustomMode() function:
```c++
ws2812fx.setCustomMode(F("Dual Larson"), dualLarson);
```
the WS2812FX code tries to keep track of which of the four custom mode "slots" you're programming with this static variable:
```c++
static uint8_t custom_mode_index = 0; // line 1607 in WS2812FX.cpp
```
Unfortunately this does not work when using multiple instances of WS2812FX, since the static custom_mode_index variable is shared across all instances of WS2812FX. :(

As a workaround you can use another form of the setCustomMode() function which explicitly specifies the custom effect "slot" you're assigning:
```c++
// assign dualLarson to custom effect slot 0
ws2812fx.setCustomMode(0, F("Dual Larson"), dualLarson);
```

Attached is my test code.
[esp8266_webinterface_zones.zip](https://github.com/kitesurfer1404/WS2812FX/files/4239616/esp8266_webinterface_zones.zip)

henmel, 2020-02-22:
Thanks Keith! You are the greatest!

It works, I just moved   modes_setup();  to be after the custom mode additions and it shows the names in the Web interface. I also, tested by adding all 4 custom modes and they also work for both zones. 

Earlier you said:

> Do you actually need / want to run different effects on your two instances
or do you want them to show the same anyway? I followed your discussion but
still not clear about your needs...

> The latter would be easier but if you need different effects effects than you may have to do as you described.

Is there an easier way to do this if I use the same effect for all zones? It would be nice if I did not have to duplicate the code for each effect, especially if I end up adding more zones...

tobi01001, 2020-02-22:
That was me...

Anyway, if each would have the same length (or quite the same) the. I would
just calculate the effect with just one ws2812fx instance (the longest) and
then just create another neopixel instance for the other.

Before writing the data you could memcpy from the ws2812fx instance to the
other neopixel without the need to actually calculate.

This could probably be done in a custom show function.

For my applications I do this a bit differently.

I have e.g a long strip and can set the number of segments which is then
equally distributed. Effects are only calculated for one segment and then
copied to the others (with reverse / mirroring etc).

You could just do the same with copying the data from the one ws2812fx to
the other (neopixel instance). This way they would also show exactly the
same.

henmel, 2020-02-22:
Your ideas sound really interesting, but I don't know how to do the things you are suggesting. I am more of a beginner with Arduino code. Most of my programming is in HTML, Javascript and some other server side languages. Everything I have learned so far was from examples included in Arduino and the web.

My two initial roof lines(zones) are about 60ft and 40ft, so per your suggestion, I could break each zone into segments 20 ft each, then copy the 20ft segment to  the other two segments in zone 1 and the 2 segments in zone 2, or something like that.

Would you have some examples on how to do things like creating the instance, copying segments from one to the other, applying reverse/mirroring, etc..? I would love to learn how to do those.

Thanks again,
Henry

tobi01001, 2020-02-23:
Hi,

happy to give you some directions. Unfortunately it's been a while since I used the "original" ws2812fx with the adafruit neopixel library.....

For the approach I have in mind, you need one of these and one adafruit neopixel object for the second one (would not hurt to have this as ws2812fx as well).

If you could share your sketch it may be easier to give you some directions.

Out of the blue its currently a bit fuzzy for me :-)

regards,
Toby

moose4lord, 2020-02-23:
Tobi has a fine idea. WS2812FX's copyPixels() function and the Heartbeat.h custom effect use memmove to copy pixel data from one spot on a strip to another, if you want to look at those for inspiration.

I played around with the idea of calculating pixel data for one segment and then copying that data to all of the zones. It works, but ultimately I wasn't happy with the results. I didn't like the way it forces splitting your strips into segments. I was using a 144 LED strip and a 30 LED strip, so there was quite a bit of difference between the lengths of the two zones, and to me, it was just too many segments. You may have better luck if the sizes of your zones aren't so dissimilar.

I've attached my test code, if you want to play with the idea.
[esp8266_webinterface_zone2.zip](https://github.com/kitesurfer1404/WS2812FX/files/4241272/esp8266_webinterface_zone2.zip)

tobi01001, 2020-02-23:
Yes, thats what I basically had in mind for this type of setup.

Glad you already prepared such example. 👍 

I also played with copying the content of a e.g. 100 leds segment frame onto a 50 leds segment with calculated partial pixels but it was never that satisfying...

So now for my use, I do divide the strip into x equal segments. The effect is only calculated on the first segment and then copied to the others.

I realized that this soemtimes looks a bit strange (depending on the effect) and therefore also added a reverse and mirror parameter.

Reverse just reverses the first segment (found that handling easier than reversing in the effect calculation itself).

Mirror just reverses every odd segment which makes the segments looking very good as the start always fits to the end...

**Anyway**, since you copy every calculated frame in the "copyPixelData" function and also write the leds in there, wouldn't it be better to set an empty customShow function to avoid additional writing time in service?

Or even more easy (what I had in mind) do the pixel copy stuff in the customShow function?

Regards,
Tobi

moose4lord, 2020-02-24:
You're right, there's an unnecessary call to show() for the first zone in the service() function. Implementing a custom show function that does nothing would fix that.

henmel, 2020-02-24:
Thanks again for your help. I will take a closer look at the example copy to see what I can learn from it. My sketch is a bit long, but based on the original esp8266_webinterface and builds on it. I simply modified it to have another GPIO output for the second zone and duplicated the code for each request for mode change, brightness, speed, etc.. As I indicated above, Keith's 

> Attached is my test code.
esp8266_webinterface_zones.zip

is very similar to what I currently have.  Your idea of breaking each zone into multiple, same length segments sounded very interesting, but I have no experience with doing that and was hoping you had an example. The sketch is already using both the WS2812FX and adafruit neopixel library, as I understand.

Looking more closely at my zone lengths, I can break them up into 9 and 6 even segments each, so if I can figure out how to take the 1st segment and copy/mirror it to each of the other segments in both zones, I think would work great. Not sure if this can be done or how to do it.

Thanks again for a great library and all of the help. Much appreciated.

henmel, 2020-02-24:
Wow! Your 

> I've attached my test code, if you want to play with the idea.
esp8266_webinterface_zone2.zip

is awesome. I did not realize that this can be done like that. It really simplifies a lot of the coding that was needed to do it the other way. I do have a couple of questions though:

- Since this is now done by copying the segment data, I am assuming it is no longer necessary to create an alternate version of the custom effects for zone 2?

- Using this technique, will WS2812FX will still see each zone as multiple segments? The reason I ask is that I am saving zone1 segment data to EEPROM, similar to ws2812fx_segments_web example, will that still work, or do I need to change it to just save the 1 segment that we are making a copy of?

- Regarding Tobi's suggestion of using a mirror or reverse for the segments, how can that be done?

-  You guys mentioned that some adjustments to the code may make it even better, 

> You're right, there's an unnecessary call to show() for the first zone in the service() function. Implementing a custom show function that does nothing would fix that. Or implementing the pixel copying in a custom show function is a good idea too.

Should I try to implement those or use as you have in the example?

Once again, thanks for all of your help. You guys Rock! 

Henry

moose4lord, 2020-02-24:
> Since this is now done by copying the segment data, I am assuming it is no longer necessary to create an alternate version of the custom effects for zone 2?

Correct

> Using this technique, will WS2812FX will still see each zone as multiple segments? The reason I ask is that I am saving zone1 segment data to EEPROM, similar to ws2812fx_segments_web example, will that still work, or do I need to change it to just save the 1 segment that we are making a copy of?

I suppose, technically, you could use this technique with multiple segments in the first zone. It would probably get confusing though. :) You'd need to adjust the ```#define SEGMENT_SIZE 30``` statement so SEGMENT_SIZE is the sum of the number of LEDs in all of your segments. The EEPROM code should work OK as is.

> Regarding Tobi's suggestion of using a mirror or reverse for the segments, how can that be done?

Instead of using memmove to do a straight copy of the pixel data, you could use a for-loop to change the order of the pixels as they're being copied. Something along these lines:
```c++
  for(uint16_t i=0; i < SEGMENT_SIZE; i++) {
    uint32_t color = ws2812fx.getPixelColor(i);
    ws2812fx_z2.setPixelColor(i, color); // forward copy
    ws2812fx_z2.setPixelColor(SEGMENT_SIZE - i - 1, color); // reverse copy
  }
``` 

> You guys mentioned that some adjustments to the code may make it even better

Tobi was just mentioning that when ws2812fx.service() is executed, the show() function updates the LEDs for zone 1, and in copyPixelData() the show() function is executed again for zone 1. The execution of show() in the service() function is unnecessary and you could inhibit it by creating a customShow() function. It's a bit of an advanced topic, but if you want to give it a try, you can create a customShow() function for zone 1 with this line in the setup() function:
```c++
ws2812fx.setCustomShow([]() { return; }); // disable the default show() function
```
and change the ```show()``` statements in copyPixelData() to these lines:
```c++
ws2812fx.Adafruit_NeoPixel::show(); // reshow zone 1
.
.
.
ws2812fx_z2.Adafruit_NeoPixel::show(); // reshow zone 2
```
But I wouldn't worry about it really. It's just an optimization.

henmel, 2020-02-25:
Thanks Keith,

I just tried to modify the copyPixelData, per your suggestion and am not sure if I did not incorporate it correctly or maybe I did not explain what I was trying to do. 

> Mirror just reverses every odd segment which makes the segments looking very good as the start always fits to the end...

Per Tobi's suggestion.

What I ended up with is a mirror within each segment, making it look like double number of segments,  instead of reversing each alternating segment. In other words if the first segment is forward direction, second one will be reverse and alternate like that. So on the even segments it would be forward and on odd ones reverse. Here is what I ended up with for the copyPixelData function:

```
void copyPixelData() {
    uint16_t numBytesPerPixel = ws2812fx.getNumBytesPerPixel();
    uint16_t numBytesPerSegment = SEGMENT_SIZE * numBytesPerPixel;
 
    // copy zone 1's first segment data to all of zone 1
    for(int i=0; i<=ws2812fx.numPixels() - SEGMENT_SIZE; i+=SEGMENT_SIZE) {
      memmove(ws2812fx.getPixels() + (i * numBytesPerPixel), ws2812fx.getPixels(), numBytesPerSegment);
      for(uint16_t i=0; i < SEGMENT_SIZE; i++) {
        uint32_t color = ws2812fx.getPixelColor(i);
        ws2812fx.setPixelColor(i, color); // forward copy
        ws2812fx.setPixelColor(SEGMENT_SIZE - i - 1, color); // reverse copy
      }
    }
    ws2812fx.show(); // reshow zone 1
    yield();

  
    // copy zone 1's first segment data to all of zone 2
    for(int i=0; i<=ws2812fx_z2.numPixels() - SEGMENT_SIZE; i+=SEGMENT_SIZE) {
      memmove(ws2812fx_z2.getPixels() + (i * numBytesPerPixel), ws2812fx.getPixels(), numBytesPerSegment);
      for(uint16_t i=0; i < SEGMENT_SIZE; i++) {
        uint32_t color = ws2812fx.getPixelColor(i);
        ws2812fx_z2.setPixelColor(i, color); // forward copy
        ws2812fx_z2.setPixelColor(SEGMENT_SIZE - i - 1, color); // reverse copy
      }
    }
    ws2812fx_z2.show(); // show zone 2
    yield();
  }

```

Thanks again!
Henry

tobi01001, 2020-02-25:
Hi Henry,

I could not judge how this would look. It should actually not complie as you are using another `int i` inside the first for loop.

Anyway, memmove is fast and straight forward to copy complete memory areas. when it comes to reverse or mirror things, you could do the more easy readable way (which needs more computing cycles but performance is not your enemy as you are using an ESP anyway ;-) ).

So what I would do is the following (depending on your physical arrangement): 
Lets consider you have divided the first stripe into 6 segments and 3 on the second stripe:

```C
#define NUMSEGMENTS1 6 // make sure you do not exceed the number of leds defined for the WS2812FX instance
#define NUMSEGMENTS2 3 // make sure you do not exceed the number of leds defined for the WS2812FX instance

void copyPixelData() {

  for (uint16_t i=0; i<NUMSEGMENTS1; i++)
  {
    for(uint16_t j=0; j<SEGMENT_SIZE; j++)
    {
      if(i & 0x01) // if the segment is a odd number
      {
        ws2812fx.setPixelColor(j + (SEGMENT_SIZE * i)) = ws2812fx.getPixelColor(SEGMENT_SIZE-1-j);
      } 
      else 
      { // even segments just get copied
        ws2812fx.setPixelColor(j + (SEGMENT_SIZE * i)) = ws2812fx.getPixelColor(j);
      } 
    }
  }
   // Repeat the same for the second strip:
  for (uint16_t i=0; i<NUMSEGMENTS2; i++)
  {
    for(uint16_t j=0; j<SEGMENT_SIZE; j++)
    {
      if(i & 0x01) // if the segment is a odd number
      {
        ws2812fx_z2.setPixelColor(j + (SEGMENT_SIZE * i)) = ws2812fx.getPixelColor(SEGMENT_SIZE-1-j);
      } 
      else 
      { // even segments just get copied
        ws2812fx_z2.setPixelColor(j + (SEGMENT_SIZE * i)) = ws2812fx.getPixelColor(j);
      } 
    }
  }
  ws2812fx.show(); // reshow zone 1
  yield(); // Don't know if we really need to yield here but does probably not hurt either...
  ws2812fx_z2.show(); // reshow zone 1
  yield(); // Don't know if we really need to yield here but does probably not hurt either...
}
```
OKAY, that' been two nested for loops doing basically the same thing... We could simplify this a bit already:
```C
#define NUMSEGMENTS1 6 // make sure you do not exceed the number of leds defined for the WS2812FX instance
#define NUMSEGMENTS2 3 // make sure you do not exceed the number of leds defined for the WS2812FX instance

void copyPixelData() {

// provided NUMSEGMENTS2 <= NUMSEGMENTS1 (one could ensure this with precompiler #if or during runtime)
// we could do the following: 

  for (uint16_t i = 0; i < NUMSEGMENTS1; i++)
  {
    for(uint16_t j = 0; j < SEGMENT_SIZE; j++)
    {
      if(i & 0x01) // if the segment is a odd number
      {
		if(i<NUMSEGMENTS1)
			ws2812fx.setPixelColor(j + (SEGMENT_SIZE * i)) = ws2812fx.getPixelColor(SEGMENT_SIZE-1-j);
		if(i<NUMSEGMENTS2)
		    ws2812fx_z2.setPixelColor(j + (SEGMENT_SIZE * i)) = ws2812fx.getPixelColor(SEGMENT_SIZE-1-j);	
      } 
      else 
      { // even segments just get copied
		if(i < NUMSEGMENTS1 && i > 0) //stay within bounds and do not copy the first segment because its the source
			ws2812fx.setPixelColor(j + (SEGMENT_SIZE * i)) = ws2812fx.getPixelColor(j);
		if(i < NUMSEGMENTS2) // stay within bounds...
			ws2812fx_z2.setPixelColor(j + (SEGMENT_SIZE * i)) = ws2812fx.getPixelColor(j);
      } 
    }
  }
  ws2812fx.show(); // reshow zone 1
  yield(); // Don't know if we really need to yield here but does probably not hurt either...
  ws2812fx_z2.show(); // reshow zone 1
  yield(); // Don't know if we really need to yield here but does probably not hurt either...
}
```

Everything from memory and untested but should give you an idea what happens.

Regards
Toby

moose4lord, 2020-02-25:
Hi Henry, sorry if I wasn't clear. My little for-loop code snippet was meant to replace the memmove statement in my "zone2" test code, but your code still has the memmove statement. Also my snippet showed both a statement for forward copy and a statement for reverse copy, but I meant that one or the other should be used depending on if you wanted a forward copy or a reverse copy of the segment.

Toby has the right idea, although I think I would take a slightly different tack. I think I would pre-calculate the reverse pixel data and then could still use the very efficient memmove statement to copy the pixel data, just choosing between the original data or the pre-calculated reverse data depending on the segment. This is what I came up with:
```c++
// copy pixel data from the first segment of zone 1 to all zones
uint8_t reversePixelData[SEGMENT_SIZE * 3]; // assumes RGB LEDs, multiply by 4 for RGBW LEDs
void copyPixelData() {
    uint8_t numBytesPerPixel = ws2812fx.getNumBytesPerPixel();
    uint16_t numBytesPerSegment = SEGMENT_SIZE * numBytesPerPixel;

    // pre-calculate the reverse pixel data for zone 1 segment 0
    uint8_t *forwardPtr = ws2812fx.getPixels();
    uint8_t *reversePtr = reversePixelData + ((SEGMENT_SIZE - 1) * numBytesPerPixel);
    for(uint16_t i=0; i < numBytesPerSegment; i+=numBytesPerPixel) {
      memmove(reversePtr - i, forwardPtr + i, numBytesPerPixel);
    }
  
    // copy zone 1 segment 0 data to all of zone 1
    bool isEvenSegment = true;
    for(int i=0; i<=ws2812fx.numPixels() - SEGMENT_SIZE; i+=SEGMENT_SIZE, isEvenSegment = !isEvenSegment) {
      if(isEvenSegment) { // even segment, forward copy
        memmove(ws2812fx.getPixels() + (i * numBytesPerPixel), ws2812fx.getPixels(), numBytesPerSegment);
      } else { // odd segment, reverse copy
        memmove(ws2812fx.getPixels() + (i * numBytesPerPixel), reversePixelData, numBytesPerSegment);
      }
    }
    ws2812fx.Adafruit_NeoPixel::show(); // reshow zone 1
    yield();

    // copy zone 1 segment 0 data to all of zone 2
    isEvenSegment = true;
    for(int i=0; i<=ws2812fx_z2.numPixels() - SEGMENT_SIZE; i+=SEGMENT_SIZE, isEvenSegment = !isEvenSegment) {
      if(isEvenSegment) { // even segment, forward copy
        memmove(ws2812fx_z2.getPixels() + (i * numBytesPerPixel), ws2812fx.getPixels(), numBytesPerSegment);
      } else { // odd segment, reverse copy
        memmove(ws2812fx_z2.getPixels() + (i * numBytesPerPixel), reversePixelData, numBytesPerSegment);
      }
    }
    ws2812fx_z2.Adafruit_NeoPixel::show(); // reshow zone 2
    yield();
  }
```

tobi01001, 2020-02-25:
Oh well, this is by far more efficient, yes. 
May tinker with something simliar that for my own application.
I already work with a complete copy of LED data (one to calculate the effect and one to actually write them) because this gives the possibility to easily fade (between the effects, on/off etc) but I did not yet think about pre-copying the reverse/mirrored effect and using memmove to distribute the data. 

@henmel Thanks for the discussion. Hope all this will help you after all.
@moose4lord Thanks for the new ideas. 👍 

One thing, since you already have a normal and mirrored segment you could still avoid the second (third) for loop:
```C
// copy pixel data from the first segment of zone 1 to all zones
uint8_t reversePixelData[SEGMENT_SIZE * 3]; // assumes RGB LEDs, multiply by 4 for RGBW LEDs
void copyPixelData() {
    uint8_t numBytesPerPixel = ws2812fx.getNumBytesPerPixel();
    uint16_t numBytesPerSegment = SEGMENT_SIZE * numBytesPerPixel;

    // pre-calculate the reverse pixel data for zone 1 segment 0
    uint8_t *forwardPtr = ws2812fx.getPixels();
    uint8_t *reversePtr = reversePixelData + ((SEGMENT_SIZE - 1) * numBytesPerPixel);
    for(uint16_t i=0; i < numBytesPerSegment; i+=numBytesPerPixel) {
      memmove(reversePtr - i, forwardPtr + i, numBytesPerPixel);
    }
  
    // copy zone 1 segment 0 data to all of zone 1
    bool isEvenSegment = true;
    for(int i=0; i<=ws2812fx.numPixels() - SEGMENT_SIZE; i+=SEGMENT_SIZE, isEvenSegment = !isEvenSegment) {
      if(isEvenSegment) { // even segment, forward copy
        memmove(ws2812fx.getPixels() + (i * numBytesPerPixel), ws2812fx.getPixels(), numBytesPerSegment);
        if(i<=ws2812fx_z2.numPixels() - SEGMENT_SIZE) {
			memmove(ws2812fx_z2.getPixels() + (i * numBytesPerPixel), ws2812fx.getPixels(), numBytesPerSegment);
		}
      } else { // odd segment, reverse copy
        memmove(ws2812fx.getPixels() + (i * numBytesPerPixel), reversePixelData, numBytesPerSegment);
		if(i<=ws2812fx_z2.numPixels() - SEGMENT_SIZE) {
			memmove(ws2812fx_z2.getPixels() + (i * numBytesPerPixel), reversePixelData, numBytesPerSegment);
		}
      }
    }
    ws2812fx.Adafruit_NeoPixel::show(); // reshow zone 1
    yield();
    ws2812fx_z2.Adafruit_NeoPixel::show(); // reshow zone 2
    yield();
  }
```

One drawback in all of this is that it makes the segments static in their application. But this may just be fine depending on the intended use.

henmel, 2020-02-25:
Both versions work great. Just what I have been looking for. This has been very helpful and your help is much appreciated. This should provide the flexibility I was looking for, especially when I am ready to add more zones. It should also make the sketch a lot more efficient and much smaller.

Again Many Thanks for all of your help! You guys Rock!
Regards,
Henry

moose4lord, 2020-02-25:
Just one caveat. Toby's refactoring would not work if zone 2 has more segments than zone 1. In that case, the for-loop to process ws2812fx would terminate before all of ws2812fx_z2's segments were processed.

Ah, those unintended consequences will get you every time. :)

closed 2020-02-25 by moose4lord

henmel, 2020-02-25:
Sorry, I have a follow up question. Not sure if i should start a new thread. I have attached a revised version of the sketch with my version of the front end and some new controls. One of the controls added was the On/Off switch, which worked fine in the original segment driven version, but in this version that copies the segments, the Off switch stops the animation, but does not turn off(clear) the LED's
[esp8266_webinterface_zone2_copysegmentdata.zip](https://github.com/kitesurfer1404/WS2812FX/files/4251412/esp8266_webinterface_zone2_copysegmentdata.zip)
. Not sure why this is. Any ideas?

By the way, feel free to use a version of the front end, if you would like.

moose4lord, 2020-02-25:
Henry, there are two problems.

The first problem is in the stop() function used to stop the animation and reset the LEDs. This is the code snippet (near line 305):
```c++
if(server.argName(i) == "z") {
      if(server.arg(i)[0] == '-') {
        system_state = false;
        ws2812fx.stop(); // add this to stop and clear the animation
```
```ws2812fx.stop();```  turns off the "running" flag to stop the animation and calls strip_off() to clear the LEDs. strip_off() calls clear() to set all the LED data to zero and then calls show() to push that data out to the LEDs. Unfortunately, when ```ws2812fx.setCustomShow([]() { return; });``` was added to inhibit the call to show() in service(), it also inhibited the show() that pushes the cleared data out to the LEDs. Bah! Again with the unintended consequences! :)

So you need to replace ```ws2812fx.stop();``` with your own code. Maybe this:
```c++
ws2812fx.pause(); // sets the running flag to false to stop animation
ws2812fx.Adafruit_NeoPixel::clear(); // set LED data to zeros
ws2812fx.Adafruit_NeoPixel::show(); // push the data out to the LEDs
```

The second problem is that the stop() function is only called for zone 1 (ws2812fx). You'll need to repeat this code for zone 2 (ws2812fx_z2).

henmel, 2020-02-25:
Thank you. I was thinking it was something like that, since I also tried the Clear functions and it did not work either. At least now I know that what caused it and how to get around it. Would restoring the original, unoptimized version of ws2812fx.show(); call and removing the custom show one be a good option too? I use the ws2812fx.clear(); function in the autocycle, to clear between the animations, in my main sketch so it may make it easier to use the built in stop and other functions.

Regarding zone 2, I was just doing the testing for now on zone 1, since I know the code works for the second zone. I have a lot of changes to make to the main sketch, that I have been working on, so I wanted to make sure the bits and pieces were going to work first.

Thank you very much!
Henry

moose4lord, 2020-02-25:
Probably the best solution is to incorporate the copyPixelData() function into a custom show function. Toby had mentioned this way back at the beginning of this discussion. You can read about custom show functions in the WS2812FX User Guide and there's a simple demo in the ws2812fx_dma example sketch.  Give it a try.

henmel, 2020-02-26:
Thanks Keith,

I moved the copyPixelData function call to the myCustomShow and replaced the Stop function call, as you suggested. Everything appears to still be working correctly. Please take a look:
```
void myCustomShow(void) {
  if(ws2812fx.isFrame(0)) {
    copyPixelData();
    return;
  }
  //ws2812fx.show(); 
}

```
```
        ws2812fx.pause(); // sets the running flag to false to stop animation
        ws2812fx.Adafruit_NeoPixel::clear(); // set LED data to zeros
        copyPixelData(); // push the data out to the LEDs (all zones)        ws2812fx.clear();

```
I tried to call the default ws2812fx.show function if isFrame was not triggered, as a fallback, but that caused the MCU to keep resetting, so I commented that out. I guess that can't be done if we use the customshow function.

Thanks again,
Henry

tobi01001, 2020-02-26:
Hi,

 I would actually skip the `if(ws2812fx.isFrame(0))` part:
> ```C
> void myCustomShow(void) {
>   //if(ws2812fx.isFrame(0)) {
>     copyPixelData();
>     return;
>   //}
>   //ws2812fx.show(); 
> }
> ```
Provided you only have one segment to be calculated during ws2812fx.service(), the `service `function should do the trick and the `copyPixelData()` function will distribute the segment information (LED data) to the other "zones".

This way, you should also be able to directly set the `copyPixelData()` as "`customShow`" in setup:
```C
ws2812fx.setCustomShow(copyPixelData); // set copyPixelData as show function and disable the default
```

in loop you just remove `isFrame` check:
```C
// if a new animation frame has been drawn for the first segment of zone 1,
// then copy the pixel data to all other zones
//  if(ws2812fx.isFrame(0)) {
//    copyPixelData();
//  }
 ```

What happens now: Only one segment is set and this frame gets calculated in `ws2812fx.service`. The service function calls `show` whenever needed. The `show` function then calls the customShow which is effectively the `copyPixelData` and so the LED data from segment one in ws2812fx gets copied to the other segments including ws2812fx_z2. Further, it calls the show functions for both strips which then writes the data.

This is what you should have achieved right now. But this should also work for stop, pause and other functions being applied to the first segment in ws2812fx and then being copied to the other zones. So you could remove your application specific calls and revert back to the originals:

> ```C
>         ws2812fx.pause(); // sets the running flag to false to stop animation
>         ws2812fx.Adafruit_NeoPixel::clear(); // set LED data to zeros
>         copyPixelData(); // push the data out to the LEDs (all zones)        ws2812fx.clear();
> ```

A call to `ws2812fx.stop()` will now clear the data from the entire strip ws2812fx and then call `show` which calls `copyPixelData` as `customShow`. The cleared data from strip 1 (segment 0) is then copied to the other zones and finally written out....

> I tried to call the default ws2812fx.show function if isFrame was not triggered, as a fallback, but that caused the MCU to keep resetting, so I commented that out. I guess that can't be done if we use the customshow function.
> 
I think you would call `show` then way too often. CustomShow would be the better approach anyway since it keeps the original functionality of the library and you just need to take care of how to get the LED data written to your (specific) setup.

I hope I did not miss something. ;-)

Regards,
Tobi

henmel, 2020-02-26:
Thank you so much,

This works like a charm! Just what was needed. This way it retains the default Library functionality with only the copyPixelData being injected, as needed. Perfect!

I am learning a lot, thanks to you both. You guys really know your stuff. Something like this would be a great example to include with the library, as I don't think most people realize that this is possible.

Thanks again,
Henry

henmel, 2020-03-10:
Hey Guys,

I am running into a stability issue when I scale the project up to the size I need. Using our finalized sketch, If I set the zone 1 to 522 LEDs and zone 2 to 348 LEDs, with segment length of 58 LEDs, and let the sketch run in auto cycle mode, my esp8266 NodeMCU v3 periodically hard resets. 

> 08:29:16.804 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
08:29:16.804 -> 
08:29:16.804 -> wdt reset
08:29:16.804 -> load 0x4010f000, len 1392, room 16 
08:29:16.804 -> tail 0
08:29:16.804 -> chksum 0xd0
08:29:16.804 -> csum 0xd0
08:29:16.804 -> v3d128e5c
08:29:16.804 -> ~ld
08:29:17.367 -> 
08:29:17.367 -> ESP full version: SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af

This happens anywhere from a couple of minutes to a number of hours and I am having hard time figuring out what the cause is. I have tried this on a couple of esp chips and they both do the same thing. Monitoring the serial monitor, it appears to happen on different effects. I even when back to the regular esp8266_webinterface example and set the number of LEDs to 870, in a single segment. The resets still happen periodically, but not as often. Have you tried letting it run for  a couple of days with a large number of LED's? You don't actually need to hookup any LEDs to the chip. Just let it run and check the serial monitor periodically for resets. When I set the number much lower, like 30 for each zone, it runs without resetting for days. Any ideas or suggestions?

gaijinsr, 2020-03-10:
Hi, before you (and others) spend a lot of time trying to figure out what the problem is in this rather unusual use case, have you tried to just get an ESP32 for 5$ and see whether it can handle the 870 LEDs more easily?

henmel, 2020-03-10:
Why is this case unusual? I have not tried getting ESP32. Do you think its a performance issue? I did try running at 160MHz instead of 80MHz and it made no difference.

moose4lord, 2020-03-11:
My longest strip is 300 LEDs long and I routinely run it for days without issue. As an experiment I set the LEDCOUNTs to 522 and 348 in my test setup and ran it overnight and didn't see any reboots. I only have V2 nodeMCUs, which are a little different than the V3 variety you have, but I don't know if that makes any difference.

The WDT resets you're seeing lead me to believe the WiFi is locking up, perhaps because interrupts are being disabled for a very long time (>24ms) when 800+ LEDs are being updated. See [https://arduino-esp8266.readthedocs.io/en/latest/reference.html#interrupts](https://arduino-esp8266.readthedocs.io/en/latest/reference.html#interrupts). You can try disabling the WiFi to see if the problem disappears. Of course you won't have network access to your ESP, so not really a solution, just an experiment to see if the problem is WiFi related.
```c++
// wifi_setup(); // comment this out
WiFi.mode( WIFI_OFF );
WiFi.forceSleepBegin();
delay(1);
```

Another thing you can try is selecting "Generic ESP8266 Module" as your board type. You'll get the option to use a different SDK (called "Espressif FW" in the Tools menu). Each SDK has a slightly different WiFI stack and you might have better luck with a newer SDK version.

Or you can try sprinkling some ```yield();``` statements in your loop(), which might help allocate more time for the WiFi stack do what it needs to do.

I'm just guessing though. Can't reproduce, so I'm just throwing out ideas.

henmel, 2020-03-11:
Thanks Keith,

I was afraid that WiFi possibly had something to do with it. I actually did some testing with Async WiFi Manger last night, which appears to made the issue a bit worst.

I will try your suggestions to see if that improves the stability and let you know how it goes. How often do you think yield(); should be triggered(seconds / milliseconds) in the void loop? I also ordered an ESP32 to play with and see if that makes a difference.

Thanks again for your support and a great library!

moose4lord, 2020-03-11:
You would call yield() after every major subroutine call in loop(). Like after server.handleClient(), ws2812fx.service() and copyPixelData(). This will give the WiFi stack more opportunities to manage the network queue and also feed the watchdog timer. It may help, it may not.

Did you order the dual core version of the ESP32? That version uses one core to manage WiFi and the other core to run your sketch. It works very well with WS2812FX. There's also a single core ESP32 that may not be any better then the ESP8266 for your situation. The ESP32 also has the advantage of using hardware DMA to drive the LEDs instead of the software driven "bit-bang" method. See the ws2812fx_esp32 example sketch.

henmel, 2020-03-11:
Thank you.

Not sure if its dual core. The description just said "ESP32 development board 2.4GHz WiFi wireless bluetooth antenna module". I thought they all were dual core. How can I tell when I get it?

moose4lord, 2020-03-12:
As far as I know the single core versions have "ESP32-SWOD" or "ESP32-SOLO" stamped on the top of the can. I would assume if the product description didn't say "single core", then you got the more common dual core variety.

henmel, 2020-03-25:
Hi Keith,

Quick update and another issue:

Per your suggestion, disabling WiFi in the sketch did resolve the resets, so it appears that WiFi and my version of NodeMCU are the  cause. Adding some yield statements helped a little but did not resolve the issue. It does seem to make a difference which router I am connection to. I have 4 here and some appear to make it more stable then others. Still waiting for my ESP32 from China.

I did run into another issue with one of the custom effects, multi comet. It appears when it is used in multiple segment setup, the effect breaks a bit, as the comets are no longer in one continuous mass. Please see the below test sketch:

```
/*
  Demo sketch for creating multiple custom effects.
  
  Keith Lord - 2018

  LICENSE

  The MIT License (MIT)

  Copyright (c) 2018  Keith Lord 

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sub-license, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
  THE SOFTWARE.
  
  CHANGELOG
  2018-07-19 initial version
*/

#include <WS2812FX.h>

// include the custom effects
// #include "custom/BlockDissolve.h"
//#include "custom/DualLarson.h"
 #include "custom/MultiComet.h"
 #include "custom/Oscillate.h"
// #include "custom/RainbowLarson.h"
// #include "custom/RandomChase.h"
// #include "custom/TriFade.h"
// #include "custom/VUMeter.h"

#define LED_COUNT 30
#define LED_PIN 4

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  
  ws2812fx.init();
  ws2812fx.setBrightness(30);

  // setup the custom effects
  uint8_t multiCometMode  = ws2812fx.setCustomMode(F("MultiComet"), multiComet);
  uint8_t OscillateMode  = ws2812fx.setCustomMode(F("Oscillate"), oscillate);

  uint32_t colors[] = {RED, BLUE, WHITE};

  ws2812fx.setSegment(0, 0, 9, multiCometMode, colors, 1000, false);
  ws2812fx.setSegment(1, 10, 19, multiCometMode, colors, 1000, false);
  //ws2812fx.setSegment(2, 20, 29, multiCometMode, colors, 1000, false);
  //ws2812fx.setSegment(1, 15, 29, OscillateMode, colors, 1000, false);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}
```
It looks like as long as only ope segment is used, it works fine, but if more then one are setup, the effect breaks.  Try commenting the second segment, the effect works fine. I did look at the library included copy of the effect, but did not see what the cause may be. Any ideas?

Thanks again for a great library!

moose4lord, 2020-03-26:
You've stumbled upon a weakness with some of the custom effects.

If you look through the library's source code you'll see that there's a small block of memory (16 bytes) set aside to store each segments runtime state. Here's the struct that defines that block:
```c++
// segment runtime parameters
    typedef struct Segment_runtime { // 16 bytes
      unsigned long next_time;
      uint32_t counter_mode_step;
      uint32_t counter_mode_call;
      uint8_t aux_param;   // auxiliary param (usually stores a color_wheel index)
      uint8_t aux_param2;  // auxiliary param (usually stores bitwise options)
      uint16_t aux_param3; // auxiliary param (usually stores a segment index)
    } segment_runtime;
```
For all of the "regular" effects, and most of the custom effects, this 16 byte struct is big enough to hold all the information needed to manage a segment at runtime. Unfortunately, there are four custom effects that are too complicated to encapsulate their runtime data in only 16 bytes:

> Heartbeat
> MultiComet
> Oscillate
> Popcorn

For those custom effects, some of the runtime data is stored in static variables. For example, in MultiComet.h:
```c++
  static int16_t comets[] = {UINT16_MAX, UINT16_MAX, UINT16_MAX, UINT16_MAX, UINT16_MAX, UINT16_MAX};
  static int8_t numComets = sizeof(comets)/sizeof(comets[0]);
```
Since static variables are shared across ALL segments, using these custom effects in more than one segment will cause the problem you see.

I can't think of an easy fix for this, other then refactoring the code to remove the static variables, which I think would be hard to do. As a work around, you can copy MultiComet.h into another custom effect, say, MultiComet2.h (renaming the function to multiComet2):
```c++
uint16_t multiComet2(void) {
  WS2812FX::Segment* seg = ws2812fx.getSegment(); // get the current segment
  int seglen = seg->stop - seg->start + 1;
.
.
.
  return (seg->speed / seglen);
}
```
Then assign a custom effect "slot" to each of the functions:
```c++
uint8_t multiCometMode  = ws2812fx.setCustomMode(F("MultiComet"), multiComet);
uint8_t multiComet2Mode  = ws2812fx.setCustomMode(F("MultiComet2"), multiComet2);
```
Then assign the effects to separate segments:
```c++
  ws2812fx.setSegment(0, 0, 9, multiCometMode, colors, 1000, false);
  ws2812fx.setSegment(1, 10, 19, multiComet2Mode, colors, 1000, false);
```
Ugh, that's a pretty ugly hack, but if you've got your heart set on two segments running the MultiComet effect, it should work.

henmel, 2020-03-26:
Thanks for a quick response. I don't think the hack would work as I have multiple segments and zones, as discussed before. I basically have two versions of my sketch for this project. One which automatically breaks up each zone into segments and lets me choose between the segment and full zone modes, the other uses the Copy Pixel Data version to fill both zones. The Copy Segment works great for those effects, since there is only one segment copied to the others, but with the other, I get this strange  side effect. I also noticed that with other custom effects that I have played with, that use static vars, as they get that skip effect with multiple segments.

Thanks again for the explanation and a great library.  Stay safe and healthy!