kitesurfer1404/WS2812FX #213: Serial Control example speed commands reversed

furstyferret-dev, 2020-02-21:
The speed commands (s+ and s-) are reversed in this example. Just needs the 1.2 and 0.8 multipliers swapping around. Thanks for a great library.

moose4lord, 2020-02-22:
You've touched on a confusing aspect of the library. The speed code was refactored a while back to change the "speed" parameter's behavior to be more of a delay-between-animation-frames. Setting speed=100 means that there will be a 100ms delay between each animation frame. This code refactoring helped solve the thorny problem of synchronizing animations across segments, but it also meant that, counterintuitively, increasing the "speed" parameter causes the animation speed to DECREASE. People assume increasing the speed parameter will increase the animation speed, but it's just the opposite. Maybe when we refactored the code we should have relabeled the speed parameter to something more intuitive, but that would have broken a lot of sketches, so it's still called "speed".

So using s+ in the serial_control example sketch does indeed arithmetically increase the speed parameter (decreasing the animation speed). If you read the book The Dark Forest by Lui Cixin, there's a procedure called a "mental seal", which basically makes someone believe a premise, no matter how irrational the premise may be. My "mental seal" is that the sketch is working as intended. :)

moose4lord, 2025-01-21:
I suppose this issue has been open long enough. It was really just a reminder about how the speed parameter behaves. Closing.