kitesurfer1404/WS2812FX #216: No issue but a custom effect request

psyqjo, 2020-04-12:
Could you generate a custom effect to populate an LED matrix with fixed color values?

moose4lord, 2020-04-12:
This is what I came up with.
```c++
#ifndef Matrix_h
#define Matrix_h

#include <WS2812FX.h>

extern WS2812FX ws2812fx;

uint32_t matrix_leds[][3][8] = { // setup for a 3x8 LED matrix
  { // page 0
  {RED,   RED,   RED,   RED,   RED,   RED,   RED,   RED},   // row 0
  {WHITE, WHITE, WHITE, WHITE, WHITE, WHITE, WHITE, WHITE}, // row 1
  {BLUE,  BLUE,  BLUE,  BLUE,  BLUE,  BLUE,  BLUE,  BLUE}   // row 2
  },
  { // page 1
  {YELLOW, YELLOW, YELLOW, YELLOW, YELLOW, YELLOW, YELLOW, YELLOW}, // row 0
  {PINK,   PINK,   PINK,   PINK,   PINK,   PINK,   PINK,   PINK},   // row 1
  {GREEN,  GREEN,  GREEN,  GREEN,  GREEN,  GREEN,  GREEN,  GREEN}   // row 2
  }
};

uint16_t matrix(void) {
  WS2812FX::Segment* seg = ws2812fx.getSegment();
  WS2812FX::Segment_runtime* segrt = ws2812fx.getSegmentRuntime();
  int seglen = seg->stop - seg->start + 1;

  const int numPages = sizeof(matrix_leds) / sizeof(matrix_leds[0]);
  const int numRows  = sizeof(matrix_leds[0]) / sizeof(matrix_leds[0][0]);
  const int numCols  = sizeof(matrix_leds[0][0]) / sizeof(matrix_leds[0][0][0]);

  uint16_t segIndex = 0;
  uint8_t pageIndex = segrt->aux_param; // aux_param will store the page index
  for(int i=0; i<numRows; i++) {
    for(int j=0; j<numCols; j++) {
      if(segIndex < seglen) {
        ws2812fx.setPixelColor(segIndex, matrix_leds[pageIndex][i][j]);
        segIndex++;
      }
    }
  }

  // increment to the next page
  segrt->aux_param < (numPages - 1) ? segrt->aux_param++ : segrt->aux_param = 0;

  return seg->speed;
}

#endif
```

You create an array of colors (matrix_leds[]) which reflects the size of your matrix and the colors you want to assign to each LED. I added code so you can create a simple animation by alternating between "pages" of data in the array. If you want a static display, just create one "page".

Hope that helps.

psyqjo, 2020-04-13:
thank you moose4lord, that's perfect.

only one thing is still not clear to me now, how do I have to set up a customEffect.h to also use the effects created with ws2812fx_segments_web?
or would it even be possible to output the code directly for the custom effects and save an effect.h file directly?

moose4lord, 2020-04-13:
I'm not sure I understand your question. All of the "normal" effects included with WS2812FX are intended for strips of LEDs, not a matrix of LEDs, so I don't think they would work with your setup.

I've attached my test sketch, which shows how to include the Matrix custom effect in a WS2812FX sketch. It is similar to the ws2812fx_custom_effect2 example sketch, which shows, in general, how to include custom effects.

Also I should mention that the Matrix sketch assumes the matrix of LEDs is wired in a unidirectional fashion like so:
[image]

If instead, the LEDs are wired in a serpentine route like this:
[image]
then you'll need some kind of transform function to arrange the data to conform to this configuration.

[matrix_test.ino.zip](https://github.com/kitesurfer1404/WS2812FX/files/4469445/matrix_test.ino.zip)

psyqjo, 2020-04-13:
everything with the matrix code worked well, there were no problems. it works without a problem.

My second question related to the general structure of effect.h

```
#ifndef Matrix_h
#define Matrix_h

#include <WS2812FX.h>

extern WS2812FX ws2812fx;
...
...
uint16_t matrix(void) {
  WS2812FX::Segment* seg = ws2812fx.getSegment();
  WS2812FX::Segment_runtime* segrt = ws2812fx.getSegmentRuntime();
  int seglen = seg->stop - seg->start + 1;
...
...
segrt->aux_param < (numPages - 1) ? segrt->aux_param++ : segrt->aux_param = 0;

  return seg->speed;
}

#endif
```

closed 2021-03-01 by moose4lord