kitesurfer1404/WS2812FX #217: Setting myModes from Web Interface

henmel, 2020-04-18:
Thanks for a great Library!

Using the webinterface example, Is there a way to pass a list of modes to myModes array, from the web interface? It would be great if I can choose which modes to autocycle through, right from the web interface.

```
      char *ptr = NULL;
      byte index = 0;
      ptr = strtok(modeslist, ",");  // takes a list of delimiters
      while(ptr != NULL)
      {
          myModes[index] = atoi(ptr);
          Serial.print("val: "); Serial.println((myModes[index]));
          index++;
          ptr = strtok(NULL, ",");  // takes a list of delimiters
      }
       sizeof_myModes = index;
       Serial.print("index: "); Serial.println((index));

```

I tried to pass a list, for example "10,12,15,30,35,42". then loop through the myModes array and assign the values, but when I try to loop back through the array to retrieve the values, the sizeof(myModes) is zero, so the loop terminates. The values do appear to be in the array, so if I loop through using the hard coded number of modes(6 in example above), the values are there. Is there a way to set the array size based on the number of values that are passed from the Web Interface?

Thanks in advance!

moose4lord, 2020-04-20:
The size of an array in C is set at compile time, so although you can change the array content at runtime, you can't change its size. But the code in the webinterface example sketch can be tweaked a bit to accommodate a myModes[] array that can be modified at runtime.

First, on the browser side, we need to send the new myModes[] array data to the ESP, so I added an \<input\> element to the html:
```HTML
<input id='myModes' name='myModes'></input>
```
and tweaked the javascript to append the input value to the query string sent to the ESP when the user clicks the "autocycle" buttons:
```JavaScript
function submitVal(name, val) {
  var xhttp = new XMLHttpRequest();
  if(name === 'a') val += '&myModes=' + document.querySelector('#myModes').value.trim();
  xhttp.open('GET', 'set?' + name + '=' + val, true);
  xhttp.send();
}
```
Then on the ESP side, set the maximum size of the array and create a variable to hold the number of modes in the array.
```C++
uint8_t myModes[20]; // maximum of 20 modes
uint8_t numMyModes = 0;
```
and everywhere the array size is referenced, change ```sizeof(myModes)``` to ```numMyModes```.

Then change the code that processes the "auto" button press to also update the myModes[] array:
```C++
    if(server.argName(i) == "a") {
      char *tmp = (char *)server.arg(i+1).c_str();  // get the myModes query parameter
      numMyModes = 0;
      char *ptr = strtok(tmp, ",");
      while(ptr != NULL) {
        int value = atoi(ptr);
        myModes[numMyModes] = value;
        numMyModes++;
        ptr = strtok(NULL, ",");
      }
      if(numMyModes > 0) {
        ws2812fx.setMode(myModes[0]);
      }

      if(server.arg(i)[0] == '-') {
        auto_cycle = false;
      } else {
        auto_cycle = true;
        auto_last_change = 0;
      }
    }
```

I'll leave it to you to make the web interface look pretty. :)

henmel, 2020-04-22:
Thank You! Works perfectly. One question though, do we need to declare the maximum number of modes? Can we just initiate with:
`uint8_t myModes[] = {};`

This appears to work too.

Thanks again for a great library!

moose4lord, 2020-04-22:
Your code snippet would initialize an array of size zero, which is surely not what you want. The C language does not check array bounds, so your code will happily read and write to the array, but you're reading and writing to memory not allocated to the array, which is a recipe for disaster.

henmel, 2020-04-22:
Thank you for the help and good explanation.

closed 2020-04-22 by henmel

henmel, 2020-05-03:
Hi Keith,

Not sure if I should start another thread for this. I have been playing around with Random Chase custom effect and have a question. It seems that no matter how long the segment length is, in access of 30 pixels, the effect does not appear to scale up. So if I have a string of 150 leds, it only appears to move up to about 30 max. The rest appear to stay black. Any ideas on how to make it scale up for longer segments?

Thanks in advance!

moose4lord, 2020-05-03:
That's odd. I do not see this kind of problem with my test setup. Is it only the Random Chase effect that seems to have the problem?

You're setting the LED_COUNT parameter at the top of the sketch, right?
```c++
#define LED_COUNT 150
```
That should be all that's necessary.

henmel, 2020-05-03:
Thanks for a quick response. I do have the LED_COUNT set correctly and Random Chase appears to be the only one effected. I do still use ESP8266. 

After some further testing, it appears that the number of LEDs lit is dependent on the strip brightness.  If I have the brightness set to 30, up to 30 are lit with a stream looking effect, at the end of the strip, which I really like. If I set the brightness up to 255, all 150 get fully lit with the chase effect and no stream effect at the end of the strip. Is there a way to keep the brightness low, but fill the strip with the stream looking effect for most of the strip length? Below is a copy of the modified ws2812fx_custom_effect2 example I used for testing, with brightness set at 30 and length at 150...

`#include <WS2812FX.h>

 #include "custom/RandomChase.h"

#define LED_COUNT 150
#define LED_PIN 4

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  
  ws2812fx.init();
  ws2812fx.setBrightness(30);

  // setup the custom effects
  uint8_t RandomChaseMode  = ws2812fx.setCustomMode(F("RandomChase"), randomChase);

  uint32_t colors[] = {RED, BLUE, WHITE};

  ws2812fx.setSegment(0, 0, LED_COUNT - 1, RandomChaseMode, colors, 50, FADE_SLOW);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}`

moose4lord, 2020-05-03:
Oh, sorry, I thought you were talking about the standard FX_MODE_CHASE_RANDOM effect, not the custom Random Chase effect. My bad. I probably should have called the custom Random Chase effect something else. :)

The problem you're seeing is caused by the for loop in the custom effect, which copies pixel data to shift the colors down the length of the strip.
```c++
for(uint16_t i=seg->stop; i>seg->start; i--) {
  ws2812fx.setPixelColor(i, ws2812fx.getPixelColor(i-1));
}
```
Unfortunately, it uses the getPixelColor() function which only returns an approximation of each LED's color when the LED strip brightness is less than 255. The color approximation is always dimmer than the actual color, so as the colors are shifted down the strip, you see the colors eventually fade to black. The problem get progressively worse as you decrease the brightness.

Fortunately, there's an easy fix. Instead of using a for loop and the getPixelColor() function, we can use the new copyPixels() function to do the pixel shifting. copyPixels() uses a more efficient memmove instruction to copy the pixel data, so does not suffer from the "color approximation" issue that getPixelColor() has.

Replace the for loop with this code:
```c++
  int seglen = seg->stop - seg->start + 1;
  ws2812fx.copyPixels(seg->start + 1, seg->start, seglen - 1);
```

henmel, 2020-05-03:
Thank you for the alternate code and the explanation. It makes more sense now. So, why is the same "getPixelColor" issue not apply further down in the code?

`
  uint32_t color = ws2812fx.getPixelColor(seg->start + 1);
  int r = random(1) != 0 ? (color >> 16 & 0xFF) : random(256);
  int g = random(1) != 0 ? (color >> 8  & 0xFF) : random(256);
  int b = random(1) != 0 ? (color       & 0xFF) : random(256);
  ws2812fx.setPixelColor(seg->start, r, g, b);

`

moose4lord, 2020-05-04:
The getPixelColor() from that section of code has the same "color approximation" issue, but it only affects the initial value of the very first pixel in the strip. So it has a negligible affect on the animation.

henmel, 2020-05-05:
Thank You! Sorry to be a bother again,

I have the myModes functionality, from above working in the sketch, but I am running into a problem saving and restoring the myModes array data in the EEPROM. I am saving both numMyModes and myModes array, but when I restore the array values, only first 4 appear to be correct, other values are not. Here are the pertinent parts from my saveToEEPROM and restoreFromEEPROM functions:

saveToEEPROM:
  EEPROM.put(sizeof(int) * 3, (int)numMyModes);
  EEPROM.put(sizeof(int) * 4, myModes);
  EEPROM.put(sizeof(int) * 5, (int)ws2812fx.getPin());
  EEPROM.put(sizeof(int) * 6, (int)ws2812fx.numPixels());
  EEPROM.put(sizeof(int) * 7, (int)ws2812fx.getBrightness());
  EEPROM.put(sizeof(int) * 8, (int)ws2812fx.getNumSegments());
  memcpy(&tmpSegments, ws2812fx.getSegments(), sizeof(tmpSegments));
  EEPROM.put(sizeof(int) * 9, tmpSegments);

restoreFromEEPROM:
    EEPROM.get(sizeof(int) * 3, numMyModes);
    EEPROM.get(sizeof(int) * 4, myModes);
    EEPROM.get(sizeof(int) * 5, pin);
    EEPROM.get(sizeof(int) * 6, length);
    EEPROM.get(sizeof(int) * 7, brightness);
    EEPROM.get(sizeof(int) * 8, numSegments);
    ws2812fx.setNumSegments(numSegments);
    EEPROM.get(sizeof(int) * 9, tmpSegments);
    memcpy(ws2812fx.getSegments(), &tmpSegments, sizeof(tmpSegments));
    WS2812FX::segment* segments = ws2812fx.getSegments();
 
What is the proper way of saving/restoring the myModes array data to EEPROM?

Thanks in advance!

moose4lord, 2020-05-05:
The first parameter of the EEPROM.get() and EEPROM.put() functions is the offset (in bytes) into the EEPROM memory. If you're just getting/putting integers, the technique you are using (using "sizeof(int) * n" to calculate the offsets) works fine.

Unfortunately, myModes and tmpSegments are arrays, not simple integers, so you have to put more effort into calculating the offset. Splitting the offset calculations into separate lines may help make it clear. Your save function would look something like this:
```c++
int offset = 0;
EEPROM.put(offset, (int)numMyModes); // store an int
offset = offset + sizeof(int); // increment the offset by the size of an int
EEPROM.put(offset, myModes);// store the myModes array
offset = offset + sizeof(myModes); // increment the offset by the size of the myModes array
EEPROM.put(offset, (int)ws2812fx.getPin()); // store an int
offset = offset + sizeof(int); // increment the offset by the size of an int
.
.
.
memcpy(&tmpSegments, ws2812fx.getSegments(), sizeof(tmpSegments));
EEPROM.put(offset, tmpSegments);
offset = offset + sizeof(tmpSegments); // increment the offset by the size of tmpSegments
.
.
.
```

henmel, 2020-05-05:
Thank you, I think I understand. I used your example from "ws2812fx_segments_web" initially to save EEPROM data. So, instead of revising all of my save and load code, in my previous code, I adjusted the location offset of all variables following myModes line by 5:
```
saveToEEPROM:
EEPROM.put(sizeof(int) * 3, (int)numMyModes);
EEPROM.put(sizeof(int) * 4, myModes);
EEPROM.put(sizeof(int) * 9, (int)ws2812fx.getPin());
EEPROM.put(sizeof(int) * 10, (int)ws2812fx.numPixels());
EEPROM.put(sizeof(int) * 11, (int)ws2812fx.getBrightness());
EEPROM.put(sizeof(int) * 12, (int)ws2812fx.getNumSegments());
memcpy(&tmpSegments, ws2812fx.getSegments(), sizeof(tmpSegments));
EEPROM.put(sizeof(int) * 13, tmpSegments);

restoreFromEEPROM:
EEPROM.get(sizeof(int) * 3, numMyModes);
EEPROM.get(sizeof(int) * 4, myModes);
EEPROM.get(sizeof(int) * 9, pin);
EEPROM.get(sizeof(int) * 10, length);
EEPROM.get(sizeof(int) * 11, brightness);
EEPROM.get(sizeof(int) * 12, numSegments);
ws2812fx.setNumSegments(numSegments);
EEPROM.get(sizeof(int) * 13, tmpSegments);
memcpy(ws2812fx.getSegments(), &tmpSegments, sizeof(tmpSegments));
WS2812FX::segment* segments = ws2812fx.getSegments();
```
This appears to give myModes enough space to save the 20 modes. I will try to rework the code as you suggested, but for now this appears to have fixed issue I ran into.

Thank you very much for your quick support and great explanations. It really helps a noob like me to learn how things actually work.

moose4lord, 2020-05-05:
Your solution will work as long as an int is 4 bytes and you don't change the size of myModes. Changing myModes[20] to myModes[30], or replacing the ESP8266 with a microprocessor that uses 2 byte integers will get you into trouble. :)