kitesurfer1404/WS2812FX #218: Porting of esp8266_webinterface example for ESP32

Sriyank681, 2020-04-24:
I can not get this Example "esp8266_webinterface" to run on ESP32 , Request to Please release this Example to run on ESP32.

moose4lord, 2020-04-24:
Can you let us know what's not working for you?

Sriyank681, 2020-04-25:
I am getting below Errors When Compiling for WEMOS LOLIN ESP32 
-------
esp8266_webinterface:85:1: error: 'ESP8266WebServer' does not name a type

 ESP8266WebServer server(HTTP_PORT);

------

esp8266_webinterface:108:3: error: 'server' was not declared in this scope

   server.on("/", srv_handle_index_html);
-----

exit status 1
'ESP8266WebServer' does not name a type
------

My Code :
I am using esp8266_webinterface example provided in library WS2812FX. 
removed below include files
//#include <ESP8266WiFi.h>
//#include <ESP8266WebServer.h>

and added below Files which are relevant to ESP32.
#include <WS2812FX.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

As this code is intended to run on ESP8266 , It seems rest of the code also needs to be edited to run on ESP32. 
For Your reference I was able to run below below code successfully on ESP32. 
https://github.com/jmponce84/WS2812FX-ESP32-web-example

moose4lord, 2020-04-25:
As you note, the webinterface sketch can be compiled for ESP32 by using that processor's WebService lib instead of the ESP8266 web server libs:
```c++
//#include <ESP8266WiFi.h>
//#include <ESP8266WebServer.h>
#include <WebServer.h>
...
//ESP8266WebServer server(HTTP_PORT);
WebServer server(HTTP_PORT);
...
//      ESP.reset();
      ESP.restart();
...
```

Unfortunately, that's not all there is to it. If you're strip of WS2812s is longer then about 30 LEDs, you'll see random glitches because of an issue with the ESP32's Wifi stack as described in issue #177.
You can workaround this issue with the ESP32's RMT hardware, as demonstrated by @jmponce84. Is there some reason you don't use his solution?

Sriyank681, 2020-04-27:
Thanks for your help, With the above modifications in code I was successfully able to run it on ESP32. 
I know the limitation of 30 LEDs, Not planning to use that.
I Used this default example because I liked the Default Dark web-interface and selecting colors is also easy.

closed 2020-04-27 by Sriyank681

abhishekduduskar, 2021-08-19:
> Thanks for your help, With the above modifications in code I was successfully able to run it on ESP32.
> I know the limitation of 30 LEDs, Not planning to use that.
> I Used this default example because I liked the Default Dark web-interface and selecting colors is also easy.

can you please share the modified code, I am getting this error after modifications 
```

exit status 1

'HTTP_PORT' was not declared in this scope
```

moose4lord, 2021-08-20:
Which modified sketch are you trying to compile?
I compiled the ESP32_test.zip sketch mentioned in issue #177 and did not see any issues.