kitesurfer1404/WS2812FX #219: LED out of timing/ out of sync

craigrobbo, 2020-04-25:
Hey there.

So I tried to do some dyncronised wipe effects to use as indicators on a headlight project im working on , however the 2 indicators are totally out of sync on real hardware and they just seem to stop wherever they want at the end of the animation (and the white just randomly fills back in)

`#include <WS2812FX.h>
#include <Arduino.h>

#define LED_COUNT 178
#define LED_PIN 9

#define BUTTON_PIN PD4

bool button_pressed = false;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  pinMode(BUTTON_PIN, INPUT);
 
  
  ws2812fx.init();
  ws2812fx.setBrightness(100);
   ws2812fx.setColor(WHITE);
   ws2812fx.setSpeed(1000);
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.start();
}

void processButton() {
  if(digitalRead(BUTTON_PIN) == 1 && button_pressed == false) {
  ws2812fx.setSegment(0,  0,  55, FX_MODE_COLOR_WIPE, ORANGE, 1000, false); 
  ws2812fx.setSegment(2, 89, 144, FX_MODE_COLOR_WIPE, ORANGE, 1000, false);  

    button_pressed = true;
    
  } else if(digitalRead(BUTTON_PIN) == 0 && button_pressed == true) {
  ws2812fx.setSegment(0,  0,  178, FX_MODE_CHASE_WHITE, WHITE, 1000, false); 
  ws2812fx.setSegment(2, 89, 144, FX_MODE_CHASE_WHITE, WHITE, 1000, false);

    button_pressed = false;

  }
}

void loop() {
  processButton();
  ws2812fx.service();
}`

moose4lord, 2020-04-25:
I changed your button setup to use the internal pull-up resistor and used the button to pull the GPIO to ground.
```C++
pinMode(BUTTON_PIN, INPUT_PULLUP);
```
Other than that minor tweak, the sketch seemed to work ok.

Although your segment setup is a bit odd. You configured segment 0 and 2, but skipped segment 1. You might want to explicitly setup four segment.
```C++
ws2812fx.setSegment(0, 0,   55,  FX_MODE_COLOR_WIPE, ORANGE, 1000, false);
ws2812fx.setSegment(1, 56,  88,  FX_MODE_STATIC,     WHITE,  1000, false);
ws2812fx.setSegment(2, 89,  144, FX_MODE_COLOR_WIPE, ORANGE, 1000, false);
ws2812fx.setSegment(3, 145, 177, FX_MODE_STATIC,     WHITE,  1000, false);
```

craigrobbo, 2020-04-25:
Thank you for that.

So I somewhat got it all working, however I figured why things are not right.

It seems the animations continue where they left off when the button is pressed rather than starting again from the first assigned pixel.

Any way to fix that?

craigrobbo, 2020-04-25:
``#include <WS2812FX.h>
#include <Arduino.h>

#define LED_COUNT 177
#define LED_PIN 6

#define BUTTON_PIN PD4

bool button_pressed = false;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
 pinMode(BUTTON_PIN, INPUT_PULLUP);
 
  
  ws2812fx.init();
  ws2812fx.setBrightness(100);
   ws2812fx.setColor(WHITE);
   ws2812fx.setSpeed(1000);
  ws2812fx.setMode(FX_MODE_CHASE_WHITE);
  ws2812fx.start();
}

void processButton() {
  if(digitalRead(BUTTON_PIN) == 1 && button_pressed == false) {
  ws2812fx.setSegment(0,  0,  55, FX_MODE_COLOR_WIPE, ORANGE, 1000, false); 
  ws2812fx.setSegment(1, 89, 144, FX_MODE_COLOR_WIPE, ORANGE, 1000, false);  

    button_pressed = true;
    
  } else if(digitalRead(BUTTON_PIN) == 0 && button_pressed == true) {
  ws2812fx.setSegment(0,  0,  55, FX_MODE_STATIC, WHITE, 1000, false); 
  ws2812fx.setSegment(1, 89, 144, FX_MODE_STATIC, WHITE, 1000, false);

    button_pressed = false;

  }
}

void loop() {
  processButton();
  ws2812fx.service();
}`

So I have changed to this now`

moose4lord, 2020-04-25:
You can add ```ws2812fx.resetSegmentRuntimes();``` before your setSegment statements to reset the animation parameters.

craigrobbo, 2020-04-25:
That seem to do the trick, Thank you very much.

I have one more question, I am currently using a 5v input to control my lights however when I add a second button in, Both buttons appear to stop working?

Here is my code for 2 x buttons for 2 different effects.

I am still fairly new to coding so I may have more questions in the future I hope you do not mind 😊

`#include <WS2812FX.h>
#include <Arduino.h>

#define LED_COUNT 177
#define LED_PIN 6

#define BUTTON_PIN PD4
#define BUTTON_PIN1 PD5

bool button_pressed = false;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
 pinMode(BUTTON_PIN, INPUT);
 pinMode(BUTTON_PIN1, INPUT);
  
  ws2812fx.init();
  ws2812fx.setBrightness(100);
   ws2812fx.setColor(WHITE);
   ws2812fx.setSpeed(1000);
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.start();
}

void processButton() {
  if(digitalRead(BUTTON_PIN) == 1 && button_pressed == false) {
  ws2812fx.setSegment(0,  0,  55, FX_MODE_COLOR_WIPE, ORANGE, 1000, false); 
  ws2812fx.setSegment(1, 89, 144, FX_MODE_COLOR_WIPE, ORANGE, 1000, false); 
  ws2812fx.resetSegmentRuntimes(); 

    button_pressed = true;
    
  } else if(digitalRead(BUTTON_PIN) == 0 && button_pressed == true) {
  ws2812fx.setSegment(0,  0,  55, FX_MODE_CHASE_WHITE, WHITE, 1000, false); 
  ws2812fx.setSegment(1, 89, 144, FX_MODE_CHASE_WHITE, WHITE, 1000, false);
  ws2812fx.resetSegmentRuntimes();

  }

    button_pressed = false;

      if(digitalRead(BUTTON_PIN1) == 1 && button_pressed == false) {
  ws2812fx.setSegment(0,  0,  55, FX_MODE_COLOR_WIPE, ORANGE, 1000, true); 
  ws2812fx.setSegment(1, 89, 144, FX_MODE_COLOR_WIPE, ORANGE, 1000, true); 
  ws2812fx.resetSegmentRuntimes(); 

    button_pressed = true;
    
  } else if(digitalRead(BUTTON_PIN1) == 0 && button_pressed == true) {
  ws2812fx.setSegment(0,  0,  55, FX_MODE_CHASE_WHITE, WHITE, 1000, true); 
  ws2812fx.setSegment(1, 89, 144, FX_MODE_CHASE_WHITE, WHITE, 1000, true);
  ws2812fx.resetSegmentRuntimes();

    button_pressed = false;

  }
}

void loop() {
  processButton();
  ws2812fx.service();
}`

kitesurfer1404, 2020-04-25:
Try to save the state of the buttons in two variables. button_pressed_1 and button_pressed_2.
Like now the first call of processButton might work as expected, but while one button is pressed, it's called a second time and the "else if" of the "other" button will reset button_pressed.

craigrobbo, 2020-04-25:
Thanks for that, I tried this and just got the error on compile

“exit status 1
'button_pressed_1' was not declared in this scope”

Would you be able to edit my code or give me the example that would work please

Again your help is highly appreciated!

kitesurfer1404, 2020-04-25:
Sounds like you have forgotten to add the declaration at the top. I added "button_presses_1" so it matches your "BUTTON_PIN" and "BUTTON_PIN1":

`#include <WS2812FX.h>
#include <Arduino.h>

#define LED_COUNT 177
#define LED_PIN 6

#define BUTTON_PIN PD4
#define BUTTON_PIN1 PD5

bool button_pressed = false;
bool button_pressed_1 = false;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
pinMode(BUTTON_PIN, INPUT);
pinMode(BUTTON_PIN1, INPUT);

ws2812fx.init();
ws2812fx.setBrightness(100);
ws2812fx.setColor(WHITE);
ws2812fx.setSpeed(1000);
ws2812fx.setMode(FX_MODE_STATIC);
ws2812fx.start();
}

void processButton() {
if(digitalRead(BUTTON_PIN) == 1 && button_pressed == false) {
ws2812fx.setSegment(0, 0, 55, FX_MODE_COLOR_WIPE, ORANGE, 1000, false);
ws2812fx.setSegment(1, 89, 144, FX_MODE_COLOR_WIPE, ORANGE, 1000, false);
ws2812fx.resetSegmentRuntimes();

button_pressed = true;

} else if(digitalRead(BUTTON_PIN) == 0 && button_pressed == true) {
ws2812fx.setSegment(0, 0, 55, FX_MODE_CHASE_WHITE, WHITE, 1000, false);
ws2812fx.setSegment(1, 89, 144, FX_MODE_CHASE_WHITE, WHITE, 1000, false);
ws2812fx.resetSegmentRuntimes();

}

button_pressed = false;

  if(digitalRead(BUTTON_PIN1) == 1 && button_pressed_1 == false) {

ws2812fx.setSegment(0, 0, 55, FX_MODE_COLOR_WIPE, ORANGE, 1000, true);
ws2812fx.setSegment(1, 89, 144, FX_MODE_COLOR_WIPE, ORANGE, 1000, true);
ws2812fx.resetSegmentRuntimes();

button_pressed_1 = true;

} else if(digitalRead(BUTTON_PIN1) == 0 && button_pressed_1 == true) {
ws2812fx.setSegment(0, 0, 55, FX_MODE_CHASE_WHITE, WHITE, 1000, true);
ws2812fx.setSegment(1, 89, 144, FX_MODE_CHASE_WHITE, WHITE, 1000, true);
ws2812fx.resetSegmentRuntimes();

button_pressed_1 = false;

}
}

void loop() {
processButton();
ws2812fx.service();
}`

kitesurfer1404, 2020-04-25:
Also, not being nitpicking in any way, better name your defines and variables in a speaking way and keep consistency in your namespace. I totally get, this code just evolved in the last minutes from first sketch to added features. But better start right away to refactor and rename stuff until it is "too lage" (too much code to go through, too many errors that can not easily be tracked down).

Which buttons pin is BUTTON_PIN for and same goes for BUTTON_PIN1. ;-) Is it for the same button but a different pin? ;-) Maybe better name it BUTTON_LEFT_PIN and BUTTON_RIGHT_PIN (or BUTTON_L_PIN and BUTTON_R_PIN, to keep it short and at same length). 
Or BUTTON_PIN_L and BUTTON_PIN_R if you prefer to have the differentiating part at the end.
Same goes for "button_pressed".

There are a lot of coding guidelines I never read. So all of this is just personal preference.

craigrobbo, 2020-04-25:
Hi there.

Please don’t feel like your nitpicking I genuinely apprte allt he advice I can get – I have only just started coding (the last couple of days) So I am ass new to this as you can imagine, as the cool kids say “total newbie”

I appreciate that there is people like yourself that’s willing to help and that you have done, I am going to get some beer tokens sent over on pay day for sure!

One last thing if you don’t mind.

The animation I have is for a car indicators (you proberbly guessed this)

I have it so the orange chases along the strip and at the end the white chases back on.

Now the white and orange animations now work great but its a little random as to when the white animation actually start, even using the reset segment code?

I can upload a video if it helps to explain?

Thanks again

Craig

kitesurfer1404, 2020-04-25:
I had the BMW-style turn lights in mind from the beginning. 

Check out the "isFrame()" function. https://github.com/kitesurfer1404/WS2812FX/blob/abece9a2a5a23027243851767c55cb8d2e19ff05/extras/WS2812FX%20change%20log.txt#L299 
If gives you information as soon one cycle of the animation has ended.

You probably don't want to switch back to white the moment the button is released but wait, until "isFrame" returns 1. You can add this maybe after

`else if(digitalRead(BUTTON_PIN1) == 0 && button_pressed_1 == true) {`

like this (untested)

`else if(digitalRead(BUTTON_PIN1) == 0 && button_pressed_1 == true) {
  if(ws2812fx.isFrame(0)) {
    // reset to white here and set button_pressed back here.
  }
}`

As said, untested.

You also might want to add some Serial.print statements for debugging in your code, so you can better watch what happens in the serial monitor.

kitesurfer1404, 2020-04-25:
Will be AFK for today. Maybe Keith still follows this if any more questions.

craigrobbo, 2020-04-25:
Thank you, I'll give that a try later :)

I'm still getting to know all the functions and how exactly they work.

Out of interest is there a place I can find all the commands for this lib?

moose4lord, 2020-04-25:
All of the lib's functions are listed in the WS2812FX.h file, although there's  no real documentation for each function.

Harm mentions the isFrame() function earlier, but I think he meant the isCycle() function. That can be used to time effect transitions when using the COLOR_WIPE, CHASE and LARSON_SCANNER effects. It is discussed in issue #199 .

craigrobbo, 2020-04-25:
> e() function earlier, but I think he meant the isCycle() function. That can be used to time effect transitions w

Thanks you very much once again!

So essentially now I have the start of the effect always resetting its animation before starting, how would I make sure it always finishes completely before starting new effect (to stop effects crossing over into each other)?

at the moment it start perfect but when the white chases its not really smooth and leaves a bit of a gap too (its also not quite consistent.

Better for the orange animation to compleatly finish then start the white?

Thank you again.

craigrobbo, 2020-04-25:
I tried both ws2812fx.isFrame(); and ws2812fx.isCyle(); and neither made any difference at all

moose4lord, 2020-04-26:
I think I would approach it a little differently. Rather than changing the effect when the button is pressed, I'd just use the COLOR_WIPE effect all the time and change the colors. This is what I came up with:
```c++
#include <WS2812FX.h>

#define LED_COUNT 144
#define LED_PIN D2

#define BUTTON_PIN D1

bool button_pressed = false;
bool button_toggled = false;

uint32_t colors_orange[] = {ORANGE, BLACK, BLACK}; // orange on a black background
uint32_t colors_white[]  = {WHITE, WHITE, WHITE}; // white on a white background

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  delay(500);
  Serial.println("\n");
  
  pinMode(BUTTON_PIN, INPUT_PULLUP);

  ws2812fx.init();
  ws2812fx.setBrightness(32);
  ws2812fx.setSegment(0,  0,  35, FX_MODE_COLOR_WIPE, colors_white, 1000, false);
  ws2812fx.setSegment(1, 72, 107, FX_MODE_COLOR_WIPE, colors_white, 1000, false);
  ws2812fx.start();
}

void processButton() {
  if(digitalRead(BUTTON_PIN) == LOW && button_pressed == false) { // button pressed
    button_pressed = true;
    button_toggled = true;
    Serial.println("button pressed");
  } else if(digitalRead(BUTTON_PIN) == HIGH && button_pressed == true) { // button released
    button_pressed = false;
    button_toggled = true;
    Serial.println("button released");
  }
}

void loop() {
  processButton();
  ws2812fx.service();

  // wait for the animation cycle to end AND for the button to change state
  if(ws2812fx.isCycle() && button_toggled) {
    if(button_pressed) { // if the button was pressed, use orange color
      ws2812fx.resetSegmentRuntimes();
      ws2812fx.setColors(0, colors_orange); // set colors for segment 0
      ws2812fx.setColors(1, colors_orange); // set colors for segment 1
    } else { // if the button was released, use white color
      ws2812fx.resetSegmentRuntimes();
      ws2812fx.setColors(0, colors_white); // set colors for segment 0
      ws2812fx.setColors(1, colors_white); // set colors for segment 1
    }
    button_toggled = false; // reset the button toggled flag
  }
}
```

craigrobbo, 2020-04-26:
> ```c++
> #include <WS2812FX.h>
>
> #define LED_COUNT 144
> #define LED_PIN D2
>
> #define BUTTON_PIN D1
>
> bool button_pressed = false;
> bool button_toggled = false;
>
> uint32_t colors_orange[] = {ORANGE, BLACK, [...]

Thanks for all your help< i think this is getting a bit too complicated for me now :(

The sketch you sent does work but its backwards, and I tried to change the true / false flags to make it reverse but it didnt work 

I need white to be on and amber on the button.

So if it helps clear up what I am doing, the controller I want to  be able to display a white light for running lights on a car, then when I press (for example ) pin 1 the left orange will light on the strip (just a portion of the strip) and then when pin 2 is pressed another part of the strip will light orange

I did figure out how to add a second output pin  so that I have 2 LED strips and 2 input pins but I honestly have no idea how to impliment this into the code you've given me :(

Sorry for being hard work

craigrobbo, 2020-04-27:
Alright so sets see if I can explain what I'm after so maybe we can simplify the design a bit.

I'm fairly new to coding and only understand the absolute basics.

So the controller I want for my headlights and I need the following functions from it.

To control sk6812 /ws2812 LEDs

There needs to be a left and right output pin (so 2 pins)

There needs to be a total of 7 inputs which are:

Day time running light
Normal Running light
Brake light
Reverse light
Fog light
Left indicator
Right indicator

And a carshow mode. 

The functions I want the controller to do is control segments of the left assigned to the various function... For example:

Headlights 
If I have a 200 led strip I might want all of it to be white for day time running light, then when I indicator left only a quarter of the strip changes from white to Amber.

Example 2

As tail lights
If I have a 100 led strip for the rear lights all 100 led maybe running light, but 50 of them need to get brighter for brake light and then there needs to be priority of some sort for when I then use the indicator too.

So that's the function, I would then need effects like wave, chase and fade effects too 

And the ability to have multi colours for the show mode.

So the way all this is controlled doesn't necessarily have to be via a mobile phone app as long as I've a way to reasonably easily change these settings for different lights

craigrobbo, 2020-04-27:
I figured out the lights being out of sync, its because i wasn't using a resistor to ground on data input pin argh...it all seems much smoother now!

closed 2021-03-01 by moose4lord

rivelta, 2022-05-06:
> You can add `ws2812fx.resetSegmentRuntimes();` before your setSegment statements to reset the animation parameters.

i have been tried but this led cant run

moose4lord, 2022-05-06:
I'm not sure what you mean? What's not working?

moose4lord, 2022-05-07:
Ah, I've seen this problem many times. In your loop() function you are reading the pushbuttons and constantly updating the segments. Every time through the loop the segments are being reset, which does not give enough time to run the animations.

What should be happening is the segments only get updated when the buttons change. If a button changes from HIGH to LOW, or from LOW to HIGH, then update the segments.
Try something like this:
```c++
#include <WS2812FX.h>

#define LED_PIN   14  // digital pin used to drive the LED strip
#define LED_COUNT  9  // number of LEDs on the strip

int button_A = 12;
int button_B = 13;
int val_A = 0;
int val_B = 0;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  delay(500);
  
  pinMode(button_A, INPUT_PULLUP); // declare pushbutton as input with pullup resistor
  pinMode(button_B, INPUT_PULLUP);

  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();

  int buttonA = digitalRead(button_A); // read button state
  int buttonB = digitalRead(button_B);
  if(buttonA == val_A && buttonB == val_B) return; // if button state hasn't changed, just return

  Serial.print(buttonA); Serial.print(","); Serial.println(buttonB);

  val_A = buttonA; // save the button state
  val_B = buttonB;

  ws2812fx.resetSegmentRuntimes(); // reset the animation runtimes

  // update the segments
  if ( val_A == HIGH && val_B == LOW ) {
    ws2812fx.setSegment(0,  0, 3,  4,  0xfffb00, 1300, true);
    ws2812fx.setSegment(1,  4, 8, 28,  0xffffff,   80, true);
  } else if ( val_A == LOW && val_B == HIGH ) {
    ws2812fx.setSegment(0,  0, 3, 28,  0xffffff,   80, true);
    ws2812fx.setSegment(1,  4, 8,  4,  0xfffb00, 1300, false);
  } else {
    ws2812fx.setSegment(0,  0, 3, 28,  0xffffff,   80);
    ws2812fx.setSegment(1,  4, 8, 28,  0xffffff,   80);
  }
}
```

rivelta, 2022-05-08:
noted , i will try next week because im little bit bussy with my office work :D