kitesurfer1404/WS2812FX #220: Help with triggers/switches

craigrobbo, 2020-04-27:
I'm sorry to be a pain guys I hope you don't mind helping me, im so close to finishing my project and feel like im failing at the last hurdle.

A freind has been helping me a lot with my code and hes been a huge help, but we're both at a point where neither of us know how to fix this issue.

This controller will control tail lights on a car, we have button presses as triggers(5v input on the board) and all the inputs work , the left and right indicator displays, the brake lights display etc...the issue is as soon as the 5v is removed (the button is no longer presssed) the lights do random things.

I know why this is, its becuse all the buttons "off" state tell the controller to do different things, and I need a way to tell the controller to just "go back to what it was before" when the button is not pressed.

or if thre is a way to just have the trigger/button as an overlay or "only do this effect when pressed"

I really hope I am making sense here. and im really sorry but massivly appreciate your time(i really owe you some beers)

Thanks

`#include <WS2812FX.h>
#include <Arduino.h>

#define LED_COUNT_STRIP_1 54             //Number of LEDs in the strip, strip 1
#define LED_COUNT_STRIP_2 54             //Number of LEDs in the strip, strip 2
#define LED_PIN_STRIP_1 8                  //Which pin are the LEDs connected to, strip 1
#define LED_PIN_STRIP_2 9                  //Which pin are the LEDs connected to, strip 2
#define BUTTON_PIN  PD3           //Button for DRL
#define BUTTON_PIN1 A1            //Button for  Running/side lights  
#define BUTTON_PIN2 PD4           //Button for  Brake lights
#define BUTTON_PIN3 A1            //Button for  Fog light            
#define BUTTON_PIN4 A1            //Button for Reverse light
#define BUTTON_PIN5 A1            //Button for Left indicator
#define BUTTON_PIN6 A1            //Button for Right indicator

bool button_pressed_0 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"
bool button_pressed_1 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

void setup() {
pinMode(BUTTON_PIN,  INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN1, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN2, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN3, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN4, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN5, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN6, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground

      //This will run when the arduino is powered on once 
  ws2812fx_strip1.init();
  ws2812fx_strip2.init();
  
  ws2812fx_strip1.setBrightness(100);
  ws2812fx_strip2.setBrightness(100);

  ws2812fx_strip1.setColor(BLACK);
  ws2812fx_strip2.setColor(BLACK);

  ws2812fx_strip1.setSpeed(1000);
  ws2812fx_strip2.setSpeed(1000);

  ws2812fx_strip1.setMode(FX_MODE_STATIC);
  ws2812fx_strip2.setMode(FX_MODE_STATIC);

  ws2812fx_strip1.start();
  ws2812fx_strip2.start();
}

void processButton() {        //Button processing down below, insert code in this area for button presses

//---------------------------------------------------------------------------------------------------

//----------This is when the button 0 DRL IS pressed -----------
  
[... 149 lines cut ...]
    button_pressed_0 = true;

//----------This is when the button 6 RIGHT INDICATOR IS NOT pressed -----------

} else if(digitalRead(BUTTON_PIN6) == 0 && button_pressed_0 == true) {

        // parameters: index, start, stop, mode, color, speed, reverse
    ws2812fx_strip1.setBrightness (50);    
    ws2812fx_strip1.setSegment(0, 0, 54, FX_MODE_COLOR_WIPE, RED, 1000, false);
    ws2812fx_strip1.resetSegmentRuntimes();       // This will make sure the segment always starts at the beginning

    button_pressed_0 = false;

}
}

//---------------------------------------------------------------------------------------------------
//Button Number *please add*

  }
void loop() {
    processButton();
    ws2812fx_strip1.service();
    ws2812fx_strip2.service();
}`

kitesurfer1404, 2020-04-27:
Do you have a pull-down-resistor at the inputs? So do they go to GND/0V after being released? Or do they just "float" without any connection?

I think you have to save ALL states of the buttons independently.
if(digitalRead(BUTTON_PIN**2**) == 1 && button_pressed_**0** == false) {

So you need more of

`bool button_pressed_0 = false; // Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"
bool button_pressed_1 = false; // Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"`

at the top for each button.

I think. Too late to wrap my head around this without trying to run it. ;-)

kitesurfer1404, 2020-04-27:
Oh, btw: Send a nice video as soon this is all running and implemented in the car. Can't wait to see that. :)

craigrobbo, 2020-04-27:
> 
> 
> Do you have a pull-down-resistor at the inputs? So do they go to GND/0V after being released? Or do they just "float" without any connection?
> 
> I think you have to save ALL states of the buttons independently. Looks like button 2 saves its state in button_pressed_0:
> 
> if(digitalRead(BUTTON_PIN**2**) == 1 && button_pressed_**0** == false) {
> 
> Same for the other buttons. So you need more of
> 
> `bool button_pressed_0 = false; // Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&" bool button_pressed_1 = false; // Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"`
> 
> at the top for each button.
> 
> I think. Too late to wrap my head around this without trying to run it. ;-)

Hi mate.
They are on a pull down resistor yeah so they  I have it wired with a resistor to ground.(this is what i did to fix my previous issue of irratic animations on button presses)

craigrobbo, 2020-04-27:
> 
> 
> Oh, btw: Send a nice video as soon this is all running and implemented in the car. Can't wait to see that. :)

Oh I deffinatly will mate for sure :)

This is a video of my current lights but this is not using adddressable leds, just standard LEDS and a very basic relay switching controller (what i want to get away from )

https://www.youtube.com/watch?v=lAswSpYJLP8

craigrobbo, 2020-04-28:
Thanks agian guy its super appreciated!

craigrobbo, 2020-04-28:
Hey guys, sorry to be that guy and to be a pain, I don't supose you got a chance to take alook at my code?

moose4lord, 2020-04-29:
Did you fix the button_pressed variable problem that @kitesurfer1404 mentioned? You only have two button_pressed variables and you need seven, one for each button.
```c++
bool button_pressed_0 = false;
bool button_pressed_1 = false;
bool button_pressed_2 = false;
bool button_pressed_3 = false;
bool button_pressed_4 = false;
bool button_pressed_5 = false;
bool button_pressed_6 = false;
```
So the code for the seventh button would look like this
```c++
  //----------This is when the button 6 RIGHT INDICATOR IS pressed -----------
  if(digitalRead(BUTTON_PIN6) == 1 && button_pressed_6 == false) { // correct button variable
    ws2812fx_strip1.setBrightness (100);
    ws2812fx_strip1.setSegment(0, 0, 54, FX_MODE_COLOR_WIPE, RED, 1000, false);
    ws2812fx_strip1.resetSegmentRuntimes();
    button_pressed_6 = true;  // correct button variable
  } else if(digitalRead(BUTTON_PIN6) == 0 && button_pressed_6 == true) { // correct button variable
    ws2812fx_strip1.setBrightness (50);    
    ws2812fx_strip1.setSegment(0, 0, 54, FX_MODE_COLOR_WIPE, RED, 1000, false);
    ws2812fx_strip1.resetSegmentRuntimes();
    button_pressed_6 = false; // correct button variable
  }
```

Also, when I copied/pasted your code into the Arduino IDE I noticed a mismatched curly brace. Look at the processButton() function and make sure the curly braces properly enclose each code block. If you indent the code blocks inside the curly braces, you'll see the error.

craigrobbo, 2020-04-29:
> ```c++
>  else if(digitalRead(BUTTON_PIN6) == 0 && button_pressed_6 == true) { // correct button variable
>     ws2812fx_strip1.setBrightness (50);    
>     ws2812fx_strip1.setSegment(0, 0, 54, FX_MODE_COLOR_WIPE, RED, 1000, false);
>     ws2812fx_strip1.resetSegmentRuntimes();
>     button_pressed_6 = false; // correct button variable
> ```

Thanks for that.

The part i need to change, but unsure on is the part "if button is not pressed"

In my code I just havin git reverting the segment back to a colour, but this won't really work after testing (well the code work but if doesnt do whaat i intended)

So lets say I have running light on RED and I activate the indicator once the indicator has finished the code tells it to go back to RED thats great.

However if the running light is not on and I indicator, well once the indicator has finished just taht segment then lights up RED

Hope I make sense here?

craigrobbo, 2020-04-29:
I edited it now to change like this, would this seem ok now?

I don't think still will go back to "previous state" though?

`#include <WS2812FX.h>
#include <Arduino.h>

#define LED_COUNT_STRIP_1 54             //Number of LEDs in the strip, strip 1
#define LED_COUNT_STRIP_2 54             //Number of LEDs in the strip, strip 2
#define LED_PIN_STRIP_1 8                  //Which pin are the LEDs connected to, strip 1
#define LED_PIN_STRIP_2 9                  //Which pin are the LEDs connected to, strip 2
#define BUTTON_PIN  PD3           //Button for DRL
#define BUTTON_PIN1 A1            //Button for  Running/side lights  
#define BUTTON_PIN2 PD4           //Button for  Brake lights
#define BUTTON_PIN3 A1            //Button for  Fog light            
#define BUTTON_PIN4 A1            //Button for Reverse light
#define BUTTON_PIN5 A1            //Button for Left indicator
#define BUTTON_PIN6 A1            //Button for Right indicator

bool button_pressed_0 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"
bool button_pressed_1 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"
bool button_pressed_2 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"
bool button_pressed_3 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"
bool button_pressed_4 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"
bool button_pressed_5 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"
bool button_pressed_6 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

void setup() {
pinMode(BUTTON_PIN,  INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN1, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN2, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN3, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN4, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN5, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN6, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground

      //This will run when the arduino is powered on once 
  ws2812fx_strip1.init();
  ws2812fx_strip2.init();
  
  ws2812fx_strip1.setBrightness(100);
  ws2812fx_strip2.setBrightness(100);

  ws2812fx_strip1.setColor(BLACK);
  ws2812fx_strip2.setColor(BLACK);

  ws2812fx_strip1.setSpeed(1000);
  ws2812fx_strip2.setSpeed(1000);

  ws2812fx_strip1.setMode(FX_MODE_STATIC);
  ws2812fx_strip2.setMode(FX_MODE_STATIC);

  ws2812fx_strip1.start();
  ws2812fx_strip2.start();
}

void processButton() {        //Button processing down below, insert code in this area for button presses

//---------------------------------------------------------------------------------------------------

//----------This is when the button 0 DRL IS pressed -----------
  
if(digitalRead(BUTTON_PIN) == 1 && button_pressed_0 == false) {

        // parameters: index, start, stop, mode, color, speed, reverse
    ws2812fx_strip1.setBrightness (100);
    ws2812fx_strip1.setSegment(0, 0, 54, FX_MODE_STATIC, WHITE, 1000, false);
[... 144 lines cut ...]
    button_pressed_6 = true;

//----------This is when the button 6 RIGHT INDICATOR IS NOT pressed -----------

} else if(digitalRead(BUTTON_PIN6) == 0 && button_pressed_6 == true) {

        // parameters: index, start, stop, mode, color, speed, reverse
    ws2812fx_strip1.setBrightness (50);    
    ws2812fx_strip1.setSegment(0, 0, 54, FX_MODE_COLOR_WIPE, RED, 1000, false);
    ws2812fx_strip1.resetSegmentRuntimes();       // This will make sure the segment always starts at the beginning

    button_pressed_6 = false;

}
}

//---------------------------------------------------------------------------------------------------
//Button Number *please add*

  }
void loop() {
    processButton();
    ws2812fx_strip1.service();
    ws2812fx_strip2.service();
}`

moose4lord, 2020-04-29:
There's still a mismatched curly brace in your code in the processButton() function.

I guess I'm still not clear what behavior you're going for. What do you want to happen when a button is released?

craigrobbo, 2020-04-29:
> 
> 
> There's still a mismatched curly brace in your code in the processButton() function.
> 
> I guess I'm still not clear what behavior you're going for. What do you want to happen when a button is released?

Sorry please for give me, Im still super new to coding :(

Im not sure which brace I hmave missmatched?

So let me try to explain best i possibly can.

The controller will essentially control tail lights.

My code at the moment does this.

If the red running light is on (receiveing 5v at the input pin) and I indicate by applying 5v at the input pin the indicator works great, and when I release the 5v from indicator the light goes back to being red.

However if I remove the 5v from the running light and then indicate by applying 5v to indicator, when I remove the 5v from the indicator the running light for that segment turns red and stays on.

I understand this is is because i have my code set to turn this segment back on after button release(or 5v is removed in this case)

What I actually need the code to do is when the button is released, just go back to "previous state" for the segment

The problem is the previous state for this segment won't always be the same, sometimes it maybe a running light on, somtimes it may be DRL sometimes it could be fog light etc etc...

I hope this makes some sense.

moose4lord, 2020-04-29:
Oh ok, I get it. So when you press a button, you want to save the LED settings, then update the LEDs; and when you release the button, you want to restore the previously saved settings.

To do that you'd need to define a variable to hold the saved settings. In this case we'll save the segment data:
```c++
WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);
WS2812FX::segment strip1_saved_segment; // variable to hold saved segment data for strip 1
```
And create a couple functions to save and restore the segment data:
```c++
void save_strip1() {
  memcpy(&strip1_saved_segment, ws2812fx_strip1.getSegment(), sizeof(WS2812FX::segment));
  ws2812fx_strip1.resetSegmentRuntimes();
}

void restore_strip1() {
  memcpy(ws2812fx_strip1.getSegment(), &strip1_saved_segment, sizeof(WS2812FX::segment));
  ws2812fx_strip1.resetSegmentRuntimes();
}
```
And adjust the button code to save the segment data when the button is pressed, and restore the segment data when the button is released.
```c++
  //----------This is when the button 0 DRL IS pressed -----------
  if(digitalRead(BUTTON_PIN) == LOW && button_pressed_0 == false) {
    save_strip1();

    ws2812fx_strip1.setBrightness (100);
    ws2812fx_strip1.setSegment(0, 0, 54, FX_MODE_STATIC, WHITE, 1000, false);
    button_pressed_0 = true;
  } else if(digitalRead(BUTTON_PIN) == HIGH && button_pressed_0 == true) {
    restore_strip1();
    button_pressed_0 = false;
  }
```

And of course repeat all that code for strip2.

craigrobbo, 2020-04-29:
> Oh ok, I get it. So when you press a button, you want to save the LED settings, then update the LEDs; and when you release the button, you want to restore the previously saved settings.
>
> To do that [...]

Thank you, Ill update my code and report back, once again i genuinly appreciate all your time. Ill get you soem beer tokens sent over when Im paid :)

craigrobbo, 2020-04-29:
Magic, so  this ALMOST works like i need it to - so close haha.

When I apply Voltage to button 0 White lights, when I release it tuns off *perfect

When I apply voltage to button 1 Red lights, when i release it turns off *perfect

When i apply Voltage to button 0 for red light (turn red) then apply voltage to button 1 the white turns on and when I then remove voltage from button 1 it goes back to red * perfect

However at this point it all goes wrong because if I remove voltage from red it doesn't revert back to being off again it stays red :(

craigrobbo, 2020-04-29:
Alright I figured that part, I need a seperate memcopy for each button to remember its state.

Thats worked great to stop buttons overlapping, but this seemingly has broken the segment feature.

If I use one segment it works fine, but if I use 2 segments the second segment starts acting irratic and doing all sorts of random things.

Also i took out some buttons from the code for now just to simplfy things on the bench

This is code for single segment:

`#include <WS2812FX.h>
#include <Arduino.h>

#define LED_COUNT_STRIP_1 54             //Number of LEDs in the strip, strip 1
#define LED_COUNT_STRIP_2 54             //Number of LEDs in the strip, strip 2
#define LED_PIN_STRIP_1 8                 //Which pin are the LEDs connected to, strip 1
#define LED_PIN_STRIP_2 9                  //Which pin are the LEDs connected to, strip 2
#define BUTTON_PIN  PD4           //Button for DRL
#define BUTTON_PIN1 PD5           //Button for  Running/side lights  

bool button_pressed_0 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"
bool button_pressed_1 = false;     //  Required for different input or buttons and must be added below and must be added after "if(digitalRead(BUTTON_PIN) == 1 &&"

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

WS2812FX::segment strip1_saved_segment; // variable to hold saved segment data for strip 1 //Storage "file name" xxx_saved_segment
WS2812FX::segment strip2_saved_segment; // variable to hold saved segment data for strip 1 //Storage "file name" xxx_saved_segment

void save_segment1() { //Saves the segments button press into memory with &xxx_saved_segment being file name
  memcpy(&strip1_saved_segment, ws2812fx_strip1.getSegment(), sizeof(WS2812FX::segment));
  ws2812fx_strip1.resetSegmentRuntimes();
}
void restore_segment1() { //Loads the segments button press into memory with &xxx_saved_segment being file name
  memcpy(ws2812fx_strip1.getSegment(), &strip1_saved_segment, sizeof(WS2812FX::segment));
  ws2812fx_strip1.resetSegmentRuntimes();
}

  void save_segment2() { //Saves the segments button press into memory with &xxx_saved_segment being file name
  memcpy(&strip2_saved_segment, ws2812fx_strip1.getSegment(), sizeof(WS2812FX::segment));
  ws2812fx_strip1.resetSegmentRuntimes();
  }

void restore_segment2() {//Loads the segments button press into memory with &xxx_saved_segment being file name
  memcpy(ws2812fx_strip1.getSegment(), &strip2_saved_segment, sizeof(WS2812FX::segment));
  ws2812fx_strip1.resetSegmentRuntimes();

}

void setup() {
pinMode(BUTTON_PIN,  INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground
pinMode(BUTTON_PIN1, INPUT);       //Defines the button type can be INPUT for 5v or INPUT_PULLUP for connecting to ground

      //This will run when the arduino is powered on once 
  ws2812fx_strip1.init();
  ws2812fx_strip2.init();
  
  ws2812fx_strip1.setBrightness(100);
  ws2812fx_strip2.setBrightness(100);

  ws2812fx_strip1.setColor(BLACK);
  ws2812fx_strip2.setColor(BLACK);

  ws2812fx_strip1.setSpeed(1000);
  ws2812fx_strip2.setSpeed(1000);

  ws2812fx_strip1.start();
  ws2812fx_strip2.start();
}

[... 130 lines cut ...]
//---------------------------------------------------------------------------------------------------

  //----------This is when the button 1 RUNNING LIGHT IS pressed -----------
  if(digitalRead(BUTTON_PIN1) == HIGH && button_pressed_1 == false) {
    save_segment2(); // File name of save

    ws2812fx_strip1.setBrightness (100);
    ws2812fx_strip1.setSegment(0, 0, 27, FX_MODE_COLOR_WIPE, ORANGE, 1000, true);
    ws2812fx_strip1.setSegment(1, 28, 54, FX_MODE_COLOR_WIPE, ORANGE, 1000, true);
    
    button_pressed_1 = true;
  } else if(digitalRead(BUTTON_PIN1) == LOW && button_pressed_1 == true) {
       restore_segment2(); // File name of save
    button_pressed_1 = false;
  }

//---------------------------------------------------------------------------------------------------
//Button Number *please add*

  }
void loop() {
    processButton();
    ws2812fx_strip1.service();
    ws2812fx_strip2.service();
}`

moose4lord, 2020-04-29:
You've made a couple mistakes transposing strip1 and strip2 data. I would rename your save/restore functions to help remind you which strip is being saved/restored. Maybe this:
```c++
void save_strip1_segment() {
  memcpy(&strip1_saved_segment, ws2812fx_strip1.getSegment(), sizeof(WS2812FX::segment));
  ws2812fx_strip1.resetSegmentRuntimes();
}

void restore_strip1_segment() {
  memcpy(ws2812fx_strip1.getSegment(), &strip1_saved_segment, sizeof(WS2812FX::segment));
  ws2812fx_strip1.resetSegmentRuntimes();
}

void save_strip2_segment() {
  memcpy(&strip2_saved_segment, ws2812fx_strip2.getSegment(), sizeof(WS2812FX::segment));
  ws2812fx_strip2.resetSegmentRuntimes();
}

void restore_strip2_segment() {
  memcpy(ws2812fx_strip2.getSegment(), &strip2_saved_segment, sizeof(WS2812FX::segment));
  ws2812fx_strip2.resetSegmentRuntimes();
}
```
Then, in the processButton() function, remember to use the strip1 save/restore functions when working with strip1; and use the strip2 save/restore functions when working with strip2.

As far as the behavior you see when you press more than one button simultaneously, this code will not work for that situation. I guess I assumed you would only press, then release, one button at a time. If you're going to press multiple buttons at a time, some kind of stack would need to be created to keep track of more than one "previous" state. Not a trivial feature to implement.

craigrobbo, 2020-04-29:
Thanks for your reply.

Once I fixed the save / restore it actually works with multiple button pressed! It did what I intended. 

However I'm struggling to explain but here's a video. 

Even just one button at a time it won't allow me more than one segment without being irraric. 

https://youtu.be/PTO-8S4LybY

moose4lord, 2020-04-30:
The save/restore code only saves and restores the first segment's data. To save/restore the data for all segments, you need this:
```c++
WS2812FX::segment strip1_saved_segments[MAX_NUM_SEGMENTS];

void save_strip1() {
  memcpy(strip1_saved_segments, ws2812fx_strip1.getSegments(), sizeof(strip1_saved_segments));
  ws2812fx_strip1.resetSegmentRuntimes();
}

void restore_strip1() {
  memcpy(ws2812fx_strip1.getSegments(), strip1_saved_segments, sizeof(strip1_saved_segments));
  ws2812fx_strip1.resetSegmentRuntimes();
}
```

craigrobbo, 2020-04-30:
Thanks again!

I feel like im 1 step forwards two steps back here and Im so sorry.

So the code you've given me works, but the first and second buttons do not work quite right  so the first button I have to set the first segment as 0 but then on the second button the first segment has to start at 1 - The issue with doing this is the animations seems to half complete (but leave some leds on) and the animations just gets shorter and shorter, really strange.

I think Im missing somthing really obvious haha

So if I leave button ones segment as zero and its own save spot and then give button 2 segment 0 and its own save location the animations work but i can only use one button at a time.

however if I give segments increasing numbers (even outside of button) they do this weed not finishing its sequence.

I can take a video if you like?

moose4lord, 2020-04-30:
I'm confused. Have you switch from using two strips to using one strip with two segments? Or do you have two strips, each with two segments?

craigrobbo, 2020-04-30:
So to simplyfy things at the moment ive just cut my code down to 1 strip and 2 buttons (just to get everything working) then eventually i will have 2 strips and 7 buttons

The buttons will be as follows:

DRL
Running lights
Brake
Reverse
Fog
Left indicator 
Right indicator

The reason multiple buttons may need to be pressed at once is for example if running light is on (thats a button) and then I brake or indicator, or even running light, brake and indicate all at the same time.

The code is so close to actually working the way I need right now but I just keep hitting these little hurdles.

I do understnd what the problems are and how they relate to the code i just don't know how to fix it unfortunatly.

From what I can gather when it says save the       strip1.getSegment I am essentially saving everything with that prefix as a result there is stuff thats "overlapping"

ideally what I need is a way for each button or each section of code to have its own save state, if thats even possible?

moose4lord, 2020-04-30:
Ok, I think I understand a little better. The idea of multiple buttons being pressed was throwing me off.

Unfortunately, I think we've been going down the wrong path. If we're allowing multiple buttons to be pressed, it seems we need a way to prioritize the buttons. As more than one button is pressed, we would traverse the priority tree and whichever button has the highest priority wins.

For example, suppose the brake lights have the highest priority, DRL lights have a medium priority, and running lights have the lowest priority. If the brake button and the running lights button are both pressed, then the brake light pattern would be displayed, because it has a higher priority than the running lights. In code it might look like this:
```c++
if(brake_lights_button_pressed) {
  // code to configure strip for brake light effect
  else if(drl_lights_button_pressed) {
    // code to configure strip for DRL effect
    } else if(running_lights_button_pressed) {
      // code to configure strip for running light effect
    }
  }
}
```
Does that make sense?

craigrobbo, 2020-04-30:
> Ok, I think I understand a little better. The idea of multiple buttons being pressed was throwing me off.
>
> Unfortunately, I think we've been going down the wrong path. If we're allowing multiple [...]

That actually makes a lot more sense now yeah.

So the highest of priority in the button presses should be at the top then filter down in order of priority?

I would implement this for every button at the section of "if"?

Would you mind at all just slotting that into my code above into one example and I can work wit from there please?

Again i really do appreciat eyour help a lot!

moose4lord, 2020-04-30:
This is what I came up with:
```c++
#include <WS2812FX.h>

// define LED strip parameters
#define LED_COUNT_STRIP_1 54 // Number of LEDs in strip 1
#define LED_COUNT_STRIP_2 54 // Number of LEDs in strip 2
#define LED_PIN_STRIP_1   D2 // GPIO pin used to drive strip 1
#define LED_PIN_STRIP_2   D3 // GPIO pin used to drive strip 2

// define button pins
#define DRL_BUTTON            D0 // GPIO pin assgned to the DRL button
#define RUNNING_LIGHTS_BUTTON D1 // GPIO pin assgned to the running lights button
#define BRAKE_LIGHTS_BUTTON   D1
#define FOG_LIGHTS_BUTTON     D1
#define REVERSE_LIGHTS_BUTTON D1
#define LEFT_LIGHTS_BUTTON    D1
#define RIGHT_LIGHTS_BUTTON   D1

// change these defines based on your button hardware setup
#define BUTTON_ACTIVE LOW          // define button active state, HIGH or LOW
#define INPUT_TYPE    INPUT_PULLUP // define GPIO input type, INPUT or INPUT_PULLUP

bool drl_button_pressed     = false;
bool running_button_pressed = false;
bool brake_button_pressed   = false;
bool fog_button_pressed     = false;
bool reverse_button_pressed = false;
bool left_button_pressed    = false;
bool right_button_pressed   = false;

bool buttons_have_changed = false; // flag when the buttons have changed

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

void setup() {
  pinMode(DRL_BUTTON,            INPUT_TYPE);
  pinMode(RUNNING_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(BRAKE_LIGHTS_BUTTON,   INPUT_TYPE);
  pinMode(FOG_LIGHTS_BUTTON,     INPUT_TYPE);
  pinMode(REVERSE_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(LEFT_LIGHTS_BUTTON,    INPUT_TYPE);
  pinMode(RIGHT_LIGHTS_BUTTON,   INPUT_TYPE);

  ws2812fx_strip1.init();
  ws2812fx_strip2.init();
  
  ws2812fx_strip1.setBrightness(100);
  ws2812fx_strip2.setBrightness(100);

  ws2812fx_strip1.setSegment(0, 0, LED_COUNT_STRIP_1 - 1, FX_MODE_STATIC, BLACK, 1000, false);
  ws2812fx_strip2.setSegment(0, 0, LED_COUNT_STRIP_2 - 1, FX_MODE_STATIC, BLACK, 1000, false);

  ws2812fx_strip1.start();
  ws2812fx_strip2.start();
}

void processButton() { //Button processing down below, insert code in this area for button presses

  // brake button
  if(digitalRead(BRAKE_LIGHTS_BUTTON) == BUTTON_ACTIVE && brake_button_pressed == false) { // brake button switched from off to on
    brake_button_pressed = true;
    buttons_have_changed = true;
  } else if(digitalRead(BRAKE_LIGHTS_BUTTON) != BUTTON_ACTIVE && brake_button_pressed == true) { // brake button switched from on to off
    brake_button_pressed = false;
    buttons_have_changed = true;
  }

  // DRL button
  if(digitalRead(DRL_BUTTON) == BUTTON_ACTIVE && drl_button_pressed == false) { // DRL button switched from off to on
    drl_button_pressed = true;
    buttons_have_changed = true;
  } else if(digitalRead(DRL_BUTTON) != BUTTON_ACTIVE && drl_button_pressed == true) { // DRL button switched from on to off
    drl_button_pressed = false;
    buttons_have_changed = true;
  }

  // Running lights button
  if(digitalRead(RUNNING_LIGHTS_BUTTON) == BUTTON_ACTIVE && running_button_pressed == false) { // Running lights button switched from off to on
    running_button_pressed = true;
    buttons_have_changed = true;
  } else if(digitalRead(RUNNING_LIGHTS_BUTTON) != BUTTON_ACTIVE && running_button_pressed == true) { // Running lights button switched from off to on
    running_button_pressed = false;
    buttons_have_changed = true;
  }
}

void processLED() {
  if(brake_button_pressed) {
    ws2812fx_strip1.setBrightness (100);
    ws2812fx_strip2.setBrightness (100);
    ws2812fx_strip1.setSegment(0, 0, LED_COUNT_STRIP_1 - 1, FX_MODE_STATIC, RED, 1000, false);
    ws2812fx_strip2.setSegment(0, 0, LED_COUNT_STRIP_2 - 1, FX_MODE_STATIC, RED, 1000, false);
    ws2812fx_strip1.resetSegmentRuntimes();
    ws2812fx_strip2.resetSegmentRuntimes();
  } else if(drl_button_pressed) {
    ws2812fx_strip1.setBrightness (50);
    ws2812fx_strip2.setBrightness (50);
    ws2812fx_strip1.setSegment(0, 0, LED_COUNT_STRIP_1 - 1, FX_MODE_STATIC, WHITE, 1000, false);
    ws2812fx_strip2.setSegment(0, 0, LED_COUNT_STRIP_2 - 1, FX_MODE_STATIC, WHITE, 1000, false);
    ws2812fx_strip1.resetSegmentRuntimes();
    ws2812fx_strip2.resetSegmentRuntimes();
  } else if(running_button_pressed) {
    ws2812fx_strip1.setBrightness (100);
    ws2812fx_strip2.setBrightness (100);
    ws2812fx_strip1.setSegment(0, 0, LED_COUNT_STRIP_1 - 1, FX_MODE_STATIC, WHITE, 1000, false);
    ws2812fx_strip2.setSegment(0, 0, LED_COUNT_STRIP_2 - 1, FX_MODE_STATIC, WHITE, 1000, false);
    ws2812fx_strip1.resetSegmentRuntimes();
    ws2812fx_strip2.resetSegmentRuntimes();
  } else { // no buttons pressed
    ws2812fx_strip1.setBrightness(0); // turn off both strips
    ws2812fx_strip2.setBrightness(0);
  }
}

void loop() {
  processButton();

  if(buttons_have_changed) {
    processLED();
    buttons_have_changed = false;
  }

  ws2812fx_strip1.service();
  ws2812fx_strip2.service();
}
```
I split up the button processing and LED processing into two functions.
Also note, my hardware setup is a little different from yours. I use INPUT_PULLUP and my buttons pull the I/O pin to ground when pressed. Saves me from having to connect a pull-down resistor to each button. Adjust these statements for your setup:
```c++
#define BUTTON_ACTIVE LOW          // define button active state, HIGH or LOW
#define INPUT_TYPE    INPUT_PULLUP // define GPIO input type, INPUT or INPUT_PULLUP
```

craigrobbo, 2020-05-01:
Hi mate, please forgive my ignorance in not replying quickly my partner has been rushed into hospital so I may not get a chance to test this out right away.

please accept my appologies

craigrobbo, 2020-05-01:
Hi, partner home safe, she's got a bad kidney infection so lots of rest.

I got a chance to try the code and while it does... Sort of work it still does some really strange things as can be seen here in video. 

I think I'm way out of my comfort zone on this project I thought controlling leds would be easyish. 

I feel a bit gutted as got all the other bits for the project and I'm failing at this hurdle. 

https://youtu.be/uaYMgvLZZ-I

moose4lord, 2020-05-01:
Hope your girl feels better soon.
I'm not sure what you're doing in your video. Can you show your code?

craigrobbo, 2020-05-01:
Hi mate thank you, yeah she's just resting up now.

The code is the one you gave me above I just changed pull up to input for 5v inputs. 

In the video I pressed red running light, then indicator separately then in the second part I'm holding red running light on then applying indicator and removing indocsot, it seems the indicator animation just stops wherever it is but also whenever indicator is applied it makes the brightness level of the entire strip go from 50% to 100% :(

moose4lord, 2020-05-02:
Well, after thinking about it some more, I wasn't happy with the previous "prioritizing" sketch I pasted above. It was incomplete, only setting up three of the seven buttons, and as I thought about adding the rest of the buttons, the priority tree was too simplistic and the logic sort of fell apart.

So here is my second attempt at a sketch that prioritizes the LED effects.
```c++
#include <WS2812FX.h>

// define LED strip parameters
#define LED_COUNT_STRIP_1 54 // Number of LEDs in strip 1
#define LED_COUNT_STRIP_2 54 // Number of LEDs in strip 2
#define LED_PIN_STRIP_1    7 // GPIO pin used to drive strip 1
#define LED_PIN_STRIP_2    8 // GPIO pin used to drive strip 2

// define button pins
#define DRL_BUTTON            0 // GPIO pin assigned to the DRL button
#define RUNNING_LIGHTS_BUTTON 1 // GPIO pin assigned to the running lights button
#define BRAKE_LIGHTS_BUTTON   2
#define FOG_LIGHTS_BUTTON     3
#define REVERSE_LIGHTS_BUTTON 4
#define LEFT_LIGHTS_BUTTON    5
#define RIGHT_LIGHTS_BUTTON   6

// change these defines based on your button hardware setup
#define BUTTON_ACTIVE LOW          // define button active state, HIGH or LOW
#define INPUT_TYPE    INPUT_PULLUP // define GPIO input type, INPUT or INPUT_PULLUP

bool drl_button_pressed     = false;
bool running_button_pressed = false;
bool brake_button_pressed   = false;
bool fog_button_pressed     = false;
bool reverse_button_pressed = false;
bool left_button_pressed    = false;
bool right_button_pressed   = false;

bool buttons_have_changed = false; // flag when the buttons have changed

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

void setup() {

//  Serial.begin(9600);
//  while(!Serial);
//  delay(500);

  pinMode(DRL_BUTTON,            INPUT_TYPE);
  pinMode(RUNNING_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(BRAKE_LIGHTS_BUTTON,   INPUT_TYPE);
  pinMode(FOG_LIGHTS_BUTTON,     INPUT_TYPE);
  pinMode(REVERSE_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(LEFT_LIGHTS_BUTTON,    INPUT_TYPE);
  pinMode(RIGHT_LIGHTS_BUTTON,   INPUT_TYPE);

  ws2812fx_strip1.init();
  ws2812fx_strip2.init();
  
  ws2812fx_strip1.setBrightness(100);
  ws2812fx_strip2.setBrightness(100);

  ws2812fx_strip1.setSegment(0, 0, LED_COUNT_STRIP_1 - 1, FX_MODE_STATIC, BLACK, 1000, false);
  ws2812fx_strip2.setSegment(0, 0, LED_COUNT_STRIP_2 - 1, FX_MODE_STATIC, BLACK, 1000, false);

  ws2812fx_strip1.start();
  ws2812fx_strip2.start();
}

void processButton() { //Button processing down below, insert code in this area for button presses

  // brake button
  if(digitalRead(BRAKE_LIGHTS_BUTTON) == BUTTON_ACTIVE && brake_button_pressed == false) { // brake button switched from off to on
    //  Serial.println("Brake");
[... 96 lines cut ...]
  }

  // left and right indicators have a medium priority and over-ride the fog/drl/running lights
  if(left_button_pressed) {
    ws2812fx_strip1.setSegment(0, 0, LED_COUNT_STRIP_1 - 1, FX_MODE_COLOR_WIPE, ORANGE, 1000, false);
  }
  if(right_button_pressed) {
    ws2812fx_strip2.setSegment(0, 0, LED_COUNT_STRIP_1 - 1, FX_MODE_COLOR_WIPE, ORANGE, 1000, false);
  }
}

void loop() {
  processButton();
  
  if(buttons_have_changed) {
    buttons_have_changed = false;
    processLED();
  }

  ws2812fx_strip1.service();
  ws2812fx_strip2.service();
}
```

See if that works any better for you.

craigrobbo, 2020-05-02:
Thanks for that.

With all buttons active its a bit difficult on the bench to do without a breadboard (im just using wires and resistors at the moment) Ive orderd a breadboard and jumper  kit so I can try this properly and illl report back.
once again thank you so much for your time

craigrobbo, 2020-05-02:
So I got this all set up and I feel like I'm being a real pain here, I'm just waiting to be blocked or ignored haha.

So the strangeness continues.

I feel like my idea is easy but in reality it's super difficult. 

If I turn the fog light on the indicator doesn't work at all, if I press brake and indicator it does strange things like part of the animation for indicator will show then stop. 

Also the DRL is 100% brightness the running light should be 50% brightness but if I press brake the whole strip goes to 100% brightness. 

Here's another video and I also tried to explain a little too as to what's going on. 

I'm really sorry for all this. 

https://youtu.be/6JmreKyxEOc

moose4lord, 2020-05-03:
Hmmmm, it's odd that your video shows your LEDs seem to be white by default. When you first power on your Arduino the default state, with no buttons pressed, should be all LEDs off. That bears some investigation.

The first part of your video looks ok to me. The brake light has the highest priority, so even when you activate the turn indicator, the brake light should remain on, because it has a higher priority than the turn indicator. The middle part of your video when you're using the fog light doesn't look right, it's not the behavior that I see with my test setup. When I turn on the fog lights, both strips turn yellow, then if I turn on the left indicator, the left strip displays the orange color wipe effect and the right strip stays on as the fog light. Look over the code in the processLED() function that prioritizes the buttons/effects. The priorities make sense to me, but I may not have implemented the priorities that you had in mind.

Lastly, in my last example sketch I only changed the colors and effects when different buttons are pressed, not the brightness. If you want to vary the brightness of the running lights, you'll have to add some setBrightness() statements. Maybe this:
```c++
void processLED() {
  ws2812fx_strip1.resetSegmentRuntimes();
  ws2812fx_strip2.resetSegmentRuntimes();
  ws2812fx_strip1.setBrightness(100);  // default brightness
  ws2812fx_strip2.setBrightness(100);
.
.
.
  } else if(running_button_pressed) {
    ws2812fx_strip1.setBrightness(50); // change brightness for running lights
    ws2812fx_strip2.setBrightness(50);
    ws2812fx_strip1.setSegment(0, 0, LED_COUNT_STRIP_1 - 1, FX_MODE_CHASE_WHITE, RED, 1000, false);
    ws2812fx_strip2.setSegment(0, 0, LED_COUNT_STRIP_2 - 1, FX_MODE_CHASE_WHITE, RED, 1000, false);
```
You can of course change the effects and colors to suit your taste.

I made a quick video of what I'm seeing:
[https://youtu.be/oacqyG5Krt8](https://youtu.be/oacqyG5Krt8)

craigrobbo, 2020-05-03:
Thanks Keith I really do appreciate the time and help and thanks for video too!

So as it stands I'm waiting on some more parts as I've ordered some more led stops, it's a bit difficult using my TV Ambi light ones at the moment so I can bench test it properly.

In the final design the setments will be somthing like this.

The daytime running light will be the entire strip in red, the normal running light will be a dimmed red(also the entire strip)

The indicator would be approx the last 10 or so leds on the strip, and brake light will be somewhere in the middle of the strip. The dog and reverse will likely be at the opposite end of the strip to the indicator taking about 5 leds each. Of course the indicator brake fog and reverse need to be 100% power.

In your example what hall s if you segment the indicator and brake light and press them both together?

I'll try my setup again over the next day or so.

Once again thank you, and like I say I'll get you some beers on pay day

https://youtu.be/oacqyG5Krt8

craigrobbo, 2020-05-03:
I've done you another youtube responce.

Im not sure if I was clear sorry so - what I have done here is put all the different buttons as different segments on the strip and this should hopefully explain the issue I'm having :)

https://www.youtube.com/watch?v=qXfnIMaY46g

craigrobbo, 2020-05-03:
I thought it maybe best to include the edited code I used too.

i havn't really changed a lot, just input pins, segments/ colors and brightness is all

`#include <WS2812FX.h>

// define LED strip parameters
#define LED_COUNT_STRIP_1 54 // Number of LEDs in strip 1
#define LED_COUNT_STRIP_2 54 // Number of LEDs in strip 2
#define LED_PIN_STRIP_1    10 // GPIO pin used to drive strip 1
#define LED_PIN_STRIP_2    11 // GPIO pin used to drive strip 2

// define button pins
#define DRL_BUTTON            2 // GPIO pin assigned to the DRL button
#define RUNNING_LIGHTS_BUTTON 3 // GPIO pin assigned to the running lights button
#define BRAKE_LIGHTS_BUTTON   4
#define FOG_LIGHTS_BUTTON     5
#define REVERSE_LIGHTS_BUTTON 6
#define LEFT_LIGHTS_BUTTON    7
#define RIGHT_LIGHTS_BUTTON   8

// change these defines based on your button hardware setup
#define BUTTON_ACTIVE LOW          // define button active state, HIGH or LOW
#define INPUT_TYPE    INPUT_PULLUP // define GPIO input type, INPUT or INPUT_PULLUP

bool drl_button_pressed     = false;
bool running_button_pressed = false;
bool brake_button_pressed   = false;
bool fog_button_pressed     = false;
bool reverse_button_pressed = false;
bool left_button_pressed    = false;
bool right_button_pressed   = false;

bool buttons_have_changed = false; // flag when the buttons have changed

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

void setup() {

//  Serial.begin(9600);
//  while(!Serial);
//  delay(500);

  pinMode(DRL_BUTTON,            INPUT_TYPE);
  pinMode(RUNNING_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(BRAKE_LIGHTS_BUTTON,   INPUT_TYPE);
  pinMode(FOG_LIGHTS_BUTTON,     INPUT_TYPE);
  pinMode(REVERSE_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(LEFT_LIGHTS_BUTTON,    INPUT_TYPE);
  pinMode(RIGHT_LIGHTS_BUTTON,   INPUT_TYPE);

  ws2812fx_strip1.init();
  ws2812fx_strip2.init();
  
  ws2812fx_strip1.setBrightness(100);
  ws2812fx_strip2.setBrightness(100);

  ws2812fx_strip1.setSegment(0, 0, LED_COUNT_STRIP_1 - 1, FX_MODE_STATIC, BLACK, 1000, false);
  ws2812fx_strip2.setSegment(0, 0, LED_COUNT_STRIP_2 - 1, FX_MODE_STATIC, BLACK, 1000, false);

  ws2812fx_strip1.start();
  ws2812fx_strip2.start();
}

void processButton() { //Button processing down below, insert code in this area for button presses

  // brake button
  if(digitalRead(BRAKE_LIGHTS_BUTTON) == BUTTON_ACTIVE && brake_button_pressed == false) { // brake button switched from off to on
    //  Serial.println("Brake");
[... 99 lines cut ...]
    ws2812fx_strip2.setSegment(0, 0, LED_COUNT_STRIP_2 - 1, FX_MODE_STATIC, BLACK, 1000, false);
  }

  // left and right indicators have a medium priority and over-ride the fog/drl/running lights
  if(left_button_pressed) {
           ws2812fx_strip1.setBrightness(100);
    ws2812fx_strip1.setSegment(0, 0, 15 , FX_MODE_COLOR_WIPE, ORANGE, 1000, false);
  }
  if(right_button_pressed) {
         ws2812fx_strip1.setBrightness(100);
    ws2812fx_strip2.setSegment(0, 0, 15, FX_MODE_COLOR_WIPE, ORANGE, 1000, false);
  }
}

void loop() {
  processButton();
  
  if(buttons_have_changed) {
    buttons_have_changed = false;
    processLED();
  }

  ws2812fx_strip1.service();
  ws2812fx_strip2.service();
}`

moose4lord, 2020-05-03:
Thanks for posting your code, Craig. That's helps a lot. I didn't realize you were partitioning the LED strips into segments like that. I thought the entire strip was devoted to a single effect, meaning, for instance, if the brake button was pressed, the entire strip showed the "brake effect". Having a brake segment, a fog light segment, a reverse light segment and a turn indicator segment, changes how I was thinking of implementing the project. I will have to think about it some more.

I'm curious about the purpose of the second LED strip (ws2812fx_strip2). Is this still part of your plan? I had thought that strip1 would be attached to the left side of the car and strip 2 to the right side of the car to accommodate the left and right turn indicators. But if the various effects are implemented in their own segments on one strip, what's strip2 for?

craigrobbo, 2020-05-03:
Ahh yes I'm really sorry for not being clear on that!

So essentially the tail light would have one single continuous strip which is segmented for various effects like brake fog etc just like I showed.

The 2 srraigs are one for left head light or tail light and one for right or left tail light. 

Using a strip per effect is kinda possible but the arduino runs out of memory and pins quickly when you end up with 5 or 6 strips per light and it also limits some effects. 

The end goal here is a controller that'll control 2 separate strips for left and right for headlights and tail lights and I can just modify the effects and segments per light I build. 

Eventually once its all dialed in I'd like to add in a show mode that can cycle the effects too but that's not important at the moment. 

For now I just want to focus on getting it working. 

Once again I genuinely appreciate your help and support!

moose4lord, 2020-05-04:
I have another test sketch for you to try. The use of segments makes the button priority code a little easier.

I created a set_all_strips() function to partition each strip into four segments; one segment for brake lights, one for fog lights, one for reverse lights and one for turn indicator lights. The function lets me set all segments to the color passed as a parameter.
```c++
void set_all_strips(uint32_t color) {
  ws2812fx_strip1.setSegment(0,  0, 15, FX_MODE_STATIC, color, 1000, false); // right/left lights segment
  ws2812fx_strip1.setSegment(1, 16, 44, FX_MODE_STATIC, color, 1000, false); // brake lights segment
  ws2812fx_strip1.setSegment(2, 45, 50, FX_MODE_STATIC, color, 1000, false); // fog lights segment
  ws2812fx_strip1.setSegment(3, 51, 53, FX_MODE_STATIC, color, 1000, false); // reverse lights segment

  ws2812fx_strip2.setSegment(0,  0, 15, FX_MODE_STATIC, color, 1000, false); // right/left lights segment
  ws2812fx_strip2.setSegment(1, 16, 44, FX_MODE_STATIC, color, 1000, false); // brake lights segment
  ws2812fx_strip2.setSegment(2, 45, 50, FX_MODE_STATIC, color, 1000, false); // fog lights segment
  ws2812fx_strip2.setSegment(3, 51, 53, FX_MODE_STATIC, color, 1000, false); // reverse lights segment
}
```
Then I adjusted the processLED() code to modify each segment as buttons are pressed. I think this better fits the behavior you have in mind.

Here is the entire test sketch:
```c++
#include <WS2812FX.h>

// define LED strip parameters
#define LED_COUNT_STRIP_1 54 // Number of LEDs in strip 1
#define LED_COUNT_STRIP_2 54 // Number of LEDs in strip 2
#define LED_PIN_STRIP_1   10 // GPIO pin used to drive strip 1
#define LED_PIN_STRIP_2   11 // GPIO pin used to drive strip 2

// define button pins
#define DRL_BUTTON            2 // GPIO pin assigned to the DRL button
#define RUNNING_LIGHTS_BUTTON 3 // GPIO pin assigned to the running lights button
#define BRAKE_LIGHTS_BUTTON   4
#define FOG_LIGHTS_BUTTON     5
#define REVERSE_LIGHTS_BUTTON 6
#define LEFT_LIGHTS_BUTTON    7
#define RIGHT_LIGHTS_BUTTON   8

// change these defines based on your button hardware setup
#define BUTTON_ACTIVE LOW // define button active state, HIGH or LOW
#define INPUT_TYPE INPUT_PULLUP // define GPIO input type, INPUT or INPUT_PULLUP

// some color intensity modifiers
#define DIM(c)      (uint32_t)(c & 0xbfbfbfbf) // 75% brightness
#define DIMMER(c)   (uint32_t)(c & 0x7f7f7f7f) // 50% brightness
#define DIMMEST(c)  (uint32_t)(c & 0x3f3f3f3f) // 25% brightness

bool drl_button_pressed = false;
bool running_button_pressed = false;
bool brake_button_pressed = false;
bool fog_button_pressed = false;
bool reverse_button_pressed = false;
bool left_button_pressed = false;
bool right_button_pressed = false;

bool buttons_have_changed = false; // flag when the buttons have changed

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

void setup() {

  // Serial.begin(9600);
  // while(!Serial);
  // delay(500);
  
  pinMode(DRL_BUTTON,            INPUT_TYPE);
  pinMode(RUNNING_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(BRAKE_LIGHTS_BUTTON,   INPUT_TYPE);
  pinMode(FOG_LIGHTS_BUTTON,     INPUT_TYPE);
  pinMode(REVERSE_LIGHTS_BUTTON, INPUT_TYPE);
[... 125 lines cut ...]
    ws2812fx_strip1.setSegment(1, 16, 44, FX_MODE_STATIC, RED, 1000, false); // brake lights segment
    ws2812fx_strip2.setSegment(1, 16, 44, FX_MODE_STATIC, RED, 1000, false); // brake lights segment
  }

  if(left_button_pressed) { // left indicator (set strip1 only)
    ws2812fx_strip1.setSegment(0,  0, 15, FX_MODE_COLOR_WIPE, ORANGE, 1000, false); // left indicator lights segment
  }

  if(right_button_pressed) { // right indicator (set strip2 only)
    ws2812fx_strip2.setSegment(0,  0, 15, FX_MODE_COLOR_WIPE, ORANGE, 1000, false); // right indocator lights segment
  }
}

void loop() {
  processButton();
  
  if(buttons_have_changed) {
    buttons_have_changed = false;
    processLED();
  }
  
  ws2812fx_strip1.service();
  ws2812fx_strip2.service();
}
```

craigrobbo, 2020-05-04:
Thanks Keith, so I gave that a try this morning and it pretty much does exactly as it says on the tin!

Thank you.

Of course, I wouldn't be me if there wasn't a "but" and I am sure you are ready for shootin gme at this point (ducks quickly)

I tried putting the running light on and indicating, works perfectly, but when I brake the animation for the indicator stops for about 1/2 second and then sort of randomly starts again?

My only other wuation would be, this code looks a lot more complicated t than I thought it would initially and I think its more than I could have done - How would I now go about modifying/adding segments for use in different lights etc

(for example my alfa romeo actually has 3 seperate indicator lights)

moose4lord, 2020-05-04:
I see your point about the indicator animation pausing for a brief moment when the brake button is pressed. This is due to the resetSegmentRuntimes() statements at the top of the processLED() function. Whenever a button is pressed, the animation for all segments is restarted from the beginning. I'll have to give it some thought. I'll see if I can simplify it too. It is a little too complicated.

craigrobbo, 2020-05-04:
Thanks again Keith, that does make sense I understand why it's doing that as for the cold I think I can work it out but I'm not 100% sure would I just need to increment the segment number each time I had in a segment but obviously all the others would go up as well accordingly?

Also how long sure if you drive or if you're into car customisation but I would like to make the offer that if we get this down and sorted that offer to make you a set of headlights free of charge justify the headlight and cover the shipping cost.

I normally charge £500 for these builds without such a fancy controller I'll do yours free :)

moose4lord, 2020-05-04:
Thank you for offering to build me some fancy headlights. That's very generous of you, but I'm afraid they would be wasted on my old, clunky Hyundai. :) But I insist you make a video of your car when you're done, and we'll call it even.

I have another version of the sketch for you to try. I changed the way animations are reset to work around the indicator pausing problem. I also tried to make it a little simpler for you to manage the segments.

Each segment number is assigned a name here:
```c++
#define INDICATOR_SEGMENT 0 // left/right indicator segment index
#define BRAKE_SEGMENT     1 // brake lights segment index
#define FOG_SEGMENT       2 // fog lights segment index
#define REVERSE_SEGMENT   3 // reverse segment index
```
If you wanted to add another segment, you'd just need to define a new name for segment 4. You can create up to 10 segments, numbered 0-9.

Then in the set_all_strips() function you configure each segment by setting it's first LED, last LED, effect, color, speed, and reverse parameters. Remember first and last LED are numbered starting from zero, so with 54 LEDs they are numbered 0-53.
```c++
void set_all_strips(uint32_t color) {
  ws2812fx_strip1.setSegment(INDICATOR_SEGMENT,  0, 15, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(BRAKE_SEGMENT,     16, 44, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(FOG_SEGMENT,       45, 50, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(REVERSE_SEGMENT,   51, 53, FX_MODE_STATIC, color, 1000, false);

  ws2812fx_strip2.setSegment(INDICATOR_SEGMENT,  0, 15, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(BRAKE_SEGMENT,     16, 44, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(FOG_SEGMENT,       45, 50, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(REVERSE_SEGMENT,   51, 53, FX_MODE_STATIC, color, 1000, false);
}
```
From there the segments just take care of themselves. The processLED() function just updates the segment color or effect mode depending on which button is active.

Here is the entire sketch:
```c++
#include <WS2812FX.h>

// define LED strip parameters
#define LED_COUNT_STRIP_1 54 // Number of LEDs in strip 1
#define LED_COUNT_STRIP_2 54 // Number of LEDs in strip 2
#define LED_PIN_STRIP_1   10 // GPIO pin used to drive strip 1
#define LED_PIN_STRIP_2   11 // GPIO pin used to drive strip 2

// define button pins
#define DRL_BUTTON            2 // GPIO pin assigned to the DRL button
#define RUNNING_LIGHTS_BUTTON 3 // GPIO pin assigned to the running lights button
#define BRAKE_LIGHTS_BUTTON   4
#define FOG_LIGHTS_BUTTON     5
#define REVERSE_LIGHTS_BUTTON 6
#define LEFT_LIGHTS_BUTTON    7
#define RIGHT_LIGHTS_BUTTON   8

// define segment indexes
#define INDICATOR_SEGMENT 0 // left/right indicator segment index
#define BRAKE_SEGMENT     1 // brake lights segment index
#define FOG_SEGMENT       2 // fog lights segment index
#define REVERSE_SEGMENT   3 // reverse segment index

// change these defines based on your button hardware setup
#define BUTTON_ACTIVE LOW // define button active state, HIGH or LOW
#define INPUT_TYPE INPUT_PULLUP // define GPIO input type, INPUT or INPUT_PULLUP

// some color intensity modifiers
#define DIM(c)      (uint32_t)(c & 0xbfbfbfbf) // 75% brightness
#define DIMMER(c)   (uint32_t)(c & 0x7f7f7f7f) // 50% brightness
#define DIMMEST(c)  (uint32_t)(c & 0x3f3f3f3f) // 25% brightness

// variables to hold each button's status
bool drl_button_pressed = false;
bool running_button_pressed = false;
bool brake_button_pressed = false;
bool fog_button_pressed = false;
bool reverse_button_pressed = false;
bool left_button_pressed = false;
[... 152 lines cut ...]
  }

  if(left_button_pressed) { // left indicator lights (set strip1 only)
    ws2812fx_strip1.setColor(INDICATOR_SEGMENT, ORANGE);
    ws2812fx_strip1.getSegment(INDICATOR_SEGMENT)->mode = FX_MODE_COLOR_WIPE;
  }

  if(right_button_pressed) { // right indicator lights (set strip2 only)
    ws2812fx_strip2.setColor(INDICATOR_SEGMENT, ORANGE);
    ws2812fx_strip2.getSegment(INDICATOR_SEGMENT)->mode = FX_MODE_COLOR_WIPE;
  }
}

void loop() {
  processButton();
  
  if(buttons_have_changed) {
    buttons_have_changed = false;
    processLED();
  }
  
  ws2812fx_strip1.service();
  ws2812fx_strip2.service();
}
```

craigrobbo, 2020-05-04:
Thats great, Ill give that a try shortly.

I'll keep my offer open to you indefinatly if you change your mind for some custom lighting :)

So if just for example I wanted to add in a second indicator segment / break up the segment I would do this?

Also is 10 segments the limit of the library?

`void set_all_strips(uint32_t color) {
  ws2812fx_strip1.setSegment(INDICATOR_SEGMENT,  0, 7, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(INDICATOR_SEGMENT,  8,15 , FX_MODE_STATIC, color, 1000, false);

  ws2812fx_strip1.setSegment(BRAKE_SEGMENT,     16, 44, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(FOG_SEGMENT,       45, 50, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(REVERSE_SEGMENT,   51, 53, FX_MODE_STATIC, color, 1000, false);

  ws2812fx_strip2.setSegment(INDICATOR_SEGMENT,  0, 7, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(INDICATOR_SEGMENT,  8,15 , FX_MODE_STATIC, color, 1000, false);

  ws2812fx_strip2.setSegment(BRAKE_SEGMENT,     16, 44, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(FOG_SEGMENT,       45, 50, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(REVERSE_SEGMENT,   51, 53, FX_MODE_STATIC, color, 1000, false);`

craigrobbo, 2020-05-04:
Alright so I have given it a try, i am not sure how its acting yuour end but for me there seems to be a half second delay between pressing a button & the light coming on, and then same when button is released the light stays on for a split second too?

Also the DRL/runnign light turns on in 2 segments?

For testing I am using your method too of internal pull ups and buttons rather than 5v - just so we're both on the same level :)

moose4lord, 2020-05-05:
You almost got it. To break up the indicator segment into two segments you first need to #define a fourth segment and give it a name:
```c++
#define INDICATOR_SEGMENT  0 // left/right indicator segment index
#define BRAKE_SEGMENT      1 // brake lights segment index
#define FOG_SEGMENT        2 // fog lights segment index
#define REVERSE_SEGMENT    3 // reverse segment index
#define INDICATOR_SEGMENT2 4 // second left/right indicator segment index
```
Then, as you've done, add it to the set_all_strips() function:
```c++
void set_all_strips(uint32_t color) {
  ws2812fx_strip1.setSegment(INDICATOR_SEGMENT,  0,  7, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(INDICATOR_SEGMENT2, 8, 15, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(BRAKE_SEGMENT,     16, 44, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(FOG_SEGMENT,       45, 50, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip1.setSegment(REVERSE_SEGMENT,   51, 53, FX_MODE_STATIC, color, 1000, false);

  ws2812fx_strip2.setSegment(INDICATOR_SEGMENT,  0,  7, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(INDICATOR_SEGMENT2, 8, 15, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(BRAKE_SEGMENT,     16, 44, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(FOG_SEGMENT,       45, 50, FX_MODE_STATIC, color, 1000, false);
  ws2812fx_strip2.setSegment(REVERSE_SEGMENT,   51, 53, FX_MODE_STATIC, color, 1000, false);
}
```
Presumably you'd want to active the second indicator segment at some point, so you'd need to add it to processLED(). Maybe activate both the first and second indicator segments when the indicator button is pressed:
```c++
  if(left_button_pressed) { // left indicator lights (set strip1 only)
    ws2812fx_strip1.setColor(INDICATOR_SEGMENT, ORANGE);
    ws2812fx_strip1.getSegment(INDICATOR_SEGMENT)->mode = FX_MODE_COLOR_WIPE;
    ws2812fx_strip1.setColor(INDICATOR_SEGMENT2, BLUE);
  }
```

And, yes, 10 segments is the libraries limit.

The half second delay you're seeing is because the default speed for all segments 1 second (1000 milliseconds).
```c++
ws2812fx_strip1.setSegment(INDICATOR_SEGMENT,  0,  7, FX_MODE_STATIC, color, **1000**, false);
```
This is basically how often the code checks to see if the LEDs need to be updated, so when you press a button, it can take up to 1000ms for the LEDs to respond. Crank the speed parameter down to 100 (100ms) and the LEDs will update more frequently, the the delay you're seeing should disappear. Unfortunately, it also means that the COLOR_WIPE animation, used for the indicator lights, will run much faster. You would need to add a setSpeed() statement to the indicator code to set the speed to 1000 when that animation is running.
```c++
  if(left_button_pressed) { // left indicator lights (set strip1 only)
    ws2812fx_strip1.setColor(INDICATOR_SEGMENT, ORANGE);
    ws2812fx_strip1.getSegment(INDICATOR_SEGMENT)->mode = FX_MODE_COLOR_WIPE;
    ws2812fx_strip1.setSpeed(INDICATOR_SEGMENT, 1000); // add this
  }
```
I'm not sure what you mean by "the DRL/running light turns on in 2 segments". That's not what I see. The DRL and running lights display all segments in red, each at slightly different intensities.

craigrobbo, 2020-05-05:
Hey Keith, so I've just done a quick video responce again I think it's much easier than explaining over text :)

https://youtu.be/KHorKtzNNCw

moose4lord, 2020-05-05:
ARG! In the libraries WS2812FX.cpp file (in the documents/Arduino/libraries/WS2812FX/src folder) this is the code for the FX_MODE_STATIC effect (near line 539):
```c++
uint16_t WS2812FX::mode_static(void) {
  for(uint16_t i=SEGMENT.start; i <= SEGMENT.stop; i++) {
    setPixelColor(i, SEGMENT.colors[0]);
  }
  return 500;
}
```
The return value sets the LED update delay, which is hardcoded to 500. So for the FX_MODE_STATIC effect, it doesn't matter what you use for a speed parameter in the setSegment() statements, it's hardcoded to 500ms.
Let me say it again...ARG!

As a workaround you'll have to edit the WS2812FX.cpp file and change the return value to this:
```c++
uint16_t WS2812FX::mode_static(void) {
  for(uint16_t i=SEGMENT.start; i <= SEGMENT.stop; i++) {
    setPixelColor(i, SEGMENT.colors[0]);
  }
  return SEGMENT.speed; // change this line
}
```
That will use the setSegment() speed parameter, not 500ms, as the LED update delay.

Hardcoding the return value to 500 is arguably a bug in the library. I'll have to give that some thought.

craigrobbo, 2020-05-05:
Thanks again Keith.

Sorry about all this I feel like I've caused you a load of agro :(

It may not be a bad thing as when I'm all done I'd like to use a fade on effect for the DRL and running lights so the tight be a work around for that static bug? However the brake lights might be a little more difficult to work around.

I do genuinely appreciate all the advice too and help of course!

craigrobbo, 2020-05-05:
Further update, So i tried to set the brake lights as  COLOR_WIPE just to see if it would remove the delay but oddly it didn't, so maybe not the library delay?

The indicator and brake I have set to have the same effect but indicator ius orange and brake as red, but the brake light still has the seeminly odd delay?

moose4lord, 2020-05-05:
Hmmmm, I don't see the delay in any of the buttons anymore. You must change the WS2812FX.cpp file as I mentioned above. After that the brake button seems very responsive, whether I use STATIC or COLOR_WIPE.

Here's my latest test code:
```c++
#include <WS2812FX.h>

// define LED strip parameters
#define LED_COUNT_STRIP_1 54 // Number of LEDs in strip 1
#define LED_COUNT_STRIP_2 54 // Number of LEDs in strip 2
#define LED_PIN_STRIP_1   10 // GPIO pin used to drive strip 1
#define LED_PIN_STRIP_2   11 // GPIO pin used to drive strip 2

// define button pins
#define DRL_BUTTON            2 // GPIO pin assigned to the DRL button
#define RUNNING_LIGHTS_BUTTON 3 // GPIO pin assigned to the running lights button
#define BRAKE_LIGHTS_BUTTON   4
#define FOG_LIGHTS_BUTTON     5
#define REVERSE_LIGHTS_BUTTON 6
#define LEFT_LIGHTS_BUTTON    7
#define RIGHT_LIGHTS_BUTTON   8

// define segment indexes
#define INDICATOR_SEGMENT  0 // left/right indicator segment index
#define BRAKE_SEGMENT      1 // brake lights segment index
#define FOG_SEGMENT        2 // fog lights segment index
#define REVERSE_SEGMENT    3 // reverse segment index
#define INDICATOR_SEGMENT2 4 // second left/right indicator segment index

// change these defines based on your button hardware setup
#define BUTTON_ACTIVE LOW // define button active state, HIGH or LOW
#define INPUT_TYPE INPUT_PULLUP // define GPIO input type, INPUT or INPUT_PULLUP

// some color intensity modifiers
#define DIM(c)      (uint32_t)(c & 0xbfbfbfbf) // 75% brightness
#define DIMMER(c)   (uint32_t)(c & 0x7f7f7f7f) // 50% brightness
#define DIMMEST(c)  (uint32_t)(c & 0x3f3f3f3f) // 25% brightness

// variables to hold each button's status
bool drl_button_pressed = false;
bool running_button_pressed = false;
bool brake_button_pressed = false;
bool fog_button_pressed = false;
bool reverse_button_pressed = false;
bool left_button_pressed = false;
bool right_button_pressed = false;

bool buttons_have_changed = false; // flag indicates when any button has changed state

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

void setup() {

   Serial.begin(115200);
   while(!Serial);
   delay(500);

  // setup button pin config
  pinMode(DRL_BUTTON,            INPUT_TYPE);
  pinMode(RUNNING_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(BRAKE_LIGHTS_BUTTON,   INPUT_TYPE);
  pinMode(FOG_LIGHTS_BUTTON,     INPUT_TYPE);
  pinMode(REVERSE_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(LEFT_LIGHTS_BUTTON,    INPUT_TYPE);
  pinMode(RIGHT_LIGHTS_BUTTON,   INPUT_TYPE);
  
  ws2812fx_strip1.init();
  ws2812fx_strip2.init();
  
  ws2812fx_strip1.setBrightness(255); // maximum brightness
[... 139 lines cut ...]
    ws2812fx_strip1.setSpeed(INDICATOR_SEGMENT, 1000);
    ws2812fx_strip1.getSegment(INDICATOR_SEGMENT)->mode = FX_MODE_COLOR_WIPE;
    ws2812fx_strip1.setColor(INDICATOR_SEGMENT2, BLUE); // test the second indicator segment
  }

  if(right_button_pressed) { // right indicator lights (set strip2 only)
    ws2812fx_strip2.setColor(INDICATOR_SEGMENT, ORANGE);
    ws2812fx_strip2.setSpeed(INDICATOR_SEGMENT, 1000);
    ws2812fx_strip2.getSegment(INDICATOR_SEGMENT)->mode = FX_MODE_COLOR_WIPE;
    ws2812fx_strip2.setColor(INDICATOR_SEGMENT2, BLUE); // test the second indicator segment
  }
}

void loop() {
  processButton();
  
  if(buttons_have_changed) {
    buttons_have_changed = false;
    processLED();
  }

  ws2812fx_strip1.service();
  ws2812fx_strip2.service();
}
```

craigrobbo, 2020-05-05:
I changed the .CPP file as requested. this did make it "marginally" quicker but it still turns on with a delay and doesnt turn off instanly.

The strange thing is the indicator is instant and goes off instantly.

In the new code the indicator lights but the second segment has that delay too?

I think i might be asking too much from an arduino 

This isn't my video this is someone else, they said they use an arduino to do this (not sure on the library)

but this was the kind of thing I wanted to end up with ideally.

https://www.youtube.com/watch?v=G5H-9q5yi54

craigrobbo, 2020-05-05:
I am wondering if I am just expecting too much, it could be possible for me to change the design of the ligths in such a way that I just use a standard LED set for Brake, reverse & fog - Just use the arduino for DRL/Running light & indicator only?

This would at least simplify the coding.

moose4lord, 2020-05-06:
I don't think you're expecting too much of this project. The hardware should be able to do what you have in mind and I think we're making progress on the software.  I've been using web apps or Android apps to control my lights for a long time, but I've never tried to use actual buttons to control them. To be sure, managing multiple buttons that control overlapping segments is challenging.

I watched the video. I doubt the guy is using the WS2812FX library. It looks like custom code to me. Pretty impressive though.

Anyway, I reworked the code again to fix a problem with the animation reseting when the DRL or running lights buttons were pressed. I gave up on trying to keep the button code separate from the LED code, so the processLED() function is gone and now everything is done in the processButton() function, like you had originally done. It's messy, but seems to work for me.

BTW, you're not really being a pest. This has turned into a good learning experience for me and I'm kinda' enjoying the debug process.

```c++
#include <WS2812FX.h>

// define LED strip parameters
#define LED_COUNT_STRIP_1 54 // Number of LEDs in strip 1
#define LED_COUNT_STRIP_2 54 // Number of LEDs in strip 2
#define LED_PIN_STRIP_1   10 // GPIO pin used to drive strip 1
#define LED_PIN_STRIP_2   11 // GPIO pin used to drive strip 2

// define button pins
#define DRL_BUTTON            2 // GPIO pin assigned to the DRL button
#define RUNNING_LIGHTS_BUTTON 3 // GPIO pin assigned to the running lights button
#define BRAKE_LIGHTS_BUTTON   4
#define FOG_LIGHTS_BUTTON     5
#define REVERSE_LIGHTS_BUTTON 6
#define LEFT_LIGHTS_BUTTON    7
#define RIGHT_LIGHTS_BUTTON   8

// define segment indexes
#define INDICATOR_SEGMENT  0 // left/right indicator segment index
#define BRAKE_SEGMENT      1 // brake lights segment index
#define FOG_SEGMENT        2 // fog lights segment index
#define REVERSE_SEGMENT    3 // reverse segment index
#define INDICATOR_SEGMENT2 4 // second left/right indicator segment index

// change these defines based on your button hardware setup
#define BUTTON_ACTIVE LOW // define button active state, HIGH or LOW
#define INPUT_TYPE INPUT_PULLUP // define GPIO input type, INPUT or INPUT_PULLUP

// some color intensity modifiers
#define DIM(c)      (uint32_t)(c & 0xbfbfbfbf) // 75% brightness
#define DIMMER(c)   (uint32_t)(c & 0x7f7f7f7f) // 50% brightness
#define DIMMEST(c)  (uint32_t)(c & 0x3f3f3f3f) // 25% brightness

// variables to hold each button's status
bool drl_button_pressed = false;
bool running_button_pressed = false;
bool brake_button_pressed = false;
bool fog_button_pressed = false;
bool reverse_button_pressed = false;
bool left_button_pressed = false;
bool right_button_pressed = false;

bool buttons_have_changed = false; // flag indicates when any button has changed state

uint32_t default_color = BLACK;
uint8_t  default_mode  = FX_MODE_STATIC;
uint16_t default_speed = 100;

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

void setup() {

  //  Serial.begin(115200);
  //  while(!Serial);
  //  delay(500);

  // setup button pin config
  pinMode(DRL_BUTTON,            INPUT_TYPE);
  pinMode(RUNNING_LIGHTS_BUTTON, INPUT_TYPE);
  pinMode(BRAKE_LIGHTS_BUTTON,   INPUT_TYPE);
[... 124 lines cut ...]
    setSegment_strip1(INDICATOR_SEGMENT,  default_color, default_mode, default_speed);
    setSegment_strip1(INDICATOR_SEGMENT2, default_color, default_mode, default_speed);
  }
  
  // Right indicator lights button
  if(digitalRead(RIGHT_LIGHTS_BUTTON) == BUTTON_ACTIVE && right_button_pressed == false) { // Right lights button switched from off to on
    // Serial.println("Right");
    right_button_pressed = true;
    setSegment_strip2(INDICATOR_SEGMENT,  ORANGE, FX_MODE_COLOR_WIPE, 1000);
    setSegment_strip2(INDICATOR_SEGMENT2, BLUE,   FX_MODE_COLOR_WIPE, 1000);
  } else if((updateBackground && !right_button_pressed) ||
            (digitalRead(RIGHT_LIGHTS_BUTTON) != BUTTON_ACTIVE && right_button_pressed == true)) { // Right lights button switched from on to off
    right_button_pressed = false;
    setSegment_strip2(INDICATOR_SEGMENT,  default_color, default_mode, default_speed);
    setSegment_strip2(INDICATOR_SEGMENT2, default_color, default_mode, default_speed);
  }
}

void loop() {
  processButton();

  ws2812fx_strip1.service();
  ws2812fx_strip2.service();
}
```

craigrobbo, 2020-05-06:
Thanks again Keith...I feel we are actually getting somewhere now properly!

This all works as intended!

The only little issue I have now is that the DRL is at a higher brightness level than the Running light, and when they are both pressed the DRl seems to take a priority but the running light really needs to be the priority here so the running light dulls when its pressed as opposed to brightening up (but without reducing brightness of other effects?)

If thats a big issue then don't worry I'll just change the way I hard wire them into the car.

One last thing, is there any way to remove the 10 segment limit as that may be quite  a limiting factor when im working with a lot of segments over different animations?

moose4lord, 2020-05-07:
The 10 segment limit is hardcoded in the libraries WS2812FX.h file (line 63).
```c++
#define MAX_NUM_SEGMENTS 10
```
You can hack it to whatever you want, but each segment uses 36 bytes of SRAM, so you can't go crazy if you're working on a standard Arduino with only 2KB of SRAM.

Speaking of memory, the sketch is already using quite a bit. You might want to consider getting an Arduino Mega560, which has much more memory then a regular Arduino. You'll be able to add a lot of LEDs and features, and not have to worry about running out of memory. Just a thought.

craigrobbo, 2020-05-07:
Thanks again Keith.

So it's quite simple to remove the hard code limit then?

Also you read my mind! The uno was just a test board but the plan is to go to a bigger better board in the long run.

I did purchase an esp8266 until I realised it didn't have enough use able input and output pins :(

moose4lord, 2020-05-07:
Yeah, the ESP8266 probably isn't a good fit for your application. Not a lot of I/O, and it's a 3.3 volt processor, so it's pain to connect to 5 volt LEDs.

craigrobbo, 2020-05-07:
> 
> 
> Yeah, the ESP8266 probably isn't a good fit for your application. Not a lot of I/O, and it's a 3.3 volt processor, so it's pain to connect to 5 volt LEDs.

I think an ESP32 or an arduino mega2560 might be a good idea to go to? the ESP has a few extra features too.

im really thinkful the code works now it makes things great!

I did have a couple of requests but if they are going to be huge changes its not a problem.

The first one is the indicator blink, is it possible to put a "delay" on the input signal so to speak

The indicator from the car is done in pulses is there a way in code that itll only go back to its previous state after a set amount of time? (if this is a lot of work then I can just use some resistors & capacitors to smooth out the pulses)

The second thing again only if its fairly simple, is it possible for one segment to start after the previous one has finished for the indicator  so just for example one halo would light then the next one (a bit like in my Alfa 159)

Thanks again :)

moose4lord, 2020-05-09:
I added two timer to manage turning the indicators off after a delay. The length of the delay is controlled by this constant:
```c++
#define INDICATOR_DELAY 3000
```

Your second request is harder. WS2812FX doesn't provide a way to sequence effects across segments. I'm not really sure how to do that.

Here's the latest code:
```c+
#include <WS2812FX.h>
#include <limits.h>

// define LED strip parameters
#define LED_COUNT_STRIP_1 54 // Number of LEDs in strip 1
#define LED_COUNT_STRIP_2 54 // Number of LEDs in strip 2
#define LED_PIN_STRIP_1   10 // GPIO pin used to drive strip 1
#define LED_PIN_STRIP_2   11 // GPIO pin used to drive strip 2

// define button pins
#define DRL_BUTTON            2 // GPIO pin assigned to the DRL button
#define RUNNING_LIGHTS_BUTTON 3 // GPIO pin assigned to the running lights button
#define BRAKE_LIGHTS_BUTTON   4
#define FOG_LIGHTS_BUTTON     5
#define REVERSE_LIGHTS_BUTTON 6
#define LEFT_LIGHTS_BUTTON    7
#define RIGHT_LIGHTS_BUTTON   8

// define segment indexes
#define INDICATOR_SEGMENT  0 // left/right indicator segment index
#define BRAKE_SEGMENT      1 // brake lights segment index
#define FOG_SEGMENT        2 // fog lights segment index
#define REVERSE_SEGMENT    3 // reverse segment index
#define INDICATOR_SEGMENT2 4 // second left/right indicator segment index

// change these defines based on your button hardware setup
#define BUTTON_ACTIVE LOW // define button active state, HIGH or LOW
#define INPUT_TYPE INPUT_PULLUP // define GPIO input type, INPUT or INPUT_PULLUP

// some color intensity modifiers
#define DIM(c)      (uint32_t)(c & 0xbfbfbfbf) // 75% brightness
#define DIMMER(c)   (uint32_t)(c & 0x7f7f7f7f) // 50% brightness
#define DIMMEST(c)  (uint32_t)(c & 0x3f3f3f3f) // 25% brightness

// timer constants
#define INDICATOR_DELAY 3000 // delay turning the indicators off by 3000ms

// variables to hold each button's status
bool drl_button_pressed = false;
bool running_button_pressed = false;
bool brake_button_pressed = false;
bool fog_button_pressed = false;
bool reverse_button_pressed = false;
bool left_button_pressed = false;
bool right_button_pressed = false;

bool buttons_have_changed = false; // flag indicates when any button has changed state

unsigned long left_indicator_timer  = ULONG_MAX;
unsigned long right_indicator_timer = ULONG_MAX;

uint32_t default_color = BLACK;
uint8_t  default_mode  = FX_MODE_STATIC;
uint16_t default_speed = 100;

WS2812FX ws2812fx_strip1 = WS2812FX(LED_COUNT_STRIP_1, LED_PIN_STRIP_1, NEO_GRB + NEO_KHZ800);
WS2812FX ws2812fx_strip2 = WS2812FX(LED_COUNT_STRIP_2, LED_PIN_STRIP_2, NEO_GRB + NEO_KHZ800);

void setup() {

//  Serial.begin(115200);
[... 151 lines cut ...]
  } else if(digitalRead(RIGHT_LIGHTS_BUTTON) != BUTTON_ACTIVE && right_button_pressed == true) {
    right_button_pressed = false;
  }
  // if indicator button pressed and timer not running, start the indicator display
  if(right_button_toggle && right_indicator_timer == ULONG_MAX) {
    setSegment_strip2(INDICATOR_SEGMENT,  ORANGE, FX_MODE_COLOR_WIPE, 1000);
    setSegment_strip2(INDICATOR_SEGMENT2, BLUE,   FX_MODE_COLOR_WIPE, 1000);
  }
  // keep the timer running as long as the button is pressed
  if(right_button_pressed) right_indicator_timer = now + INDICATOR_DELAY;
  // if the background needs to be updated or the timer expires, reset the indicator display
  if((updateBackground && right_indicator_timer == ULONG_MAX) || now > right_indicator_timer) {
      right_indicator_timer = ULONG_MAX; // reset the indicator timer
      setSegment_strip2(INDICATOR_SEGMENT,  default_color, default_mode, default_speed);
      setSegment_strip2(INDICATOR_SEGMENT2, default_color, default_mode, default_speed);
  }
}

void loop() {
  processButton();

  ws2812fx_strip1.service();
  ws2812fx_strip2.service();
}
```

craigrobbo, 2020-05-09:
Thanks again Keith!

Yeah I think doing the effect across several segments might not be possible.

For example if I had to Halos for one to do the animation then the one after it to do the animation after the first one.

unless i could implement a delay function some how so the second segment starts after xx time?

[4 more replies]

closed 2020-06-17 by craigrobbo